tbsdtv/linux_media #284: Issue with installing driver in Ubuntu 22.04

dorinilut, 2022-06-05:
hello,

Since moving to ubuntu 22.04 i am getting the following error when building followed by a kernel panic whenever I am doing the reboot:

`root@dorin:/home/dorin/TBS/media_build# make rmmod
make -C /home/dorin/TBS/media_build/v4l rmmod
make[1]: Entering directory '/home/dorin/TBS/media_build/v4l'
scripts/rmmod.pl unload
found 737 modules
/usr/sbin/rmmod videodev
Pulseaudio is running with UUID(s): 1000
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module videodev is in use by: cx2341x videobuf2_v4l2 videobuf2_common cx23885
/usr/sbin/rmmod videobuf2_vmalloc
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module videobuf2_vmalloc is in use by: dvb_core
/usr/sbin/rmmod videobuf2_memops
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module videobuf2_memops is in use by: videobuf2_vmalloc videobuf2_dma_sg
/usr/sbin/rmmod tveeprom
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module tveeprom is in use by: cx23885
/usr/sbin/rmmod cx23885
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod altera_ci
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod cx2341x
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod dvb_core
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module dvb_core is in use by: saa716x_tbs_dvb m88ds3103 videobuf2_dvb saa716x_core
/usr/sbin/rmmod tda18271
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod cx23885
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module cx23885 is not currently loaded
/usr/sbin/rmmod saa716x_tbs_dvb
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod m88ds3103
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod saa716x_core
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod videobuf2_dvb
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod altera_ci
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module altera_ci is not currently loaded
/usr/sbin/rmmod dvb_core
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod videobuf2_vmalloc
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
/usr/sbin/rmmod videobuf2_dma_sg
No PulseAudio daemon running, or not running as session daemon.
[... 111 lines cut ...]
#new build
git clone https://github.com/tbsdtv/media_build.git -b extra
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media

cd media_build
make dir DIR=../media

./install.sh

#build and make tbs opensource drivers
sudo make rmmod
sudo modprobe cx23885
sudo modprobe saa716x_tbs_dvb int_type=1
lspci
dmesg | grep frontend

#load "saa716x_tbs-dvb" modules
#echo saa716x_tbs-dvb | sudo tee /etc/modules-load.d/tbs.conf

#install dvb-apps
sudo apt --yes install dvb-apps
sudo lsdvb

#sudo reboot
`

crazycat69, 2022-06-05:
what dvb devices installed ?

dorinilut, 2022-06-05:
it's a 6590. here are the details from lspci

0b:00.0 Multimedia controller: TBS Technologies DVB Tuner PCIe Card
        Subsystem: Device 6590:0001
        Flags: bus master, fast devsel, latency 0, IOMMU group 19
        Memory at c4000000 (32-bit, non-prefetchable) [size=256K]
        Capabilities: [50] Power Management version 3
        Capabilities: [70] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [90] Express Endpoint, MSI 00
        Capabilities: [100] Device Serial Number 00-00-00-00-00-00-00-00
        Kernel modules: tbsecp3

crazycat69, 2022-06-05:
so why loaded cx23885 and saa716x_tbs_dvb  ?

dorinilut, 2022-06-05:
i know this is going to sound stupid, but because that what it said in the guide. should i remove that step? this script has worked for me since 18.04. and I also have a  Hauppauge WinTV-dualHD DVB.

dorinilut, 2022-06-07:
I commented out the section with the loading modules. I will try again and post. I would kindly ask if there are any other recommendations.

dorinilut, 2022-06-07:
I commented the loading of the modules and tied a new install. This time it worked. I will close this now.