tbsdtv/linux_media #285: Lots of compiling errors on 5.18.2

ich777, 2022-06-07:
Hi,
got a lot of compiling erros when compiling for stable Kernel version 5.18.2:

```
  CC [M]  /usr/src/TBS_OS/media_build/v4l/ivtv-ioctl.o
In file included from /usr/src/TBS_OS/media_build/v4l/ivtv-driver.c:46:
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_might_use_pio':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:20:19: error: 'PCI_DMA_NONE' undeclared (first use in this function); did you mean 'DMA_NONE'?
   20 |  return s->dma == PCI_DMA_NONE || (SLICED_VBI_PIO && s->type == IVTV_ENC_STREAM_TYPE_VBI);
      |                   ^~~~~~~~~~~~
      |                   DMA_NONE
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:20:19: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_use_pio':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:27:19: error: 'PCI_DMA_NONE' undeclared (first use in this function); did you mean 'DMA_NONE'?
   27 |  return s->dma == PCI_DMA_NONE ||
      |                   ^~~~~~~~~~~~
      |                   DMA_NONE
  CC [M]  /usr/src/TBS_OS/media_build/v4l/ivtv-irq.o
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_might_use_dma':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:33:19: error: 'PCI_DMA_NONE' undeclared (first use in this function); did you mean 'DMA_NONE'?
   33 |  return s->dma != PCI_DMA_NONE;
      |                   ^~~~~~~~~~~~
      |                   DMA_NONE
  CC [M]  /usr/src/TBS_OS/media_build/v4l/ivtv-mailbox.o
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_buf_sync_for_cpu':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:44:3: error: implicit declaration of function 'pci_dma_sync_single_for_cpu'; did you mean 'dma_sync_single_for_cpu'? [-Werror=implicit-function-declaration]
   44 |   pci_dma_sync_single_for_cpu(s->itv->pdev, buf->dma_handle,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   dma_sync_single_for_cpu
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_buf_sync_for_device':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:51:3: error: implicit declaration of function 'pci_dma_sync_single_for_device'; did you mean 'dma_sync_single_for_device'? [-Werror=implicit-function-declaration]
   51 |   pci_dma_sync_single_for_device(s->itv->pdev, buf->dma_handle,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   dma_sync_single_for_device
In file included from /usr/src/TBS_OS/media_build/v4l/ivtv-fileops.c:13:
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_might_use_pio':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:20:19: error: 'PCI_DMA_NONE' undeclared (first use in this function); did you mean 'DMA_NONE'?
   20 |  return s->dma == PCI_DMA_NONE || (SLICED_VBI_PIO && s->type == IVTV_ENC_STREAM_TYPE_VBI);
      |                   ^~~~~~~~~~~~
      |                   DMA_NONE
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:20:19: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_stream_sync_for_cpu':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:74:36: error: 'PCI_DMA_TODEVICE' undeclared (first use in this function); did you mean 'DMA_TO_DEVICE'?
   74 |    sizeof(struct ivtv_sg_element), PCI_DMA_TODEVICE);
      |                                    ^~~~~~~~~~~~~~~~
      |                                    DMA_TO_DEVICE
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_use_pio':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:27:19: error: 'PCI_DMA_NONE' undeclared (first use in this function); did you mean 'DMA_NONE'?
   27 |  return s->dma == PCI_DMA_NONE ||
      |                   ^~~~~~~~~~~~
      |                   DMA_NONE
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_might_use_dma':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:33:19: error: 'PCI_DMA_NONE' undeclared (first use in this function); did you mean 'DMA_NONE'?
   33 |  return s->dma != PCI_DMA_NONE;
      |                   ^~~~~~~~~~~~
      |                   DMA_NONE
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_stream_sync_for_device':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:81:36: error: 'PCI_DMA_TODEVICE' undeclared (first use in this function); did you mean 'DMA_TO_DEVICE'?
   81 |    sizeof(struct ivtv_sg_element), PCI_DMA_TODEVICE);
      |                                    ^~~~~~~~~~~~~~~~
      |                                    DMA_TO_DEVICE
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h: In function 'ivtv_buf_sync_for_cpu':
/usr/src/TBS_OS/media_build/v4l/ivtv-queue.h:44:3: error: implicit declaration of function 'pci_dma_sync_single_for_cpu'; did you mean 'dma_sync_single_for_cpu'? [-Werror=implicit-function-declaration]
   44 |   pci_dma_sync_single_for_cpu(s->itv->pdev, buf->dma_handle,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   dma_sync_single_for_cpu
In file included from /usr/src/TBS_OS/media_build/v4l/ivtv-driver.c:47:
/usr/src/TBS_OS/media_build/v4l/ivtv-udma.h: In function 'ivtv_udma_sync_for_device':
/usr/src/TBS_OS/media_build/v4l/ivtv-udma.h:27:30: error: 'PCI_DMA_TODEVICE' undeclared (first use in this function); did you mean 'DMA_TO_DEVICE'?
   27 |   sizeof(itv->udma.SGarray), PCI_DMA_TODEVICE);
[... 414 lines cut ...]
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h: In function 'cx18_buf_sync_for_cpu':
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h:18:2: error: implicit declaration of function 'pci_dma_sync_single_for_cpu'; did you mean 'dma_sync_single_for_cpu'? [-Werror=implicit-function-declaration]
   18 |  pci_dma_sync_single_for_cpu(s->cx->pci_dev, buf->dma_handle,
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  dma_sync_single_for_cpu
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h: In function 'cx18_buf_sync_for_device':
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h:25:2: error: implicit declaration of function 'pci_dma_sync_single_for_device'; did you mean 'dma_sync_single_for_device'? [-Werror=implicit-function-declaration]
   25 |  pci_dma_sync_single_for_device(s->cx->pci_dev, buf->dma_handle,
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  dma_sync_single_for_device
In file included from /usr/src/TBS_OS/media_build/v4l/cx18-dvb.c:12:
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h: In function 'cx18_buf_sync_for_cpu':
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h:18:2: error: implicit declaration of function 'pci_dma_sync_single_for_cpu'; did you mean 'dma_sync_single_for_cpu'? [-Werror=implicit-function-declaration]
   18 |  pci_dma_sync_single_for_cpu(s->cx->pci_dev, buf->dma_handle,
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  dma_sync_single_for_cpu
  CC [M]  /usr/src/TBS_OS/media_build/v4l/cx23885-av.o
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h: In function 'cx18_buf_sync_for_device':
/usr/src/TBS_OS/media_build/v4l/cx18-queue.h:25:2: error: implicit declaration of function 'pci_dma_sync_single_for_device'; did you mean 'dma_sync_single_for_device'? [-Werror=implicit-function-declaration]
   25 |  pci_dma_sync_single_for_device(s->cx->pci_dev, buf->dma_handle,
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |  dma_sync_single_for_device
cc1: some warnings being treated as errors
make[3]: [scripts/Makefile.build:288: /usr/src/TBS_OS/media_build/v4l/cx18-dvb.o] Error 1
```

tryptichon, 2022-06-22:
You could try the patch I hacked together here, although you're seeing some errors that I have not seen while I was creating the patch.

https://github.com/tbsdtv/linux_media/issues/283#issuecomment-1162672107

ich777, 2022-06-22:
@tryptichon on what Kernel version did you try to build the drivers?

I will try your patch shortly and report back. Thank you!

tryptichon, 2022-06-22:
See the link to the other comment. It is 5.18.5-200 from Fedora 36.

ich777, 2022-07-09:
@tryptichon thank you for the patch from #283, fixes this issue here too. :)
Helps me a lot since I currently don't have the time to patch the driver myself!

Now I am finally able to create the driver package for Unraid again (tested against Kernel version 5.18.10).

closed 2022-07-09 by ich777