tbsdtv/linux_media #286: Build fails in Fedora 36 kernel 5.18.5-200.fc36.x86_64

fe31nz, 2022-06-22:
I tried compiling in a VirtualBox Fedora 36 with the current kernel (5.18.5-200.fc36.x86_64) and it fails on incompatible pointer types:

/root/projects/tbs6909-drivers/media_build/v4l/msi001.c:504:27: error: initialization of 'void (*)(struct spi_device *)' from incompatible pointer type 'int (*)(struct spi_device *)' [-Werror=incompatible-pointer-types]
  504 |         .remove         = msi001_remove,
      |                           ^~~~~~~~~~~~~

It looks like this warning is being upgraded to an error as a result of an option being passed to gcc, or maybe it is now the default with gcc 12.1.1 which is current on Fedora 36.  I have attached my full build script file and the log of its output:

./download_and_build.sh 2>&1 | tee build.log

fe31nz, 2022-06-22:
Attaching the files did not work, so here they are:

download_and_build.sh:
#!/bin/bash

set -x
cd /root/projects/tbs6909-drivers || exit 1
rm -r media
rm -r media_build
git clone https://github.com/tbsdtv/media_build.git || exit 2
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media || exit 3

cd media_build
make dir DIR=../media || exit 11
make allyesconfig || exit 12
make -j$(($(getconf _NPROCESSORS_ONLN)+1)) || exit 13
#make || exit 13

make install
#restart

build.log:
+ cd /root/projects/tbs6909-drivers
+ rm -r media
rm: cannot remove 'media': No such file or directory
+ rm -r media_build
+ git clone https://github.com/tbsdtv/media_build.git
Cloning into 'media_build'...
+ git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
Cloning into './media'...
Updating files: 100% (74560/74560), done.
+ cd media_build
+ make dir DIR=../media
make -C linux/ dir DIR="../../media"
make[1]: Entering directory '/root/projects/tbs6909-drivers/media_build/linux'
rm -rf drivers include sound mm .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
Searching in ../../media/Makefile for kernel version.
./use_dir.pl ../../media
sync file: include/uapi/linux/cec.h
sync file: include/linux/atmel-isc-media.h
sync file: include/misc/altera.h
sync file: include/uapi/linux/v4l2-mediabus.h
sync file: include/linux/mmc/sdio_ids.h
sync file: include/uapi/linux/videodev2.h
sync file: include/uapi/linux/ccs.h
sync file: include/trace/events/pwc.h
sync file: include/uapi/linux/cec-funcs.h
sync file: include/linux/ti_wilink_st.h
sync file: include/uapi/linux/meye.h
sync file: include/uapi/linux/ivtvfb.h
sync file: include/linux/of_graph.h
sync file: include/uapi/linux/xilinx-v4l2-controls.h
sync file: sound/pci/bt87x.c
sync file: include/uapi/linux/ivtv.h
sync file: include/uapi/linux/lirc.h
sync file: include/trace/events/vb2.h
sync file: include/uapi/linux/media.h
sync file: include/sound/aci.h
sync file: include/uapi/linux/max2175.h
sync file: include/uapi/linux/v4l2-controls.h
sync file: include/uapi/linux/usb/video.h
sync file: include/trace/events/v4l2.h
sync file: include/linux/via-core.h
sync file: include/linux/hdmi.h
sync file: include/uapi/linux/uvcvideo.h
sync file: include/uapi/linux/v4l2-subdev.h
sync file: include/linux/fixp-arith.h
sync file: include/linux/dma/xilinx_dma.h
sync file: include/uapi/linux/v4l2-dv-timings.h
sync file: include/linux/pci_ids.h
[... 247 lines cut ...]
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/tda18271-common.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/tda18271-fe.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/tda18273.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/xc5000.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/xc4000.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/msi001.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/mt2060.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/mt2063.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/mt2266.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/qt1010.o
/root/projects/tbs6909-drivers/media_build/v4l/msi001.c:504:27: error: initialization of 'void (*)(struct spi_device *)' from incompatible pointer type 'int (*)(struct spi_device *)' [-Werror=incompatible-pointer-types]
  504 |         .remove         = msi001_remove,
      |                           ^~~~~~~~~~~~~
/root/projects/tbs6909-drivers/media_build/v4l/msi001.c:504:27: note: (near initialization for 'msi001_driver.remove')
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/mt2131.o
  CC [M]  /root/projects/tbs6909-drivers/media_build/v4l/mxl5005s.o
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:288: /root/projects/tbs6909-drivers/media_build/v4l/msi001.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1838: /root/projects/tbs6909-drivers/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/kernels/5.18.5-200.fc36.x86_64'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/projects/tbs6909-drivers/media_build/v4l'
make: *** [Makefile:26: all] Error 2
+ exit 13

tryptichon, 2022-06-22:
If you are familiar with patching, try this: https://github.com/tbsdtv/linux_media/issues/283#issuecomment-1162672107

crazycat69, 2023-03-22:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-22 by crazycat69