tbsdtv/linux_media #288: Build fails on ubuntu 22.04LTS kernel 5.15.0-41-generic

4bu8Repricant, 2022-07-25:
Hi there.

I am working on installation of TBS6902 on PC running ubuntu 22.04LTS.
A few days ago, the kernel was upgraded to 5.15.0-41-generic.
It was an accident and MKDS is not under deploy, so that I started updating drivers as folloing.

~ $ cd media
~/media $ git remote update
~/media $ git pull
~/media $ cd ../media_build
~/media_build $ git remote update
~/media_build $ git pull
~/media_build $ make 

Just like mentioned in "Prerequisites, Build errors and Rebuilding" in wiki.
But after make command issued, I got the error.

kantotv@kantotv-desktop:~/media_build$ make
make -C /home/kantotv/media_build/v4l
make[1]: Entering directory '/home/kantotv/media_build/v4l'
make[1]: *** No rule to make target '../linux/drivers/media/platform/stm32/Makefile', needed by 'Makefile.media'.  Stop.
make[1]: Leaving directory '/home/kantotv/media_build/v4l'
make: *** [Makefile:26: all] Error 2

It seems that the rule in Makefile is missing and the file has been updated recently.

It will be very helpful if some sort of advice is provided.

crazycat69, 2022-07-25:
Just tested on this kernel - build ok
maybe need **make distclean** before build ?

4bu8Repricant, 2022-07-26:
Thank you so much.
I am sorry it was my careless mistake.

it was "make release"!

After that, make and make install were fine and driver is loaded successfully.

Thank very much and I think I can close this issue.

closed 2022-07-26 by 4bu8Repricant

yazun, 2023-04-30:
Hello,
Just updated to Ubuntu22, fetched new sources and getting errors on compile related to nonexisting patches (on a kernel I do not have)

```
Sync'ing file drivers/media/rc/nuvoton-cir.c (orig = 35b997bab912881597cabcc6ecc139a5ee774736, copy = 46f83769d4fab750910471a11adba82f4430b210, patched = )
Sync'ing file drivers/media/rc/ir-imon-decoder.c (orig = c76815791c66c864d0ca6092bfff46b327df66ba, copy = d8350b2e332c41ca382612107cac942a5e55fc13, patched = )
Sync'ing file drivers/media/rc/mceusb.c (orig = 7973b739911a9529167b9933a74f8b1b0fd4f064, copy = 33fea060cfdc44e4fc1b9971f832dd97bbdcaece, patched = )
Sync'ing file drivers/media/rc/winbond-cir.c (orig = c2845e6bbfe8452bfc53d4d8625265ba61ebcc91, copy = 73111f12ff32de8ecd76b6cd25ba92a1f5b6650a, patched = )
Sync'ing file drivers/media/rc/ene_ir.c (orig = 233dcadf8bd4daf1551a608b44a1216c59239099, copy = 823c75c09991d18eed358d81c10933a3e2ca7dd9, patched = )
make[3]: Entering directory '/root/source/media_build/linux'
Unapplying patches
patch -s -f -R -p1 -i ../backports/v5.18_v4l2.patch
14 out of 14 hunks FAILED -- saving rejects to file drivers/media/v4l2-core/v4l2-ioctl.c.rej
make[3]: Leaving directory '/root/source/media_build/linux'
Applying patches for kernel 5.15.0-71-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/ccs.patch
patch -s -f -N -p1 -i ../backports/rc-cec.patch
patch -s -f -N -p1 -i ../backports/v6.2_class.patch
patch -s -f -N -p1 -i ../backports/v6.2_remove.patch
patch -s -f -N -p1 -i ../backports/v6.2_v4l2-core.patch
patch -s -f -N -p1 -i ../backports/v6.1_class.patch
patch -s -f -N -p1 -i ../backports/v6.1_i2c_probe.patch
patch -s -f -N -p1 -i ../backports/v6.1_videobuf2.patch
patch -s -f -N -p1 -i ../backports/v6.1_timer.patch
patch -s -f -N -p1 -i ../backports/v6.0_i2c_remove.patch
patch -s -f -N -p1 -i ../backports/v6.0_v4l2-tpg.patch
patch -s -f -N -p1 -i ../backports/v6.0_pt3.patch
patch -s -f -N -p1 -i ../backports/v6.0_testdrv.patch
patch -s -f -N -p1 -i ../backports/v6.0_i2c_remove_tbs.patch
patch -s -f -N -p1 -i ../backports/v5.18_rc.patch
patch -s -f -N -p1 -i ../backports/v5.18_rc2.patch
patch -s -f -N -p1 -i ../backports/v5.18_v4l2.patch
patch -s -f -N -p1 -i ../backports/v5.18_misc.patch
patch: **** Can't open patch file ../backports/v5.18_misc.patch : No such file or directory
make[2]: *** [Makefile:135: apply_patches] Error 1
make[2]: Leaving directory '/root/source/media_build/linux'
make[1]: *** [Makefile:375: allyesconfig] Error 2
make[1]: Leaving directory '/root/source/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2
root@sm:~/source/media_build# uname -a
Linux sm 5.15.0-71-generic #78-Ubuntu SMP Tue Apr 18 09:00:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```
What could be the problem? (done `make distclean` etc)

yazun, 2023-04-30:
Removal of `v5.18_misc.patch ` from the list helped.

yazun, 2023-04-30:
but then we get 
```
 CC [M]  /root/source/media_build/v4l/imx290.o
  CC [M]  /root/source/media_build/v4l/imx296.o
  CC [M]  /root/source/media_build/v4l/imx355.o
/root/source/media_build/v4l/imx290.c:1413:9: error: implicit declaration of function 'RUNTIME_PM_OPS'; did you mean 'SET_RUNTIME_PM_OPS'? [-Werror=implicit-function-declaration]
 1413 |         RUNTIME_PM_OPS(imx290_runtime_suspend, imx290_runtime_resume, NULL)
      |         ^~~~~~~~~~~~~~
      |         SET_RUNTIME_PM_OPS
/root/source/media_build/v4l/imx290.c:1413:9: warning: initialization of 'int (*)(struct device *)' from 'int' makes pointer from integer without a cast [-Wint-conversion]
/root/source/media_build/v4l/imx290.c:1413:9: note: (near initialization for 'imx290_pm_ops.prepare')
/root/source/media_build/v4l/imx290.c:1413:9: error: initializer element is not constant
/root/source/media_build/v4l/imx290.c:1413:9: note: (near initialization for 'imx290_pm_ops.prepare')
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:297: /root/source/media_build/v4l/imx290.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1907: /root/source/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-71-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/source/media_build/v4l'
make: *** [Makefile:26: all] Error 2
root@sm:~/source/media_build#
```
default gcc from ubuntu
```
gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
```

crazycat69, 2023-04-30:
Fixed
https://github.com/tbsdtv/media_build/commit/8855e24e31d0bb23e8e3a9dca58534f8eceb93b9

yazun, 2023-04-30:
Thanks, yes, `SET_RUNTIME_PM_OPS` helped locally.