tbsdtv/linux_media #289: Compilation fails on 5.19.2

ich777, 2022-08-18:
Compilation fails on Kernel 5.19.2 with this error:

```
/usr/src/TBS_OS/media_build/v4l/tm6000-video.c: In function 'tm6000_prepare_isoc':
/usr/src/TBS_OS/media_build/v4l/tm6000-video.c:573:9: error: too many arguments to function 'usb_maxpacket'
  573 |  size = usb_maxpacket(dev->udev, pipe, usb_pipeout(pipe));
      |         ^~~~~~~~~~~~~
In file included from /usr/src/TBS_OS/media_build/v4l/tm6000-video.c:20:
./include/linux/usb.h:1972:19: note: declared here
 1972 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
  CC [M]  /usr/src/TBS_OS/media_build/v4l/firedtv-fe.o
make[3]: [scripts/Makefile.build:249: /usr/src/TBS_OS/media_build/v4l/tm6000-video.o] Error 1 (ignored)
  CC [M]  /usr/src/TBS_OS/media_build/v4l/firedtv-fw.o
./scripts/check-local-export: nm failed
make[3]: [scripts/Makefile.build:249: /usr/src/TBS_OS/media_build/v4l/tm6000-video.o] Error 1 (ignored)
  CC [M]  /usr/src/TBS_OS/media_build/v4l/vidtv_tuner.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/vidtv_demod.o
  CC [M]  /usr/src/TBS_OS/media_build/v4l/vidtv_bridge.o
/usr/src/TBS_OS/media_build/v4l/tm6000-input.c: In function '__tm6000_ir_int_start':
/usr/src/TBS_OS/media_build/v4l/tm6000-input.c:343:9: error: too many arguments to function 'usb_maxpacket'
  343 |  size = usb_maxpacket(dev->udev, pipe, usb_pipeout(pipe));
      |         ^~~~~~~~~~~~~
In file included from /usr/src/TBS_OS/media_build/v4l/tm6000-input.c:13:
./include/linux/usb.h:1972:19: note: declared here
 1972 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
make[3]: [scripts/Makefile.build:250: /usr/src/TBS_OS/media_build/v4l/tm6000-input.o] Error 1 (ignored)
  CC [M]  /usr/src/TBS_OS/media_build/v4l/vidtv_common.o
/usr/src/TBS_OS/media_build/v4l/tm6000-dvb.c: In function 'tm6000_start_stream':
/usr/src/TBS_OS/media_build/v4l/tm6000-dvb.c:123:9: error: too many arguments to function 'usb_maxpacket'
  123 |  size = usb_maxpacket(dev->udev, pipe, usb_pipeout(pipe));
      |         ^~~~~~~~~~~~~
In file included from /usr/src/TBS_OS/media_build/v4l/tm6000-dvb.c:10:
./include/linux/usb.h:1972:19: note: declared here
 1972 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
./scripts/check-local-export: nm failed
```

ich777, 2022-08-24:
Completely overlooked this issue: tbsdtv/media_build#56
Will leave this open until media_build supports building on `5.19+`

crazycat69, 2023-03-22:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-22 by crazycat69