tbsdtv/linux_media #290: complilation frustrated in tumbleweeb 5.19.2

miguel1963, 2022-08-25:
Compilation fails on kernel 5.19.2 tumbleweed error:
  CC [M]  /home/Miguel/media_build/v4l/rc-tt-1500.o
  CC [M]  /home/Miguel/media_build/v4l/rc-twinhan1027.o
  CC [M]  /home/Miguel/media_build/v4l/rc-twinhan-dtv-cab-ci.o
  CC [M]  /home/Miguel/media_build/v4l/rc-vega-s9x.o
  CC [M]  /home/Miguel/media_build/v4l/rc-videomate-m1f.o
  CC [M]  /home/Miguel/media_build/v4l/rc-videomate-s350.o
  CC [M]  /home/Miguel/media_build/v4l/rc-videomate-tv-pvr.o
  CC [M]  /home/Miguel/media_build/v4l/rc-videostrong-kii-pro.o
  CC [M]  /home/Miguel/media_build/v4l/rc-wetek-hub.o
  CC [M]  /home/Miguel/media_build/v4l/rc-wetek-play2.o
  CC [M]  /home/Miguel/media_build/v4l/rc-winfast.o
  CC [M]  /home/Miguel/media_build/v4l/rc-winfast-usbii-deluxe.o
  CC [M]  /home/Miguel/media_build/v4l/rc-x96max.o
  CC [M]  /home/Miguel/media_build/v4l/rc-xbox-360.o
  CC [M]  /home/Miguel/media_build/v4l/rc-xbox-dvd.o
  CC [M]  /home/Miguel/media_build/v4l/rc-zx-irdec.o
  CC [M]  /home/Miguel/media_build/v4l/rc-main.o
  CC [M]  /home/Miguel/media_build/v4l/rc-ir-raw.o
  CC [M]  /home/Miguel/media_build/v4l/lirc_dev.o
  CC [M]  /home/Miguel/media_build/v4l/ir-imon-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-jvc-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-mce_kbd-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-nec-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-rc5-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-rc6-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-rcmm-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-sanyo-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-sharp-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-sony-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ir-xmp-decoder.o
  CC [M]  /home/Miguel/media_build/v4l/ene_ir.o
  CC [M]  /home/Miguel/media_build/v4l/fintek-cir.o
  CC [M]  /home/Miguel/media_build/v4l/igorplugusb.o
  CC [M]  /home/Miguel/media_build/v4l/iguanair.o
  CC [M]  /home/Miguel/media_build/v4l/imon.o
  CC [M]  /home/Miguel/media_build/v4l/imon_raw.o
  CC [M]  /home/Miguel/media_build/v4l/ite-cir.o
  CC [M]  /home/Miguel/media_build/v4l/mceusb.o
  CC [M]  /home/Miguel/media_build/v4l/nuvoton-cir.o
/home/Miguel/media_build/v4l/mceusb.c: In function 'mceusb_dev_probe':
/home/Miguel/media_build/v4l/mceusb.c:1754:16: error: too many arguments to function 'usb_maxpacket'
 1754 |         maxp = usb_maxpacket(dev, pipe, usb_pipeout(pipe));
      |                ^~~~~~~~~~~~~
In file included from /home/Miguel/media_build/v4l/mceusb.c:29:
/usr/src/linux-5.19.2-1/include/linux/usb.h:1972:19: note: declared here
 1972 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
make[3]: *** [/usr/src/linux-5.19.2-1/scripts/Makefile.build:250: /home/Miguel/media_build/v4l/mceusb.o] Error 1
make[3]: *** Se espera a que terminen otras tareas....
make[2]: *** [../../../linux-5.19.2-1/Makefile:1863: /home/Miguel/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-5.19.2-1-obj/x86_64/default'
make[1]: *** [Makefile:53: default] Error 2
make[1]: se sale del directorio '/home/Miguel/media_build/v4l'
make: *** [Makefile:26: all] Error 2

ich777, 2022-08-31:
See this #289 and also more important tbsdtv/media_build#56

crazycat69, 2023-03-22:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-22 by crazycat69