tbsdtv/linux_media #294: Installation of the driver brakes vaapi

lucky79, 2022-09-22:
Hello,

before I installed this driver for my TBS cards, VAAPI was working fine and vainfo gave me list of codecs, which are supported for acceleration. But after installation, vainfo just shows the following:

error: can't connect to X server!
error: failed to initialize display

May I know if thats expected behaviour or what I can do to have both TBS drivers and VAAPI working?

Thank you!
Best regards
Lukas

Debian bullseye - kernel 5.18.0-0.deb11.4-amd64
CPU is Intel i3-12100, GPU Intel UHD graphics 730

lucky79, 2022-09-22:
Ok seems the same problem like #250. But I cannot downgrade the kernel, because the GPU is not supported. So I guess I just need to wait, right?

crazycat69, 2022-09-22:
use [media_build/install.sh](https://github.com/tbsdtv/media_build/blob/extra/install.sh) script for compile/install

closed 2023-03-22 by crazycat69