tbsdtv/linux_media #302: Build is failing on Fedora 37 6.0.15-300.fc37.x86_64

gbirchley, 2022-12-30:
The build is failing on Fedora 37 6.0.15-300.fc37.x86_64

I get this result:

cd Downloads
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make

....

                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  CC [M]  /home/mythtv/Downloads/media_build/v4l/m5mols_capture.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  LD [M]  /home/mythtv/Downloads/media_build/v4l/m5mols.o
  CC [M]  /home/mythtv/Downloads/media_build/v4l/max9271.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  CC [M]  /home/mythtv/Downloads/media_build/v4l/ml86v7667.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
[... 894 lines cut ...]
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
/home/mythtv/Downloads/media_build/v4l/tda1997x.c:2520:10: error: 'struct snd_soc_component_driver' has no member named 'non_legacy_dai_naming'; did you mean 'legacy_dai_naming'?
 2520 |         .non_legacy_dai_naming  = 1,
      |          ^~~~~~~~~~~~~~~~~~~~~
      |          legacy_dai_naming
make[3]: *** [scripts/Makefile.build:249: /home/mythtv/Downloads/media_build/v4l/tda1997x.o] Error 1
make[2]: *** [Makefile:1856: /home/mythtv/Downloads/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/kernels/6.0.15-300.fc37.x86_64'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/mythtv/Downloads/media_build/v4l'
make: *** [Makefile:26: all] Error 2

bluzee, 2022-12-31:
Do you have a tda1997x device that you need a driver for it?  If not disable the driver in config and run make again.  If memory serves the config file is .v4l/config  open that in a text editor and search for 
`CONFIG_VIDEO_TDA1997X=m `
Change that line to 
`# CONFIG_VIDEO_TDA1997X is not set`

gbirchley, 2023-01-01:
Thanks, I don't need the driver so this would be a possible workaround. Hacking the config file in this way pushes the build further on, but it breaks again:

....

   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  CC [M]  /home/mythtv/Downloads/media_build/v4l/pvrusb2-sysfs.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  CC [M]  /home/mythtv/Downloads/media_build/v4l/pvrusb2-debugifc.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  LD [M]  /home/mythtv/Downloads/media_build/v4l/pvrusb2.o
  CC [M]  /home/mythtv/Downloads/media_build/v4l/stk1160-core.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/mythtv/Downloads/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
  CC [M]  /home/mythtv/Downloads/media_build/v4l/stk1160-v4l.o
In file included from ./include/linux/device.h:29,
                 from ./include/linux/input.h:19,
[... 123 lines cut ...]
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/mmdebug.h:5,
                 from /home/mythtv/Downloads/media_build/v4l/config-compat.h:12,
                 from /home/mythtv/Downloads/media_build/v4l/compat.h:10:
./include/linux/compiler.h:246: note: this is the location of the previous definition
  246 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
/home/mythtv/Downloads/media_build/v4l/tm6000-video.c: In function 'tm6000_prepare_isoc':
/home/mythtv/Downloads/media_build/v4l/tm6000-video.c:573:16: error: too many arguments to function 'usb_maxpacket'
  573 |         size = usb_maxpacket(dev->udev, pipe, usb_pipeout(pipe));
      |                ^~~~~~~~~~~~~
In file included from /home/mythtv/Downloads/media_build/v4l/tm6000-video.c:20:
./include/linux/usb.h:1976:19: note: declared here
 1976 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:249: /home/mythtv/Downloads/media_build/v4l/tm6000-video.o] Error 1
make[2]: *** [Makefile:1856: /home/mythtv/Downloads/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/kernels/6.0.15-300.fc37.x86_64'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/mythtv/Downloads/media_build/v4l'
make: *** [Makefile:26: all] Error 2

AndyHarvey99, 2023-01-01:
I've being using the patch to the .config here: https://github.com/tbsdtv/media_build/issues/56#issuecomment-1288681754
It's compiles ok on 6.0.15-300.fc37.x86_64.

bluzee, 2023-01-01:
You have to do the same thing with any of these modules that won't compile. Find the entry for the module in config and disable it in the same manner then run make again. There are usually two or three at least.

gbirchley, 2023-01-02:
Editing the config file as described in the linked thread worked, eventually - although deleting drivers too recklessly initially meant the MCE remote stopped working. Now up and running in any event.

For anyone reading, follow the configuration steps as normal then edit the /media_build/v4l/.config file that allyesconfig produces, removing any broken drivers,then proceed with building.

This isn't a long term solution really, but works fir the time being. Surely the its time for TBS to step up if there are problems maintaining the open source driver build system?

Thanks for your help.

closed 2023-01-02 by gbirchley