tbsdtv/linux_media #303: upstream new kernel

bastard79, 2023-01-29:
is any chance to sync sources with latest linuxtv with support kernel 6.x? or any tutorial how do this by myself?

kafisc1, 2023-02-05:
Indeed! Would be nice to be able to use kernel 6.1.

gpancot, 2023-02-21:
i am testing this version for 6.1.12, it seems to work with my tbs6281:
https://github.com/gpancot/linux_media

kafisc1, 2023-02-28:
Thank's for your feeback.
Not working for me for TBS 6209 on pve-kernel-6.1.10-1-pve.
The error occurs on the step "make":

 CC [M]  /home/tbs_6_1/media_build/v4l/rdacm20.o
/home/tbs_6_1/media_build/v4l/ov7740.c:1217:14: error: initialization of 'void (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
 1217 |  .remove   = ov7740_remove,

gpancot, 2023-03-15:
I corrected the error, retry

crazycat69, 2023-03-22:
Updated, now compatible with 4.14 - 6.2

closed 2023-03-22 by crazycat69

kafisc1, 2023-04-29:
> Updated, now compatible with 4.14 - 6.2
Thanks! Working perfectly with Proxmox Kernel 6.2.11-1-pve.