tbsdtv/linux_media #304: STV091x: array index out of bounds

jmwislez, 2023-02-05:
I'm running Ubuntu 22.04 LTS with the latest open source TBS drivers, and 3 TBS boards: TBS 6910 DVB-S/S2, TBS 6903 DVB-S/S2, and TBS 6281SE DVB-T/T2/C.

I'm getting the following error message in dmesg:
```
[  225.844628] ================================================================================
[  225.844633] UBSAN: array-index-out-of-bounds in /root/tbsdriver/media_build/v4l/stv091x.c:751:23
[  225.844635] index 14 is out of range for type 'u32 [2]'
[  225.844636] CPU: 6 PID: 2314 Comm: tvh:mtimer Tainted: G           OE     5.15.0-58-generic #64-Ubuntu
[  225.844638] Hardware name: Gigabyte Technology Co., Ltd. Z690 UD DDR4/Z690 UD DDR4, BIOS F20b 10/07/2022
[  225.844639] Call Trace:
[  225.844640]  <TASK>
[  225.844641]  show_stack+0x52/0x5c
[  225.844645]  dump_stack_lvl+0x4a/0x63
[  225.844648]  dump_stack+0x10/0x16
[  225.844650]  ubsan_epilogue+0x9/0x49
[  225.844651]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[  225.844653]  read_status+0x1084/0x10c0 [stv091x]
[  225.844656]  dvb_frontend_handle_ioctl+0x36a/0xc90 [dvb_core]
[  225.844662]  ? hrtimer_cancel+0x15/0x40
[  225.844664]  ? futex_wait+0x232/0x260
[  225.844665]  dvb_frontend_do_ioctl+0x89/0x100 [dvb_core]
[  225.844669]  dvb_usercopy+0x55/0x1a0 [dvb_core]
[  225.844671]  ? dvb_frontend_handle_ioctl+0xc90/0xc90 [dvb_core]
[  225.844675]  ? futex_wake+0x7c/0x190
[  225.844676]  ? __fget_files+0x86/0xc0
[  225.844678]  dvb_frontend_ioctl+0x1f/0x40 [dvb_core]
[  225.844681]  __x64_sys_ioctl+0x92/0xd0
[  225.844683]  do_syscall_64+0x59/0xc0
[  225.844685]  ? exit_to_user_mode_prepare+0x37/0xb0
[  225.844686]  ? syscall_exit_to_user_mode+0x27/0x50
[  225.844688]  ? do_syscall_64+0x69/0xc0
[  225.844689]  ? do_syscall_64+0x69/0xc0
[  225.844690]  ? exit_to_user_mode_prepare+0x37/0xb0
[  225.844691]  ? syscall_exit_to_user_mode+0x27/0x50
[  225.844692]  ? __x64_sys_read+0x19/0x20
[  225.844693]  ? do_syscall_64+0x69/0xc0
[  225.844694]  ? __x64_sys_access+0x1c/0x30
[  225.844695]  ? do_syscall_64+0x69/0xc0
[  225.844696]  ? exc_page_fault+0x89/0x170
[  225.844697]  entry_SYSCALL_64_after_hwframe+0x61/0xcb
[  225.844700] RIP: 0033:0x7f3d0018baff
[  225.844702] Code: 00 48 89 44 24 18 31 c0 48 8d 44 24 60 c7 04 24 10 00 00 00 48 89 44 24 08 48 8d 44 24 20 48 89 44 24 10 b8 10 00 00 00 0f 05 <41> 89 c0 3d 00 f0 ff ff 77 1f 48 8b 44 24 18 64 48 2b 04 25 28 00
[  225.844704] RSP: 002b:00007f3cf4c49910 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  225.844706] RAX: ffffffffffffffda RBX: 00007f3cd4008fe0 RCX: 00007f3d0018baff
[  225.844707] RDX: 00007f3cf4c49988 RSI: 0000000080046f45 RDI: 000000000000001d
[  225.844707] RBP: 000056189ab07d10 R08: 0000000000000000 R09: 0000000000000000
[  225.844708] R10: 0000000000000001 R11: 0000000000000246 R12: 0000561898ab9f90
[  225.844708] R13: 00007f3cf4c49a00 R14: 000056189ab08288 R15: 0000000000000001
[  225.844711]  </TASK>
[  225.844711] ================================================================================
```

deeptho, 2023-07-23:
Probably in  the code at the end of the function
static u32 DVBS2_nBCH(enum DVBS2_ModCod ModCod, enum DVBS2_FECType FECType)
in
media/drivers/media/dvb-frontends/stv091x.c 
at line 749 is wrong.

Replace 
```
	if (ModCod >= DVBS2_QPSK_1_4 &&
	    ModCod <= DVBS2_32APSK_9_10 && FECType <= DVBS2_16K)
		return nBCH[FECType][ModCod];
	return 64800;
```
with

```
	if (ModCod >= 0 &&
			ModCod < sizeof(nBCH)/ sizeof(nBCH[0]) &&
			FECType >=0 && FECType < 2)
		return nBCH[ModCod][FECType];
```
Recompile, reinstall, reboot, and report if it works

jmwislez, 2023-07-31:
Thank you for looking at this!

Same error message with the updated code, unfortunately:

```
[   90.265802] ================================================================================
[   90.265817] UBSAN: array-index-out-of-bounds in /root/tbsdriver/media_build/v4l/stv091x.c:751:23
[   90.265827] index 14 is out of range for type 'u32 [2]'
[   90.265833] CPU: 10 PID: 2200 Comm: tvh:mtimer Tainted: G           OE     5.15.0-78-generic #85-Ubuntu
[   90.265841] Hardware name: Gigabyte Technology Co., Ltd. Z690 UD DDR4/Z690 UD DDR4, BIOS F20b 10/07/2022
[   90.265843] Call Trace:
[   90.265847]  <TASK>
[   90.265851]  show_stack+0x52/0x5c
[   90.265862]  dump_stack_lvl+0x4a/0x63
[   90.265872]  dump_stack+0x10/0x16
[   90.265876]  ubsan_epilogue+0x9/0x36
[   90.265880]  __ubsan_handle_out_of_bounds.cold+0x44/0x49
[   90.265886]  ? read_regs.isra.0+0x71/0xa0 [stv091x]
[   90.265894]  read_status+0x1084/0x10c0 [stv091x]
[   90.265903]  dvb_frontend_handle_ioctl+0x36a/0xc90 [dvb_core]
[   90.265930]  dvb_frontend_do_ioctl+0x89/0x100 [dvb_core]
[   90.265949]  dvb_usercopy+0x55/0x1a0 [dvb_core]
[   90.265963]  ? dvb_frontend_handle_ioctl+0xc90/0xc90 [dvb_core]
[   90.265981]  ? fput+0x13/0x20
[   90.265988]  ? exit_to_user_mode_prepare+0x37/0xb0
[   90.265997]  ? syscall_exit_to_user_mode+0x27/0x50
[   90.266004]  ? __fget_files+0x86/0xc0
[   90.266013]  dvb_frontend_ioctl+0x1f/0x40 [dvb_core]
[   90.266031]  __x64_sys_ioctl+0x92/0xd0
[   90.266036]  do_syscall_64+0x59/0xc0
[   90.266044]  entry_SYSCALL_64_after_hwframe+0x61/0xcb
[   90.266053] RIP: 0033:0x7f023bc1aaff
[   90.266060] Code: 00 48 89 44 24 18 31 c0 48 8d 44 24 60 c7 04 24 10 00 00 00 48 89 44 24 08 48 8d 44 24 20 48 89 44 24 10 b8 10 00 00 00 0f 05 <41> 89 c0 3d 00 f0 ff ff 77 1f 48 8b 44 24 18 64 48 2b 04 25 28 00
[   90.266064] RSP: 002b:00007f02306d5910 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[   90.266072] RAX: ffffffffffffffda RBX: 00007f020400f300 RCX: 00007f023bc1aaff
[   90.266076] RDX: 00007f02306d5988 RSI: 0000000080046f45 RDI: 000000000000001e
[   90.266079] RBP: 0000559dc1aca810 R08: 0000000000000000 R09: 0000000000000000
[   90.266082] R10: 0000000000000001 R11: 0000000000000246 R12: 0000559dc01c2830
[   90.266085] R13: 00007f02306d5a00 R14: 0000559dc1acad88 R15: 0000000000000001
[   90.266091]  </TASK>
[   90.266093] ================================================================================
```

crazycat69, 2023-07-31:
Look like old DD bug
https://github.com/DigitalDevices/dddvb/commit/de34e2ebbd82256c2bea292e0b736df4728eea27

crazycat69, 2023-07-31:
Fixed https://github.com/tbsdtv/linux_media/commit/38860714bcd629c160f968e82b61b9bd1f2d9343

closed 2023-11-18 by crazycat69