tbsdtv/linux_media #306: m88rs6060

majortom9, 2023-03-05:
Noticed this modcod doesn't look right and can confirm tuning a known s2x TP of 16 apsk fec 25/36 results in an incorrect  report of 16 apsk fec 8/15.

`diff --git a/drivers/media/dvb-frontends/m88rs6060.c b/drivers/media/dvb-frontends/m88rs6060.c
index d26a0e438ec0..a5d0fbd2ae6e 100644
--- a/drivers/media/dvb-frontends/m88rs6060.c
+++ b/drivers/media/dvb-frontends/m88rs6060.c
@@ -257,8 +257,8 @@ static  struct MT_FE_PLS_INFO mPLSInfoTable[] =
         {0xA3,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_23_36,     TRUE,   FALSE,                 0},
         {0xA4,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk_L,   MtFeCodeRate_2_3,       FALSE,  FALSE,                 0},
         {0xA5,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk_L,   MtFeCodeRate_2_3,       TRUE,   FALSE,                 0},
-        {0xA6,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_8_15,      FALSE,  FALSE,                 0},
-        {0xA7,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_8_15,      TRUE,   FALSE,                 0},
+        {0xA6,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_25_36,     FALSE,  FALSE,                 0},
+        {0xA7,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_25_36,     TRUE,   FALSE,                 0},
         {0xA8,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_13_18,     FALSE,  FALSE,                 0},
         {0xA9,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_13_18,     TRUE,   FALSE,                 0},
         {0xAA,           TRUE,   MtFeType_DvbS2X,        MtFeModMode_16Apsk,     MtFeCodeRate_7_9,       FALSE,  FALSE,                 0},
`

zainkamal95, 2023-04-15:
Is this driver working on coreelec 21?
My device has internal dvb tuner m88rs6060

crazycat69, 2023-04-17:
you need kernel with amlogic dvb subsystem

crazycat69, 2023-04-26:
https://github.com/tbsdtv/linux_media/commit/d94513e97be429847a8bf6ba6bcf74492d3d24d9

closed 2023-04-26 by crazycat69