tbsdtv/linux_media #307: Failed compilation on 6.1.0-6-amd64 kernel (several -Werror=incompatible-pointer-types errors)

Tnix76, 2023-03-12:
I need to compile a driver for a TBS5520 device. I do all the steps but when I execute `make -j4` I get the next errors:

```
make -C /home/victor/Documentos/TBS5520/media_build/v4l 
make[1]: se entra en el directorio '/home/victor/Documentos/TBS5520/media_build/v4l'
scripts/make_makefile.pl
./scripts/make_myconfig.pl
[ ! -f "./config-mycompat.h" ] && echo "/* empty config-mycompat.h */" > "./config-mycompat.h" || true
make -C firmware prep
creating symbolic links...
perl scripts/make_config_compat.pl /lib/modules/6.1.0-6-amd64/source ./.myconfig ./config-compat.h
make[2]: Entering directory '/home/victor/Documentos/TBS5520/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/victor/Documentos/TBS5520/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/victor/Documentos/TBS5520/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/victor/Documentos/TBS5520/media_build/v4l/firmware'
Kernel build directory is /lib/modules/6.1.0-6-amd64/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/victor/Documentos/TBS5520/media_build/linux'
Syncing with dir ../../media
Patches for 6.1.0-6-amd64 already applied.
make[2]: Leaving directory '/home/victor/Documentos/TBS5520/media_build/linux'
make -C /lib/modules/6.1.0-6-amd64/build M=/home/victor/Documentos/TBS5520/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-6.1.0-6-amd64'
  CC [M]  /home/victor/Documentos/TBS5520/media_build/v4l/max2175.o
  CC [M]  /home/victor/Documentos/TBS5520/media_build/v4l/ad5820.o
  CC [M]  /home/victor/Documentos/TBS5520/media_build/v4l/ad9389b.o
  CC [M]  /home/victor/Documentos/TBS5520/media_build/v4l/adp1653.o
In file included from /usr/src/linux-headers-6.1.0-6-common/include/linux/device.h:29,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/input.h:19,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/victor/Documentos/TBS5520/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from /usr/src/linux-headers-6.1.0-6-common/include/asm-generic/bug.h:5,
                 from /usr/src/linux-headers-6.1.0-6-common/arch/x86/include/asm/bug.h:87,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/bug.h:5,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/mmdebug.h:5,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/config-compat.h:12,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/compat.h:10:
/usr/src/linux-headers-6.1.0-6-common/include/linux/compiler.h:238: note: this is the location of the previous definition
  238 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
In file included from /usr/src/linux-headers-6.1.0-6-common/include/linux/device.h:29,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/input.h:19,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/victor/Documentos/TBS5520/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from /usr/src/linux-headers-6.1.0-6-common/include/asm-generic/bug.h:5,
                 from /usr/src/linux-headers-6.1.0-6-common/arch/x86/include/asm/bug.h:87,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/bug.h:5,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/mmdebug.h:5,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/config-compat.h:12,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/compat.h:10:
/usr/src/linux-headers-6.1.0-6-common/include/linux/compiler.h:238: note: this is the location of the previous definition
  238 | #define is_signed_type(type) (((type)(-1)) < (__force type)1)
      | 
In file included from /usr/src/linux-headers-6.1.0-6-common/include/linux/device.h:29,
                 from /usr/src/linux-headers-6.1.0-6-common/include/linux/input.h:19,
                 from /home/victor/Documentos/TBS5520/media_build/v4l/compat.h:47,
                 from <command-line>:
/home/victor/Documentos/TBS5520/media_build/v4l/../linux/include/linux/overflow.h:33: warning: "is_signed_type" redefined
   33 | #define is_signed_type(type)       (((type)(-1)) < (type)1)
      | 
In file included from /usr/src/linux-headers-6.1.0-6-common/include/asm-generic/bug.h:5,
[... 30 lines cut ...]
      |                           ^~~~~~~~~~~~~~
/home/victor/Documentos/TBS5520/media_build/v4l/max2175.c:1435:27: note: (near initialization for 'max2175_driver.remove')
/home/victor/Documentos/TBS5520/media_build/v4l/adp1653.c:542:27: error: initialization of 'void (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
  542 |         .remove         = adp1653_remove,
      |                           ^~~~~~~~~~~~~~
/home/victor/Documentos/TBS5520/media_build/v4l/adp1653.c:542:27: note: (near initialization for 'adp1653_i2c_driver.remove')
/home/victor/Documentos/TBS5520/media_build/v4l/ad9389b.c:1212:19: error: initialization of 'void (*)(struct i2c_client *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
 1212 |         .remove = ad9389b_remove,
      |                   ^~~~~~~~~~~~~~
/home/victor/Documentos/TBS5520/media_build/v4l/ad9389b.c:1212:19: note: (near initialization for 'ad9389b_driver.remove')
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.1.0-6-common/scripts/Makefile.build:255: /home/victor/Documentos/TBS5520/media_build/v4l/ad5820.o] Error 1
make[3]: *** Se espera a que terminen otras tareas....
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.1.0-6-common/scripts/Makefile.build:255: /home/victor/Documentos/TBS5520/media_build/v4l/adp1653.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.1.0-6-common/scripts/Makefile.build:255: /home/victor/Documentos/TBS5520/media_build/v4l/max2175.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.1.0-6-common/scripts/Makefile.build:255: /home/victor/Documentos/TBS5520/media_build/v4l/ad9389b.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.1.0-6-common/Makefile:2030: /home/victor/Documentos/TBS5520/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.0-6-amd64'
make[1]: *** [Makefile:53: default] Error 2
make[1]: se sale del directorio '/home/victor/Documentos/TBS5520/media_build/v4l'
make: *** [Makefile:26: all] Error 2
```

Tnix76, 2023-03-22:
Tested again with the corrections. Compiles and runs fine.

Thanks a lot.

Tnix76, 2023-03-22:
Now it compiles and runs fine.

closed 2023-03-22 by Tnix76

crazycat69, 2023-03-23:
Updated, now compatible with 4.14 - 6.2