tbsdtv/linux_media #310: TBS 5930 (m88rs6060) + Radxa ROCK3 Model A boot fail

yurym, 2023-04-23:
If I first turn on the TBS 5930 and then turn on the Rock 3 board, then an error appears during the board boot. But if I turn on the TBS 5930 when the Rock 3 board is already on, then everything works fine.

```
Apr 23 12:17:07 rock-3a kernel: [   11.345472] mc: loading out-of-tree module taints kernel.
Apr 23 12:17:07 rock-3a kernel: [   11.345635] mc: module verification failed: signature and/or required key missing - tainting kernel
Apr 23 12:17:07 rock-3a kernel: [   11.346878] mc: Linux media interface: v0.10
Apr 23 12:17:07 rock-3a kernel: [   11.383403] videodev: Linux video capture interface: v2.00
Apr 23 12:17:07 rock-3a kernel: [   11.383421] WARNING: You are using an experimental version of the media stack.
Apr 23 12:17:07 rock-3a kernel: [   11.383421] 	As the driver is backported to an older kernel, it doesn't offer
Apr 23 12:17:07 rock-3a kernel: [   11.383421] 	enough quality for its usage in production.
Apr 23 12:17:07 rock-3a kernel: [   11.383421] 	Use it with care.
Apr 23 12:17:07 rock-3a kernel: [   11.383421] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
Apr 23 12:17:07 rock-3a kernel: [   11.383421] 	d94513e97be429847a8bf6ba6bcf74492d3d24d9 media: dvb-frontends/m88rs6060: Fix modcode table.
Apr 23 12:17:07 rock-3a kernel: [   11.389027] WARNING: You are using an experimental version of the media stack.
Apr 23 12:17:07 rock-3a kernel: [   11.389027] 	As the driver is backported to an older kernel, it doesn't offer
Apr 23 12:17:07 rock-3a kernel: [   11.389027] 	enough quality for its usage in production.
Apr 23 12:17:07 rock-3a kernel: [   11.389027] 	Use it with care.
Apr 23 12:17:07 rock-3a kernel: [   11.389027] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
Apr 23 12:17:07 rock-3a kernel: [   11.389027] 	d94513e97be429847a8bf6ba6bcf74492d3d24d9 media: dvb-frontends/m88rs6060: Fix modcode table.
Apr 23 12:17:07 rock-3a kernel: [   11.397860] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in cold state, will try to load a firmware
Apr 23 12:17:07 rock-3a kernel: [   11.411241] dvb-usb: downloading firmware from file 'dvb-usb-id5930.fw'
Apr 23 12:17:07 rock-3a kernel: [   11.411271] tbs5930: start downloading TBS5930 firmware
Apr 23 12:17:07 rock-3a kernel: [   11.413976] cfg80211: Loading compiled-in X.509 certificates for regulatory database
Apr 23 12:17:07 rock-3a kernel: [   11.420365] usb 7-1: USB disconnect, device number 2
Apr 23 12:17:07 rock-3a kernel: [   11.421955] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Apr 23 12:17:07 rock-3a kernel: [   11.432511] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
Apr 23 12:17:07 rock-3a kernel: [   11.434394] hantro-vpu fdea0000.video-codec: Adding to iommu group 0
Apr 23 12:17:07 rock-3a kernel: [   11.434990] hantro-vpu fdea0000.video-codec: registered rockchip,rk3568-vpu-dec as /dev/video0
Apr 23 12:17:07 rock-3a kernel: [   11.435428] hantro-vpu fdee0000.video-codec: Adding to iommu group 1
Apr 23 12:17:07 rock-3a kernel: [   11.436165] hantro-vpu fdee0000.video-codec: registered rockchip,rk3568-vepu-enc as /dev/video1
Apr 23 12:17:07 rock-3a kernel: [   11.474789] Bluetooth: Core ver 2.22
Apr 23 12:17:07 rock-3a kernel: [   11.474931] NET: Registered PF_BLUETOOTH protocol family
Apr 23 12:17:07 rock-3a kernel: [   11.474939] Bluetooth: HCI device and connection manager initialized
Apr 23 12:17:07 rock-3a kernel: [   11.474958] Bluetooth: HCI socket layer initialized
Apr 23 12:17:07 rock-3a kernel: [   11.474966] Bluetooth: L2CAP socket layer initialized
Apr 23 12:17:07 rock-3a kernel: [   11.474987] Bluetooth: SCO socket layer initialized
Apr 23 12:17:07 rock-3a kernel: [   11.513685] Intel(R) Wireless WiFi driver for Linux
Apr 23 12:17:07 rock-3a kernel: [   11.513870] pci 0000:00:00.0: of_irq_parse_pci: failed with rc=-22
Apr 23 12:17:07 rock-3a kernel: [   11.513928] iwlwifi 0000:01:00.0: enabling device (0000 -> 0002)
Apr 23 12:17:07 rock-3a kernel: [   11.525219] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in warm state.
Apr 23 12:17:07 rock-3a kernel: [   11.542744] iwlwifi 0000:01:00.0: api flags index 2 larger than supported by driver
Apr 23 12:17:07 rock-3a kernel: [   11.542821] iwlwifi 0000:01:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
Apr 23 12:17:07 rock-3a kernel: [   11.545826] iwlwifi 0000:01:00.0: loaded firmware version 72.daa05125.0 ty-a0-gf-a0-72.ucode op_mode iwlmvm
Apr 23 12:17:07 rock-3a kernel: [   11.577171] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
Apr 23 12:17:07 rock-3a kernel: [   11.577278] dvbdev: DVB: registering new adapter (TurboSight TBS 5930 DVB-S/S2/S2x)
Apr 23 12:17:07 rock-3a kernel: [   11.577289] tbs5930: read eeprom failed.
Apr 23 12:17:07 rock-3a kernel: [   11.577293] dvb-usb: MAC address reading failed.
Apr 23 12:17:07 rock-3a kernel: [   11.633693] rockchip-pinctrl pinctrl: pin gpio1-9 already requested by vcc-cam; cannot claim for fe410000.i2s
Apr 23 12:17:07 rock-3a kernel: [   11.633710] rockchip-pinctrl pinctrl: pin-41 (fe410000.i2s) status -22
Apr 23 12:17:07 rock-3a kernel: [   11.633717] rockchip-pinctrl pinctrl: could not request pin 41 (gpio1-9) from group i2s1m0-sdi2  on device rockchip-pinctrl
Apr 23 12:17:07 rock-3a kernel: [   11.633724] rockchip-i2s-tdm fe410000.i2s: Error applying setting, reverse things back
Apr 23 12:17:07 rock-3a kernel: [   11.637215] m88rs6060 7-0069: probe failed = 0
Apr 23 12:17:07 rock-3a kernel: [   11.637489] usbcore: registered new interface driver btusb
Apr 23 12:17:07 rock-3a kernel: [   11.656747] Bluetooth: hci0: Device revision is 0
Apr 23 12:17:07 rock-3a kernel: [   11.656771] Bluetooth: hci0: Secure boot is enabled
Apr 23 12:17:07 rock-3a kernel: [   11.656776] Bluetooth: hci0: OTP lock is enabled
Apr 23 12:17:07 rock-3a kernel: [   11.656779] Bluetooth: hci0: API lock is enabled
Apr 23 12:17:07 rock-3a kernel: [   11.656783] Bluetooth: hci0: Debug lock is disabled
Apr 23 12:17:07 rock-3a kernel: [   11.656786] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
Apr 23 12:17:07 rock-3a kernel: [   11.656793] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
Apr 23 12:17:07 rock-3a kernel: [   11.657179] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
Apr 23 12:17:07 rock-3a kernel: [   11.657197] Mem abort info:
Apr 23 12:17:07 rock-3a kernel: [   11.657200]   ESR = 0x0000000096000044
Apr 23 12:17:07 rock-3a kernel: [   11.657204]   EC = 0x25: DABT (current EL), IL = 32 bits
Apr 23 12:17:07 rock-3a kernel: [   11.657209]   SET = 0, FnV = 0
Apr 23 12:17:07 rock-3a kernel: [   11.657212]   EA = 0, S1PTW = 0
Apr 23 12:17:07 rock-3a kernel: [   11.657214]   FSC = 0x04: level 0 translation fault
Apr 23 12:17:07 rock-3a kernel: [   11.657217] Data abort info:
Apr 23 12:17:07 rock-3a kernel: [   11.657220]   ISV = 0, ISS = 0x00000044
Apr 23 12:17:07 rock-3a kernel: [   11.657222]   CM = 0, WnR = 1
[... 41 lines cut ...]
Apr 23 12:17:07 rock-3a kernel: [   11.680880]  load_module+0x1bf8/0x1f88
Apr 23 12:17:07 rock-3a kernel: [   11.681216]  __do_sys_finit_module+0xd4/0x170
Apr 23 12:17:07 rock-3a kernel: [   11.681609]  __arm64_sys_finit_module+0x20/0x30
Apr 23 12:17:07 rock-3a kernel: [   11.682015]  invoke_syscall+0x44/0x108
Apr 23 12:17:07 rock-3a kernel: [   11.682354]  el0_svc_common.constprop.3+0x84/0xf8
Apr 23 12:17:07 rock-3a kernel: [   11.682776]  do_el0_svc+0x2c/0xc8
Apr 23 12:17:07 rock-3a kernel: [   11.683075]  el0_svc+0x20/0x50
Apr 23 12:17:07 rock-3a kernel: [   11.683353]  el0t_64_sync_handler+0x98/0xc0
Apr 23 12:17:07 rock-3a kernel: [   11.683727]  el0t_64_sync+0x170/0x174
Apr 23 12:17:07 rock-3a kernel: [   11.684059] Code: f100405f 540000c3 a9401c26 a97f348c (a9001c06) 
Apr 23 12:17:07 rock-3a kernel: [   11.684601] ---[ end trace 0000000000000000 ]---
Apr 23 12:17:07 rock-3a kernel: [   11.752579] iwlwifi 0000:01:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
Apr 23 12:17:07 rock-3a kernel: [   11.756044] thermal thermal_zone2: failed to read out thermal zone (-61)
Apr 23 12:17:07 rock-3a kernel: [   11.973588] iwlwifi 0000:01:00.0: loaded PNVM version 64acdc51
Apr 23 12:17:07 rock-3a kernel: [   11.989958] iwlwifi 0000:01:00.0: Detected RF GF, rfid=0x10d000
Apr 23 12:17:07 rock-3a kernel: [   12.061790] iwlwifi 0000:01:00.0: base HW address: f0:d4:15:50:de:17
Apr 23 12:17:07 rock-3a kernel: [   12.251910] iwlwifi 0000:01:00.0 wlp1s0: renamed from wlan0
```

```
root@rock-3a:~# uname -r
6.1.11-rk35xx
```

linux_media build commit d94513e97be429847a8bf6ba6bcf74492d3d24d9.

sielicki, 2023-11-05:
Same issue in #329 with Orange Pi 5 Plus under 6.6 next

clear-sky, 2024-03-10:
hi all!
i have the same issue 
https://github.com/tbsdtv/media_build/issues/66

sielicki, 2024-03-10:
Interesting that this is not aarch64 specific in your case, I was thinking this was going to be a weakly ordered specific race condition.

clear-sky, 2024-03-12:
as far as i understood kernel crashes in loading module in cold state and could not read eeprom

crazycat69, 2024-03-12:
module crash on **tbs5930_frontend_attach()**

clear-sky, 2024-03-13:
24cxx eeproms are inherently slow, I think we should give them more time to read their contents

clear-sky, 2024-03-13:
HI dear crazycat69 -thanks for the answer
24cxx eeproms are inherently slow, I think we should give them more time to read their contents

crazycat69, 2024-03-13:
But demod chip also not answer. This some USB host issue, maybe incomplete SoC init when DVB device init started.

clear-sky, 2024-03-13:
thanks for the answer! is there any way to init demod chip after complete boot?
for avoiding from kernel crash

crazycat69, 2024-03-13:
1.) Add the line **blacklist dvb-usb-tbs5930** into the file **/etc/modprobe.d/blacklist.conf**
To prevent that the module is loaded

2.) Add the line /sbin/modprobe dvb-usb-tbs5930 into the file /etc/rc.local
In order to load the module at the end of the boot process.

P.S. [How to Enable /etc/rc.local with Systemd](https://www.linuxbabe.com/linux-server/how-to-enable-etcrc-local-with-systemd). Usual rc.local enabled by default

clear-sky, 2024-03-13:
Hi thanks for this hint
before this hint i thought that problem is related timing for loading module
but now i am sure there is a serious problem in module 
because if we even load module manually it causes kernel crash

clear-sky, 2024-03-14:
Hi Dear crazycat69
do you have the hardware?

clear-sky, 2024-03-14:
hi dear crazycat69!
thanks for your effort!
i see your hardware is different but the same demod chip!
i threw  dvb-usb-tbs5930 in black list thus system boots without problem but
when i load module manually ( modprobe dvb-usb-tbs5930 ) via terminal i give a "KILLED" error message and kernel crashes
what is the problem?

clear-sky, 2024-03-14:
another thing!
I see you have the same problem in this lines :
```
14.03.2024 17:51 tbs5530 read eeprom failed.
14.03.2024 17:51 dvb-usb MAC address reading failed.
14.03.2024 17:51 i2c i2c-1 cxd2878:Can not decete the chip.
14.03.2024 17:51 i2c i2c-1 cxd2878:error attaching frontend.
```
as far as i see you disconnect usb cable of device and connect it again!
it is unacceptable! 
i have installed tbs5930 inside the case and have used internal usb port ( so this is very hard to reconnect usb cabe)

crazycat69, 2024-03-14:
> are you reconnect usb cable manually or by the software? i have installed tbs5930 inside the case and have used internal usb port ( so this is very hard to reconnect usb cabe)

First time device detected in cold state (Cypress FX2 without firmware), next step firmware load and FX2 MCU reset, next step first attempt EEPROM read and demod init - this failed and device disconnected by driver.
Second-time device in warm state, but FX2 firmware load again and  EEPROM read and demod init succeeded.

P.S. This init sequence same since first TBS QBOX device.

crazycat69, 2024-03-14:
Well, look like 5930 driver bug :)

> 14.03.2024 23:02	usb 2-1.1	new high-speed USB device number 6 using ehci-pci
> 14.03.2024 23:02	usb 2-1.1	New USB device found, idVendor=734c, idProduct=5930, bcdDevice= 0.01
> 14.03.2024 23:02	usb 2-1.1	New USB device strings: Mfr=0, Product=0, SerialNumber=0
> 14.03.2024 23:02	dvb-usb	found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in cold state, will try to load a firmware
> 14.03.2024 23:02	dvb-usb	downloading firmware from file 'dvb-usb-id5930.fw'
> 14.03.2024 23:02	tbs5930	start downloading TBS5930 firmware
> 14.03.2024 23:02	dvb-usb	found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in warm state.
> 14.03.2024 23:02	dvb-usb	will pass the complete MPEG2 transport stream to the software demuxer.
> 14.03.2024 23:02	dvbdev	DVB: registering new adapter (TurboSight TBS 5930 DVB-S/S2/S2x)
> 14.03.2024 23:02	tbs5930	read eeprom failed.
> 14.03.2024 23:02	dvb-usb	MAC address reading failed.
> 14.03.2024 23:02	m88rs6060 14-0069	probe failed = 0
> 14.03.2024 23:02	usb 2-1.1	USB disconnect, device number 6
> 14.03.2024 23:02	BUG	kernel NULL pointer dereference, address: 0000000000000008
> 14.03.2024 23:02	#PF	supervisor write access in kernel mode
> 14.03.2024 23:02	#PF	error_code(0x0002) - not-present page
> 14.03.2024 23:02		PGD 0 P4D 0
> 14.03.2024 23:02	Oops	0002 [#1] PREEMPT SMP PTI
>

clear-sky, 2024-03-15:
Hi dear crazycat69!
look at my hardware:
[image]
[image]
as you see there is no place to add additional parts but you said that 5530 has more parts bacause of dvb-t and dvb-c
if possible please put your hardware pictures here! to see differences
thus i am interested how you replace eeprom contents? via usb port or desoldring and resoldering eeprom?
some times eeprom damages! if possible please share this bin files for some people who has such damaged hardware
thanks for notice!

crazycat69, 2024-03-15:
https://github.com/tbsdtv/linux_media/commit/8ac9d8f7e002bd807b9be444f10feae95ca5c12a
https://github.com/tbsdtv/linux_media/commit/58d9cc554e521639b90c2ac4b64a3ba78fee2610

crazycat69, 2024-03-15:
> as you see there is no place to add additional parts but you said that 5530 has more parts bacause of dvb-t and dvb-c
> if possible please put your hardware pictures here! to see differences

[image]
[image]

clear-sky, 2024-03-16:
thank you dear crazycat69!
test new driver and i confirm that problem solved!
thanks again for sharing your hardware image! very very nice hardware but can not be found in the market easily
may be it is a limited product!
but cold not find my last question (eeprom contents)
if this bin files is in windows driver is there any way to extract it in linux? or you have got it from TBS?

sielicki, 2024-03-16:
Thank you, should be able to test this over the weekend.

crazycat69, 2024-03-16:
> my last question (eeprom contents) if this bin files are included in windows driver is there any way to extract it in linux? or you have got it from TBS?

TBS USB devices have small 256 bytes EEPROM for FX2 config and some vendor info like MAC and windows driver mode (for 5530, 5230)
[eep.zip](https://github.com/tbsdtv/linux_media/files/14624708/eep.zip)

clear-sky, 2024-03-17:
thank you! 
crazycat69 you are very kind!
hope victory for you and all people of your great country!

sielicki, 2024-09-02:
Still seeing initialization issues for this device:

```
❯ modinfo dvb-usb-tbs5930
filename:       /run/booted-system/kernel-modules/lib/modules/6.10.7/updates/extra/media/usb/dvb-usb/dvb-usb-tbs5930.ko.xz
license:        GPL
version:        1.0
description:    TurboSight TBS 5930 driver
author:         Davin zhang<Davin@tbsdtv.com>
srcversion:     7505CD78912D725F00A74E3
alias:          usb:v734Cp5930d*dc*dsc*dp*ic*isc*ip*in*
depends:        dvb-usb
name:           dvb_usb_tbs5930
vermagic:       6.10.7 SMP mod_unload aarch64
parm:           debug:set debugging level (1=info 2=xfer (or-able)). (debugging is not enabled) (int)
parm:           adapter_nr:DVB adapter numbers (array of short)

❯ modinfo m88rs6060
filename:       /run/booted-system/kernel-modules/lib/modules/6.10.7/updates/extra/media/dvb-frontends/m88rs6060.ko.xz
firmware:       dvb-demod-m88rs6060.fw
license:        GPL
description:    Montage M88RS6060 driver
author:         Davin zhang <Davin@tbsdtv.com>
alias:          i2c:m88rs6060
depends:
name:           m88rs6060
vermagic:       6.10.7 SMP mod_unload aarch64

❯ uname -a
Linux goldendog 6.10.7 #1-NixOS SMP Thu Aug 29 15:36:13 UTC 2024 aarch64 GNU/Linux

sielicki in 🌐 goldendog in ~
❯ sudo dmesg | tail -n 75
[   61.814551] dvb_usb: loading out-of-tree module taints kernel.
[   61.817824] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in cold state, will try to load a firmware
[   61.820878] dvb-usb: downloading firmware from file 'dvb-usb-id5930.fw'
[   61.820914] tbs5930: start downloading TBS5930 firmware
[   61.840503] usb 1-1: USB disconnect, device number 2
[   61.945960] dvb-usb: found a 'TurboSight TBS 5930 DVB-S/S2/S2x' in warm state.
[   61.946783] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   61.947063] dvbdev: DVB: registering new adapter (TurboSight TBS 5930 DVB-S/S2/S2x)
[   61.947093] tbs5930: read eeprom failed.
[   61.947564] dvb-usb: MAC address reading failed.
[   61.948929] Unable to handle kernel NULL pointer dereference at virtual address 00000000000001af
[   61.949861] Mem abort info:
[   61.950163]   ESR = 0x0000000096000021
[   61.950558]   EC = 0x25: DABT (current EL), IL = 32 bits
[   61.951111]   SET = 0, FnV = 0
[   61.951433]   EA = 0, S1PTW = 0
[   61.951774]   FSC = 0x21: alignment fault
[   61.952198] Data abort info:
[   61.952505]   ISV = 0, ISS = 0x00000021, ISS2 = 0x00000000
[   61.953070]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[   61.953596]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[   61.954148] user pgtable: 4k pages, 48-bit VAs, pgdp=000000017cab5000
[   61.954812] [00000000000001af] pgd=0000000000000000, p4d=0000000000000000
[   61.955521] Internal error: Oops: 0000000096000021 [#1] SMP
[   61.956097] Modules linked in: dvb_usb_tbs5930(O+) dvb_usb(O) dvb_core xt_tcpudp xt_mark nft_compat x_tables nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 cfg80211 rfkill nf_tables libcrc32c nls_iso8859_1 nls_cp437 r8169 snd_soc_simple_card hantro_vpu rtc_hym8563 crct10dif_ce snd_soc_simple_card_utils polyval_ce polyval_generic v4l2_vp9 v4l2_h264 snd_soc_es8328_i2c sm4 v4l2_mem2mem rk805_pwrkey videobuf2_dma_contig snd_soc_es8328 videobuf2_memops pwm_fan videobuf2_v4l2 cmdlinepart gpio_ir_recv snd_soc_simple_amplifier spi_rockchip_sfc phy_rockchip_naneng_combphy rockchip_saradc videodev rockchip_thermal rockchipdrm snd_soc_rockchip_i2s snd_soc_rockchip_i2s_tdm rockchip_dfi dw_mipi_dsi analogix_dp dw_hdmi drm_display_helper videobuf2_common cec mc drm_dma_helper adc_keys uio_pdrv_genirq uio sch_fq_codel tls tap macvlan bridge stp llc fuse nfnetlink dmi_sysfs nvme nvme_core nvme_auth dm_mod dax
[   61.964502] CPU: 3 PID: 1620 Comm: (udev-worker) Tainted: G           O       6.10.7 #1-NixOS
[   61.965381] Hardware name: Orange Pi Orange Pi 5 Plus/Orange Pi 5 Plus, BIOS v0.11.2 07/11/2024
[   61.966266] pstate: a04000c9 (NzCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   61.966984] pc : ida_alloc_range+0xa8/0x528
[   61.967439] lr : media_device_register_entity+0x78/0x228 [mc]
[   61.968077] sp : ffff800085c43400
[   61.968426] x29: ffff800085c43470 x28: ffff0a71bdfa9460 x27: 61c8864680b583eb
[   61.969172] x26: 0000000000000001 x25: ffffab0a07c324b0 x24: 0000000000000001
[   61.969915] x23: 0000000000000000 x22: 0000000000000000 x21: 00000000001fffff
[   61.970666] x20: 000000007fffffff x19: ffff0a71acdd8000 x18: 0000000000000000
[   61.971409] x17: 0000000000000000 x16: ffffab0a06d210d8 x15: 0000000000000000
[   61.972152] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[... 34 lines cut ...]
[   62.031865]  el0t_64_sync+0x190/0x198
[   62.036942] Code: d50343df d503201f 52800001 52800022 (88e17c02)
[   62.042236] ---[ end trace 0000000000000000 ]---
[   62.123539] note: (udev-worker)[1620] exited with irqs disabled
```

That's with media_build and media from master:

```
  media = fetchFromGitHub rec {
    name = repo;
    owner = "tbsdtv";
    repo = "linux_media";
    rev = "06f0fc5ccc29edd1ba0378087fc044c0b63a46d1";
    sha256 = "sha256-HKDWcflkbjCKKkpYTvtcQGIplrT29Uvf9UPo0u66KNY=";
  };

  build = fetchFromGitHub rec {
    name = repo;
    owner = "tbsdtv";
    repo = "media_build";
    rev = "593ed8e0a8d075efddbaafdd8caea3e469751a56";
    sha256 = "sha256-wTvJ3Yq+Kjgt49B+52KmjOBzhwOqZbL8+msVuSOY9W0=";
  };
```