tbsdtv/linux_media #311: kernel 5.16 Compile error

elkizana, 2023-04-28:
`home/idoom/media_build/v4l/ov2740.c:1204:8: error: type defaults to 'int' in declaration of 'DEFINE_SIMPLE_DEV_PM_OPS' [-Werror=implicit-int]
 1204 | static DEFINE_SIMPLE_DEV_PM_OPS(ov2740_pm_ops, ov2740_suspend, ov2740_resume);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/home/idoom/media_build/v4l/ov2740.c:1204:1: warning: parameter names (without types) in function declaration
 1204 | static DEFINE_SIMPLE_DEV_PM_OPS(ov2740_pm_ops, ov2740_suspend, ov2740_resume);
      | ^~~~~~
/home/idoom/media_build/v4l/ov2740.c:1216:23: error: implicit declaration of function 'pm_sleep_ptr' [-Werror=implicit-function-declaration]
 1216 |                 .pm = pm_sleep_ptr(&ov2740_pm_ops),
      |                       ^~~~~~~~~~~~
/home/idoom/media_build/v4l/ov2740.c:1216:37: error: 'ov2740_pm_ops' undeclared here (not in a function); did you mean 'ov2740_pad_ops'?
 1216 |                 .pm = pm_sleep_ptr(&ov2740_pm_ops),
      |                                     ^~~~~~~~~~~~~
      |                                     ov2740_pad_ops
/home/idoom/media_build/v4l/ov2740.c:1204:8: warning: 'DEFINE_SIMPLE_DEV_PM_OPS' declared 'static' but never defined [-Wunused-function]
 1204 | static DEFINE_SIMPLE_DEV_PM_OPS(ov2740_pm_ops, ov2740_suspend, ov2740_resume);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/home/idoom/media_build/v4l/ov2740.c:839:12: warning: 'ov2740_resume' defined but not used [-Wunused-function]
  839 | static int ov2740_resume(struct device *dev)
      |            ^~~~~~~~~~~~~
/home/idoom/media_build/v4l/ov2740.c:825:12: warning: 'ov2740_suspend' defined but not used [-Wunused-function]
  825 | static int ov2740_suspend(struct device *dev)
      |            ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:287: /home/idoom/media_build/v4l/ov2740.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1852: /home/idoom/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.16.20-051620-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/idoom/media_build/v4l'
make: *** [Makefile:26: all] Error 2
`

elkizana, 2023-04-28:
compiled fine with "master"
to close

crazycat69, 2023-04-28:
487a1584559fee4fe1478ddf7e35b304fed54f3b

elkizana, 2023-04-28:
thanks 
how can update the the media_folder ?

crazycat69, 2023-04-28:
another [update](https://github.com/tbsdtv/media_build/commit/281a3f023034f96ebb571838119de60249d1d30d), tested with 5.16.20-051620-generic

elkizana, 2023-04-28:
compiled but unfortunately the graphics are broken, nouveau doesn't seem to work 

[image]

crazycat69, 2023-04-28:
Use install.sh script

elkizana, 2023-04-28:
did it the problem still

crazycat69, 2023-04-28:
Maybe you delete kernel media modules before install ?

elkizana, 2023-04-29:
> Maybe you delete kernel media modules before install ?

suggest me something to do ?

crazycat69, 2023-05-01:
what dvb device used ?

elkizana, 2023-05-02:
TechniSat SkyStar 2 eXpress HD 
I have tested several kernels; kernels 5.10 and below work fine. The ones above 5.10 install but broke the graphics.

elkizana, 2023-05-02:
[dmesg.txt](https://github.com/tbsdtv/linux_media/files/11377410/dmesg.txt)

elkizana, 2023-05-02:
it's working now. I installed it with the the and executed the commands. Strange.

crazycat69, 2023-05-02:
ok
you use old unsupported kernels 5.16, 5.18 instead current ubuntu kernels ?

elkizana, 2023-05-03:
Now i'm using 5.18.0
Thank you for the work you are doing for the community @crazycat69

closed 2023-05-06 by elkizana