tbsdtv/linux_media #313: Compilation fails on Raspberry Pi 6.1.21-v8+

yurym, 2023-06-03:
```
sudo apt install patchutils git libproc-processtable-perl
sudo apt install raspberrypi-kernel-headers

git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media

cd media_build
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j4
```

On last commit I'm getting an error message:

```
/home/yury/media_build/v4l/xilinx-vipp.c:640:3: error: 'struct platform_driver' has no member named 'remove_new'; did you mean 'remove'?
  640 |  .remove_new = xvip_composite_remove,
      |   ^~~~~~~~~~
      |   remove
/home/yury/media_build/v4l/xilinx-vipp.c:640:16: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
  640 |  .remove_new = xvip_composite_remove,
      |                ^~~~~~~~~~~~~~~~~~~~~
/home/yury/media_build/v4l/xilinx-vipp.c:640:16: note: (near initialization for 'xvip_composite_driver.remove')
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:250: /home/yury/media_build/v4l/xilinx-vipp.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/yury/media_build/v4l/xilinx-csi2rxss.c:1087:3: error: 'struct platform_driver' has no member named 'remove_new'; did you mean 'remove'?
 1087 |  .remove_new  = xcsi2rxss_remove,
      |   ^~~~~~~~~~
      |   remove
/home/yury/media_build/v4l/xilinx-csi2rxss.c:1087:17: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
 1087 |  .remove_new  = xcsi2rxss_remove,
      |                 ^~~~~~~~~~~~~~~~
/home/yury/media_build/v4l/xilinx-csi2rxss.c:1087:17: note: (near initialization for 'xcsi2rxss_driver.remove')
/home/yury/media_build/v4l/xilinx-tpg.c:924:3: error: 'struct platform_driver' has no member named 'remove_new'; did you mean 'remove'?
  924 |  .remove_new  = xtpg_remove,
      |   ^~~~~~~~~~
      |   remove
/home/yury/media_build/v4l/xilinx-tpg.c:924:17: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
  924 |  .remove_new  = xtpg_remove,
      |                 ^~~~~~~~~~~
/home/yury/media_build/v4l/xilinx-tpg.c:924:17: note: (near initialization for 'xtpg_driver.remove')
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:250: /home/yury/media_build/v4l/xilinx-csi2rxss.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:250: /home/yury/media_build/v4l/xilinx-tpg.o] Error 1
make[2]: *** [Makefile:2012: /home/yury/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/yury/media_build/v4l'
make: *** [Makefile:26: all] Error 2
```

But on older commit 103e41310c399a11f8535fbb878479ac36ca6069 compiles just fine:

```
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=10 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
git checkout 4fd76794ab87353ed38e0e3291c9610c4876c841
cd ../media
git checkout 103e41310c399a11f8535fbb878479ac36ca6069

cd ../media_build
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j4
```

Casandro, 2023-07-15:
I can confirm this.

crazycat69, 2023-07-15:
media_build updated
https://github.com/tbsdtv/media_build/commit/214eec655f5a1dc4aa5169c0b273c637c6e87d40

Casandro, 2023-07-16:
It still doesn't seem to work:
`/home/casandro/tbs/media_build/v4l/isl7998x.c:1617:12: error: initialization of 'int (*)(struct i2c_client *, const struct i2c_device_id *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
 1617 |  .probe  = isl7998x_probe,` 

Apparently some i2c device drivers are still incompatible. I have tried fixing them, by adding the missing parameter, but then I get stuck at:
`/home/casandro/tbs/media_build/v4l/xilinx-vtc.c: In function 'xvtc_of_get':                                                                                                                                         
/home/casandro/tbs/media_build/v4l/xilinx-vtc.c:257:7: error: implicit declaration of function 'of_property_present'; did you mean 'fwnode_property_present'? [-Werror=implicit-function-declaration]                 257 |  if (!of_property_present(np, "xlnx,vtc"))                                                        
      |       ^~~~~~~~~~~~~~~~~~~                                                                         
      |       fwnode_property_present                    `

crazycat69, 2023-07-16:
media_build updated
https://github.com/tbsdtv/media_build/commit/a8e584b52516a8f00b4a92d20c28cabe57f65740

GuiAlDuS, 2023-08-15:
A similar issue happening to me on Raspbian kernel v 6.1.21-v8+

```
/home/pi/media_build/v4l/max9286.c:1719:12: error: initialization of 'int (*)(struct i2c_client *, const struct i2c_device_id *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
 1719 |  .probe  = max9286_probe,
      |            ^~~~~~~~~~~~~

.................................
In file included from <command-line>:
/home/pi/media_build/v4l/compat.h:1076: warning: "untagged_addr" redefined
 1076 | #define untagged_addr(addr) (addr)
      | 
In file included from ./arch/arm64/include/asm/thread_info.h:17,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/arm64/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/smp.h:110,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/input.h:8,
                 from /home/pi/media_build/v4l/compat.h:47,
                 from <command-line>:
./arch/arm64/include/asm/memory.h:230: note: this is the location of the previous definition
  230 | #define untagged_addr(addr) ({     \
      | 
make[2]: *** [Makefile:2012: /home/pi/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
make: *** [Makefile:26: all] Error 2
```

crazycat69, 2023-08-16:
yet another media_build update :)
https://github.com/tbsdtv/media_build/commit/b50470e0647217cdf95c5f6490d232fa0752e796

GuiAlDuS, 2023-08-16:
Thanks!!
I run it with the new build but got these other errors:

```
./arch/arm64/include/asm/memory.h:230: note: this is the location of the previous definition
  230 | #define untagged_addr(addr) ({     \
      | 
/home/pi/media_build/v4l/xilinx-vtc.c: In function 'xvtc_of_get':
/home/pi/media_build/v4l/xilinx-vtc.c:257:7: error: implicit declaration of function 'of_property_present'; did you mean 'fwnode_property_present'? [-Werror=implicit-function-declaration]
  257 |  if (!of_property_present(np, "xlnx,vtc"))
      |       ^~~~~~~~~~~~~~~~~~~
      |       fwnode_property_present
In file included from <command-line>:
/home/pi/media_build/v4l/compat.h:1076: warning: "untagged_addr" redefined
 1076 | #define untagged_addr(addr) (addr)
      | 
In file included from ./arch/arm64/include/asm/thread_info.h:17,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/arm64/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/smp.h:110,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/input.h:8,
                 from /home/pi/media_build/v4l/compat.h:47,
                 from <command-line>:
./arch/arm64/include/asm/memory.h:230: note: this is the location of the previous definition
  230 | #define untagged_addr(addr) ({     \
      | 
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:250: /home/pi/media_build/v4l/xilinx-vtc.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from <command-line>:
/home/pi/media_build/v4l/compat.h:1076: warning: "untagged_addr" redefined
 1076 | #define untagged_addr(addr) (addr)
      | 
In file included from ./arch/arm64/include/asm/thread_info.h:17,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/arm64/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/smp.h:110,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/input.h:8,
                 from /home/pi/media_build/v4l/compat.h:47,
                 from <command-line>:
./arch/arm64/include/asm/memory.h:230: note: this is the location of the previous definition
  230 | #define untagged_addr(addr) ({     \
      | 
/home/pi/media_build/v4l/video-mux.c:485:3: error: 'struct platform_driver' has no member named 'remove_new'; did you mean 'remove'?
  485 |  .remove_new = video_mux_remove,
      |   ^~~~~~~~~~
      |   remove
/home/pi/media_build/v4l/video-mux.c:485:16: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
  485 |  .remove_new = video_mux_remove,
      |                ^~~~~~~~~~~~~~~~
/home/pi/media_build/v4l/video-mux.c:485:16: note: (near initialization for 'video_mux_driver.remove')
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:250: /home/pi/media_build/v4l/video-mux.o] Error 1
In file included from <command-line>:
/home/pi/media_build/v4l/compat.h:1076: warning: "untagged_addr" redefined
 1076 | #define untagged_addr(addr) (addr)
      | 
In file included from ./arch/arm64/include/asm/thread_info.h:17,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/arm64/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/smp.h:110,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/input.h:8,
                 from /home/pi/media_build/v4l/compat.h:47,
                 from <command-line>:
./arch/arm64/include/asm/memory.h:230: note: this is the location of the previous definition
  230 | #define untagged_addr(addr) ({     \
      | 
make[2]: *** [Makefile:2012: /home/pi/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
make: *** [Makefile:26: all] Error 2
```

crazycat69, 2023-08-18:
Disable VIDEO_XILINX.
https://github.com/tbsdtv/media_build/commit/26ec79bf8fe730bea6fddec21a587398cf71001d

GuiAlDuS, 2023-08-18:
Thank you very much!
Now it got this other: 

```

/home/pi/media_build/v4l/video-mux.c:485:3: error: 'struct platform_driver' has no member named 'remove_new'; did you mean 'remove'?
  485 |  .remove_new = video_mux_remove,
      |   ^~~~~~~~~~
      |   remove
/home/pi/media_build/v4l/video-mux.c:485:16: error: initialization of 'int (*)(struct platform_device *)' from incompatible pointer type 'void (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]
  485 |  .remove_new = video_mux_remove,
      |                ^~~~~~~~~~~~~~~~

```

crazycat69, 2023-08-19:
new update :)
https://github.com/tbsdtv/media_build/commit/f03e9524ca9f55b2b6f54d97bd12114c8cd8912b

GuiAlDuS, 2023-08-19:
It worked! thanks a lot crazycat69!

closed 2023-08-19 by crazycat69