tbsdtv/linux_media #314: TBS5520SE errors : err -2

xfiles2007, 2023-07-11:
Ive tried to install the drivers , they were working on Ubuntu 20 but now have stopped. Not sure what's going on , have tried to remove module and reinstall but hitting this :

[   67.133778] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[   67.282521] usb 1-1: New USB device found, idVendor=734c, idProduct=5521, bcdDevice= 0.01
[   67.282531] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   67.304940] dvb_usb: Unknown symbol rc_register_device (err -2)
[   67.304961] dvb_usb: Unknown symbol rc_free_device (err -2)
[   67.304978] dvb_usb: Unknown symbol rc_allocate_device (err -2)
[   67.305003] dvb_usb: Unknown symbol rc_unregister_device (err -2)

crazycat69, 2023-07-11:
use [install.sh](https://github.com/tbsdtv/media_build/blob/extra/install.sh) for driver configure/compile/install

xfiles2007, 2023-07-11:
I’ve tried that also . 
I’ll try it again.

xfiles2007, 2023-07-12:
Ok tried the install.sh script same issue..

Not sure what else I can do 

[  107.118724] usb 1-1: new high-speed USB device number 2 using ehci-pci
[  107.280552] usb 1-1: New USB device found, idVendor=734c, idProduct=5521, bcdDevice= 0.01
[  107.280558] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  107.316195] mc: Linux media interface: v0.10
[  107.349893] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	face8d83a57d96bfde29c3adfeb6e4a879dd2270 media: pci/tbsmod: Minor fix for 6.4+ kernels.
[  107.353558] dvb_usb: Unknown symbol rc_register_device (err -2)
[  107.353573] dvb_usb: Unknown symbol rc_free_device (err -2)
[  107.353583] dvb_usb: Unknown symbol rc_allocate_device (err -2)
[  107.353599] dvb_usb: Unknown symbol rc_unregister_device (err -2)

xfiles2007, 2023-07-12:
I followed the pdf guide from the tbs site, all ok now. Sorry for the noise

closed 2023-07-12 by xfiles2007