tbsdtv/linux_media #315: Compilation fails on opensuse15.5

hanshorst, 2023-07-19:
No Compilation under opensuse 15.5

Linux lp03 5.14.21-150500.53-default #1 SMP PREEMPT_DYNAMIC Wed May 10 07:56:26 UTC 2023 (b630043) x86_64 x86_64 x86_64 GNU/Linux
gcc version 7.5.0 (SUSE Linux)

Thanks for your help! 
############################################################################

  ```
  CC [M]  /usr/local/src/media_build/v4l/vb2-trace.o
  CC [M]  /usr/local/src/media_build/v4l/videobuf2-dma-contig.o
  CC [M]  /usr/local/src/media_build/v4l/videobuf2-dma-sg.o
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dc_vaddr':
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:103:22: error: storage size of 'map' isn't known
   struct dma_buf_map map;
                      ^~~
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:103:22: warning: unused variable 'map' [-Wunused-variable]
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c: At top level:
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:441:64: warning: 'struct dma_buf_map' declared inside parameter list will not be visible outside of this definition or declaration
 static int vb2_dc_dmabuf_ops_vmap(struct dma_buf *dbuf, struct dma_buf_map *map)
                                                                ^~~~~~~~~~~
  CC [M]  /usr/local/src/media_build/v4l/videobuf2-dvb.o
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dc_dmabuf_ops_vmap':
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:451:2: error: implicit declaration of function 'dma_buf_map_set_vaddr'; did you mean 'iosys_map_set_vaddr'? [-Werror=implicit-function-declaration]
  dma_buf_map_set_vaddr(map, vaddr);
  ^~~~~~~~~~~~~~~~~~~~~
  iosys_map_set_vaddr
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c: At top level:
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:471:10: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .vmap = vb2_dc_dmabuf_ops_vmap,
          ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:471:10: note: (near initialization for 'vb2_dc_dmabuf_ops.vmap')
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dc_unmap_dmabuf':
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:736:9: error: variable 'map' has initializer but incomplete type
  struct dma_buf_map map = DMA_BUF_MAP_INIT_VADDR(buf->vaddr);
         ^~~~~~~~~~~
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:736:27: error: implicit declaration of function 'DMA_BUF_MAP_INIT_VADDR'; did you mean 'IOSYS_MAP_INIT_VADDR'? [-Werror=implicit-function-declaration]
  struct dma_buf_map map = DMA_BUF_MAP_INIT_VADDR(buf->vaddr);
                           ^~~~~~~~~~~~~~~~~~~~~~
                           IOSYS_MAP_INIT_VADDR
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:736:21: error: storage size of 'map' isn't known
  struct dma_buf_map map = DMA_BUF_MAP_INIT_VADDR(buf->vaddr);
                     ^~~
/usr/local/src/media_build/v4l/videobuf2-dma-contig.c:736:21: warning: unused variable 'map' [-Wunused-variable]
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-5.14.21-150500.55.7/scripts/Makefile.build:276: /usr/local/src/media_build/v4l/videobuf2-dma-contig.o] Fehler 1
make[4]: *** Es wird auf noch nicht beendete Prozesse gewartet....
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c: In function 'vb2_dma_sg_vaddr':
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:308:21: error: storage size of 'map' isn't known
  struct dma_buf_map map;
                     ^~~
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:308:21: warning: unused variable 'map' [-Wunused-variable]
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c: At top level:
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:488:68: warning: 'struct dma_buf_map' declared inside parameter list will not be visible outside of this definition or declaration
 static int vb2_dma_sg_dmabuf_ops_vmap(struct dma_buf *dbuf, struct dma_buf_map *map)
                                                                    ^~~~~~~~~~~
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c: In function 'vb2_dma_sg_dmabuf_ops_vmap':
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:492:2: error: implicit declaration of function 'dma_buf_map_set_vaddr'; did you mean 'iosys_map_set_vaddr'? [-Werror=implicit-function-declaration]
  dma_buf_map_set_vaddr(map, buf->vaddr);
  ^~~~~~~~~~~~~~~~~~~~~
  iosys_map_set_vaddr
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c: At top level:
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:512:10: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .vmap = vb2_dma_sg_dmabuf_ops_vmap,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:512:10: note: (near initialization for 'vb2_dma_sg_dmabuf_ops.vmap')
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c: In function 'vb2_dma_sg_unmap_dmabuf':
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:579:9: error: variable 'map' has initializer but incomplete type
  struct dma_buf_map map = DMA_BUF_MAP_INIT_VADDR(buf->vaddr);
         ^~~~~~~~~~~
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:579:27: error: implicit declaration of function 'DMA_BUF_MAP_INIT_VADDR'; did you mean 'IOSYS_MAP_INIT_VADDR'? [-Werror=implicit-function-declaration]
  struct dma_buf_map map = DMA_BUF_MAP_INIT_VADDR(buf->vaddr);
                           ^~~~~~~~~~~~~~~~~~~~~~
                           IOSYS_MAP_INIT_VADDR
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:579:21: error: storage size of 'map' isn't known
  struct dma_buf_map map = DMA_BUF_MAP_INIT_VADDR(buf->vaddr);
                     ^~~
/usr/local/src/media_build/v4l/videobuf2-dma-sg.c:579:21: warning: unused variable 'map' [-Wunused-variable]
cc1: some warnings being treated as errors

```

crazycat69, 2023-07-19:
Look like kernel with backported patches from new kernels.
Try disable **v5.17_iosys.patch** in [media_build/backports/backports.txt](https://github.com/tbsdtv/media_build/blob/extra/backports/backports.txt#L64)

hanshorst, 2023-07-20:
Tip was successful - thanks!
Two changes are required.
Disabled:
**v5.14_bus_void_return.patch
v5.17_iosys.patch**

```
--- ./backports.txt.ori 2023-07-20 09:03:45.540491134 +0200
+++ ./backports.txt     2023-07-20 09:04:40.272494057 +0200
@@ -61,14 +61,14 @@

 [5.17.1023]
 add v5.17_spi.patch
-add v5.17_iosys.patch
+delete v5.17_iosys.patch
 add v5.17_overflow.patch

 [5.15.1023]
 add v5.15_container_of.patch

 [5.14.1023]
-add v5.14_bus_void_return.patch
+delete  v5.14_bus_void_return.patch

 [5.13.1023]
 add v5.13_stk1160.patch

```

closed 2023-07-20 by hanshorst