tbsdtv/linux_media #319: "loadable modules turned off" on Raspberry Pi 6.1.47-v8+ / 6.1.21-v8+

LukeLR, 2023-09-01:
Hi everyone,

after a kernel update, my TBS5580 stopped showing up, so I tried building the modules again. Both updating my old build environment with `git pull` as well as following the installation steps in the wiki, `make allyesconfig` fails with:

```
File not found: /lib/modules/6.1.47-v8+/build/.config at ./scripts/make_kconfig.pl line 33, <IN> line 4.
```

So I copy my `media_build/v4l/.config` there, then it fails with

```
pi@mediapi:~/git/media_build $ make allyesconfig
make -C /home/pi/git/media_build/v4l allyesconfig
make[1]: Entering directory '/home/pi/git/media_build/v4l'
make[2]: Entering directory '/home/pi/git/media_build/linux'
Syncing with dir ../../media
Patches for 6.1.47-v8+ already applied.
make[2]: Leaving directory '/home/pi/git/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/6.1.47-v8+/build /lib/modules/6.1.47-v8+/build 1
Preparing to compile for kernel version 6.1.47
You appear to have loadable modules turned off in your kernel.  You can
not compile the v4l-dvb drivers, as modules, and use them with a kernel
that has modules disabled.

If you want to compile these drivers into your kernel, you should
use 'make kernel-links' to link the source for these drivers into
your kernel tree.  Then configure and compile the kernel.
make[1]: *** [Makefile:376: allyesconfig] Error 255
make[1]: Leaving directory '/home/pi/git/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2
```

Any ideas what "loadable modules turned off" means? I'm running the stock Raspbian updated to the latest version, so I don't understand where this is coming from. Am I missing something? Thanks!

crazycat69, 2023-09-04:
maybe old kernel 6.1.47-v8+ deleted ?

tmn505, 2023-09-04:
Remove `media_build/v4l/.version` or change `SRCDIR` path to new kernel in that file. And don't forget to remove Your `.config`.

LukeLR, 2023-09-09:
I completely wiped and reinstalled raspbian on the Pi 4, the error about the missing .config appears again, but this time, I copied the config over from a Pi 3 which apparently has such a config in `/lib/modules/6.1.21-v7+/build/`. However, now it's failing with

```
make[2]: Entering directory '/usr/lib/modules/6.1.21-v8+/build'
make[2]: *** No rule to make target 'modules'.  Stop.
```

So while I'm not sure if the loadable modules problem is fixed, or if it's just a new error that's occurring first. However, it still doesn't build. Any ideas?

tmn505, 2023-09-10:
It appears as if You don't have any linux-headers installed?

LukeLR, 2023-09-10:
Thanks for the suggeetion. however, raspberry-kernel-headers is installed and up-to-Date, so that's probably not the cause. Any other ideas?

tmn505, 2023-09-10:
Without whole console output at which this occurs it's hard to guess.
Anyway, clone media_build to new dir and run `./build` (You'll probably have to install packages it asks).

LukeLR, 2023-09-11:
Same error message when running ./build from a fresh media_build clone. See the full output attached:

```
raspberrypi:~/git/media_build $ ./build
Checking if the needed tools for Raspbian GNU/Linux 11 (bullseye) are available
Needed package dependencies are met.

************************************************************
* This script will download the latest tarball and build it*
* Assuming that your kernel is compatible with the latest  *
* drivers. If not, you'll need to add some extra backports,*
* ./backports/<kernel> directory.                          *
* It will also update this tree to be sure that all compat *
* bits are there, to avoid compilation failures            *
************************************************************
************************************************************
* All drivers and build system are under GPLv2 License     *
* Firmware files are under the license terms found at:     *
* https://github.com/tbsdtv/media_build/releases/download/latest *
* Please abort in the next 5 secs if you don't agree with  *
* the license                                              *
************************************************************

Not aborted. It means that the licence was agreed. Proceeding...

****************************
Updating the building system
****************************
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
From https://github.com/tbsdtv/media_build
 * branch            master     -> FETCH_HEAD
Already up to date.
make: Entering directory '/home/lukas/git/media_build/linux'
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
--2023-09-11 01:26:35--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/f43f6471-805c-47ad-b55f-14c3825d8a46?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=9bb8d1bae4b5d99733c0644ce7ebc49bc4105fed3490b53585702c164eff3566&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream [following]
--2023-09-11 01:26:36--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/f43f6471-805c-47ad-b55f-14c3825d8a46?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=9bb8d1bae4b5d99733c0644ce7ebc49bc4105fed3490b53585702c164eff3566&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 91 [application/octet-stream]
Saving to: ‘linux-media.tar.bz2.md5.tmp’

linux-media.tar. 100%[==========>]      91  --.-KB/s    in 0s      

2023-09-11 01:26:36 (2.14 MB/s) - ‘linux-media.tar.bz2.md5.tmp’ saved [91/91]

cat: linux-media.tar.bz2.md5: No such file or directory
--2023-09-11 01:26:36--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/bdb19af2-8772-4909-8a95-b33acd0fa789?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=ebe6d28a6df9047faff9217dbf3ff2b79b6487b6324cd2d9f5c902f58389cb02&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2&response-content-type=application%2Foctet-stream [following]
--2023-09-11 01:26:36--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/bdb19af2-8772-4909-8a95-b33acd0fa789?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=ebe6d28a6df9047faff9217dbf3ff2b79b6487b [... cut ...]
make[2]: Leaving directory '/home/lukas/git/media_build/linux'
make -C /lib/modules/6.1.21-v8+/build M=/home/lukas/git/media_build/v4l  modules
make[2]: Entering directory '/usr/lib/modules/6.1.21-v8+/build'
make[2]: *** No rule to make target 'modules'.  Stop.
make[2]: Leaving directory '/usr/lib/modules/6.1.21-v8+/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/lukas/git/media_build/v4l'
make: *** [Makefile:26: all] Error 2
build failed at ./build line 554
```

LukeLR, 2023-09-11:
Same error message when running ./build from a fresh media_build clone. See the full output attached:

<details>
<summary>Output</summary>
```
raspberrypi:~/git/media_build $ ./build
Checking if the needed tools for Raspbian GNU/Linux 11 (bullseye) are available
Needed package dependencies are met.

************************************************************
* This script will download the latest tarball and build it*
* Assuming that your kernel is compatible with the latest  *
* drivers. If not, you'll need to add some extra backports,*
* ./backports/<kernel> directory.                          *
* It will also update this tree to be sure that all compat *
* bits are there, to avoid compilation failures            *
************************************************************
************************************************************
* All drivers and build system are under GPLv2 License     *
* Firmware files are under the license terms found at:     *
* https://github.com/tbsdtv/media_build/releases/download/latest *
* Please abort in the next 5 secs if you don't agree with  *
* the license                                              *
************************************************************

Not aborted. It means that the licence was agreed. Proceeding...

****************************
Updating the building system
****************************
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
From https://github.com/tbsdtv/media_build
 * branch            master     -> FETCH_HEAD
Already up to date.
make: Entering directory '/home/lukas/git/media_build/linux'
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
--2023-09-11 01:26:35--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/f43f6471-805c-47ad-b55f-14c3825d8a46?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=9bb8d1bae4b5d99733c0644ce7ebc49bc4105fed3490b53585702c164eff3566&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream [following]
--2023-09-11 01:26:36--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/f43f6471-805c-47ad-b55f-14c3825d8a46?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=9bb8d1bae4b5d99733c0644ce7ebc49bc4105fed3490b53585702c164eff3566&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 91 [application/octet-stream]
Saving to: ‘linux-media.tar.bz2.md5.tmp’

linux-media.tar. 100%[==========>]      91  --.-KB/s    in 0s      

2023-09-11 01:26:36 (2.14 MB/s) - ‘linux-media.tar.bz2.md5.tmp’ saved [91/91]

cat: linux-media.tar.bz2.md5: No such file or directory
--2023-09-11 01:26:36--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/bdb19af2-8772-4909-8a95-b33acd0fa789?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=ebe6d28a6df9047faff9217dbf3ff2b79b6487b6324cd2d9f5c902f58389cb02&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2&response-content-type=application%2Foctet-stream [following]
--2023-09-11 01:26:36--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/bdb19af2-8772-4909-8a95-b33acd0fa789?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230911%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230911T002636Z&X-Amz-Expires=300&X-Amz-Signature=ebe [... cut ...]
make[2]: Leaving directory '/home/lukas/git/media_build/linux'
make -C /lib/modules/6.1.21-v8+/build M=/home/lukas/git/media_build/v4l  modules
make[2]: Entering directory '/usr/lib/modules/6.1.21-v8+/build'
make[2]: *** No rule to make target 'modules'.  Stop.
make[2]: Leaving directory '/usr/lib/modules/6.1.21-v8+/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/lukas/git/media_build/v4l'
make: *** [Makefile:26: all] Error 2
build failed at ./build line 554
``` 
</details>

LukeLR, 2023-09-11:
Solved it! The issue was that the 32 bit Raspberry Pi OS does not supply kernel headers for the ARMv8 architecture used by Raspberry Pi 4, although being the default option. Switching to 64 bit Raspberry Pi OS fixed the issue, as now the `/lib/modules/$(uname -r)/build` folder exists. Thanks for all your help!

closed 2023-09-11 by LukeLR