tbsdtv/linux_media #321: DVB: Unable to find symbol av201x_attach

AndyHarvey99, 2023-09-14:
Compiled drivers ok for kernel 6.4.15 on Fedora 37. 
At runtime I see the failure:
Sep 14 12:26:14 mythtv kernel:failing symbol_get of non-GPLONLY symbol av201x_attach.
Sep 14 12:26:14 mythtv kernel: DVB: Unable to find symbol av201x_attach()

After a bit of digging I think this is related to GPL-only exports (https://lwn.net/Articles/939842/)

Simply fixed by changing EXPORT_SYMBOL(av201x_attach) to EXPORT_SYMBOL_GPL(av201x_attach);

I'm not sure if this would be a permanent solution.

tmn505, 2023-09-19:
It is the permanent solution https://lore.kernel.org/lkml/20230908092035.3815268-2-gregkh@linuxfoundation.org,
unless module has incompatible license.

crazycat69, 2023-09-24:
https://github.com/tbsdtv/linux_media/commit/4acdd089dfe99cfcd4436f729e133d625212b1e1
https://github.com/tbsdtv/linux_media/commit/bfcdb6fc52a62ee744969cc97017615aa8507c98

AndyHarvey99, 2023-09-29:
Tested successfully on kernel 6.5.5 (Fedora 37)
Thank you.

closed 2023-09-29 by AndyHarvey99