tbsdtv/linux_media #322: Compilation fails: RockPro64 - Dietpi

hartontw, 2023-09-15:
Hello,

I'm following the [guide in the wiki](https://github.com/tbsdtv/linux_media/wiki). 
Probably I'm doing something wrong but I'm not able to determine what.

**Hardware:** `RockPro64`
**Arch:** `aarch64`
**OS:** `DietPi v8.21.1 (Bookworm)`
**Kernel:** `6.1.50-current-rockchip64`

Workaround (fresh install):
```bash
apt install linux-headers-current-rockchip64 patchutils gcc make git
mkdir linux_media
cd linux_media
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j4
```

Errors:
```shell
./include/linux/minmax.h:36:9: error: expected identifier or '(' before '__builtin_choose_expr'
   36 |         __builtin_choose_expr(__safe_cmp(x, y), \
``` 
```shell
./include/linux/minmax.h:33:49: error: expected identifier or '(' before ')' token
   33 |                 __cmp(unique_x, unique_y, op); })
``` 

<details>
  <summary>Log Error</summary>

```shell
make -C /home/drivers/media_build/v4l
make[1]: Entering directory '/home/drivers/media_build/v4l'
make -C firmware prep
creating symbolic links...
make[2]: Entering directory '/home/drivers/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/drivers/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/drivers/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/drivers/media_build/v4l/firmware'
Kernel build directory is /lib/modules/6.1.50-current-rockchip64/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/drivers/media_build/linux'
Syncing with dir ../../media
Patches for 6.1.50-current-rockchip64 already applied.
make[2]: Leaving directory '/home/drivers/media_build/linux'
make -C /lib/modules/6.1.50-current-rockchip64/build M=/home/drivers/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-6.1.50-current-rockchip64'
  CC [M]  /home/drivers/media_build/v4l/rockchip_av1_filmgrain.o
  CC [M]  /home/drivers/media_build/v4l/hantro_h264.o
  CC [M]  /home/drivers/media_build/v4l/hantro_hevc.o
  CC [M]  /home/drivers/media_build/v4l/hantro_mpeg2.o
In file included from <command-line>:
/home/drivers/media_build/v4l/compat.h:1076: warning: "untagged_addr" redefined
 1076 | #define untagged_addr(addr) (addr)
      |
In file included from ./arch/arm64/include/asm/thread_info.h:17,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/arm64/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:78,
[... 159 lines cut ...]
/home/drivers/media_build/v4l/compat.h:1076: warning: "untagged_addr" redefined
 1076 | #define untagged_addr(addr) (addr)
      |
In file included from ./arch/arm64/include/asm/thread_info.h:17,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/arm64/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:78,
                 from ./include/linux/smp.h:110,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/input.h:8,
                 from /home/drivers/media_build/v4l/compat.h:47:
./arch/arm64/include/asm/memory.h:230: note: this is the location of the previous definition
  230 | #define untagged_addr(addr)     ({                                      \
      |
make[2]: *** [Makefile:2012: /home/drivers/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.50-current-rockchip64'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/drivers/media_build/v4l'
make: *** [Makefile:26: all] Error 2           
```
</details>

crazycat69, 2023-09-15:
https://github.com/tbsdtv/media_build/commit/18a3abecbca0cc57bb46b9bfac959d61367e35b3

hartontw, 2023-09-15:
Thanks for the response.

I've repeated the process described with the same result.

I've checked first `v4l/versions.txt`.

hartontw, 2023-09-17:
**Hardware:** `RockPro64`
**Arch:** `aarch64`
**OS:** `Armbian (Bookworm)`
**Kernel:** `6.1.50-current-rockchip64`

Same workaround, same errors.

hartontw, 2023-09-17:
I've found the problem and the solution. Tomorrow I will do a Pull Request with a better solution.

For now this is a quick fix:
```shell
apt install gcc make git patchutils libproc-processtable-perl linux-headers-current-rockchip64
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
``` 

Edit `./media/drivers/media/platform/verisilicon/rockchip_av1_filmgrain.c`  commenting lines `195-198`:
```c
//static inline s32 clamp(s32 value, s32 low, s32 high)
//{
//	return value < low ? low : (value > high ? high : value);
//}
``` 

```shell
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j4
make install
reboot
```

closed 2023-09-17 by hartontw