tbsdtv/linux_media #323: Compilation failure incompatible pointer types

iotic-labs-markwharton, 2023-09-17:
```
/home/mark/3rd/tbs/media_build/v4l/ad5820.c:373:27: error: initialization of 'int (*)(struct i2c_client *, const struct i2c_device_id *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
  373 |         .probe          = ad5820_probe,
```

crazycat69, 2023-09-17:
kernel version ?

iotic-labs-markwharton, 2023-09-19:
$ uname -a
Linux windy 6.2.0-32-generic #32~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug 18 10:40:13 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

iotic-labs-markwharton, 2023-09-19:
commit 0a3fbba551ddc2cf6a74ff5d4b9077be7aad0e36 (HEAD -> latest, origin/latest, origin/HEAD)
Author: CrazyCat <crazycat69@narod.ru>
Date:   Tue Sep 12 01:12:05 2023 +0300

    media: dvb-frontends\stid135: Change LLR calculation.

iotic-labs-markwharton, 2023-09-19:
gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

iotic-labs-markwharton, 2023-09-19:
Hope that's enough info
mark@windy:~/3rd/tbs/media_build
$ git branch -a
* extra
  master
  remotes/origin/HEAD -> origin/extra
  remotes/origin/extra
  remotes/origin/latest
  remotes/origin/master

but on media_build
mark@windy:~/3rd/tbs/media
$ git branch -a
  issue-305
* latest
  remotes/origin/HEAD -> origin/latest
  remotes/origin/latest

iotic-labs-markwharton, 2023-09-19:
are those branches compatible?

crazycat69, 2023-09-19:
media_build/extra branch renamed to latest

iotic-labs-markwharton, 2023-09-20:
Sorry if I'm being dim, but does that mean I'm ok (git-wise) and therefore *shouldn't* be getting these errors?
```
 error: initialization of 'int (*)(struct i2c_client *, const struct i2c_device_id *)' from incompatible pointer type 'int (*)(struct i2c_client *)' [-Werror=incompatible-pointer-types]
  373 |         .probe          = ad5820_probe,
      |                           ^~~~~~~~~~~~
```

iotic-labs-markwharton, 2023-09-20:
switching to `origin/latest` make the compilation errors go away, but that leaves me in detatched HEAD
```
git branch -a
* (HEAD detached at origin/latest)
  extra
  master
  remotes/origin/HEAD -> origin/extra
  remotes/origin/extra
  remotes/origin/latest
  remotes/origin/master
```

iotic-labs-markwharton, 2023-09-20:
I'll close this as the compilation errors have gone away

closed 2023-09-20 by iotic-labs-markwharton