tbsdtv/linux_media #324: Module does not compile on opensuse Tumbleweed Kernel 6.3.5

TBScardOwner, 2023-09-21:
Klone nach 'media_build'...
remote: Enumerating objects: 7340, done.
remote: Counting objects: 100% (1384/1384), done.
remote: Compressing objects: 100% (465/465), done.
remote: Total 7340 (delta 953), reused 1336 (delta 916), pack-reused 5956
Empfange Objekte: 100% (7340/7340), 1.51 MiB | 2.53 MiB/s, fertig.
Löse Unterschiede auf: 100% (5282/5282), fertig.
Klone nach './media'...
remote: Enumerating objects: 86172, done.
remote: Counting objects: 100% (86172/86172), done.
remote: Compressing objects: 100% (81337/81337), done.
remote: Total 86172 (delta 8222), reused 29774 (delta 3934), pack-reused 0
Empfange Objekte: 100% (86172/86172), 238.56 MiB | 2.24 MiB/s, fertig.
Löse Unterschiede auf: 100% (8222/8222), fertig.
Aktualisiere Dateien: 100% (81335/81335), fertig.
make -C linux/ dir DIR="../../media"
make[1]: Entering directory '/root/media_build/linux'
rm -rf drivers include sound mm .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
Searching in ../../media/Makefile for kernel version.
./use_dir.pl ../../media
sync file: include/linux/mmc/sdio_ids.h
sync file: include/uapi/linux/max2175.h
sync file: include/uapi/linux/media-bus-format.h
sync file: include/linux/of_graph.h
sync file: include/uapi/linux/uvcvideo.h
sync file: include/trace/events/pwc.h
sync file: include/uapi/linux/usb/video.h
sync file: include/uapi/linux/ivtv.h
sync file: include/linux/atmel-isc-media.h
sync file: include/uapi/linux/v4l2-subdev.h
sync file: include/trace/events/vb2.h
sync file: include/uapi/linux/xilinx-v4l2-controls.h
sync file: include/uapi/linux/ivtvfb.h
sync file: include/uapi/linux/smiapp.h
sync file: include/uapi/linux/v4l2-common.h
sync file: include/uapi/linux/lirc.h
sync file: include/linux/kconfig.h
sync file: include/uapi/linux/v4l2-mediabus.h
sync file: include/uapi/linux/ccs.h
sync file: include/trace/events/v4l2.h
sync file: include/uapi/linux/cec.h
sync file: include/linux/fixp-arith.h
sync file: include/uapi/linux/v4l2-dv-timings.h
sync file: include/linux/hdmi.h
sync file: include/linux/pci_ids.h
sync file: include/linux/ti_wilink_st.h
sync file: include/linux/dma/xilinx_dma.h
sync file: include/linux/via-core.h
sync file: include/linux/usb/uvc.h
sync file: include/uapi/linux/v4l2-controls.h
sync file: sound/pci/bt87x.c
sync file: include/misc/altera.h
sync file: include/linux/overflow.h
sync file: include/sound/aci.h
sync file: include/uapi/linux/cec-funcs.h
sync file: include/uapi/linux/media.h
sync file: include/uapi/linux/videodev2.h
sync dir: include/uapi/linux/dvb/
sync dir: drivers/staging/media/
sync dir: drivers/misc/altera-stapl/
sync dir: drivers/media/
sync dir: include/dt-bindings/media/
sync dir: include/linux/platform_data/media/
sync dir: include/media/
make[1]: Leaving directory '/root/media_build/linux'
make -C /root/media_build/v4l allyesconfig
make[1]: Entering directory '/root/media_build/v4l'
No version yet, using 6.5.3-1-vanilla
make[2]: Entering directory '/root/media_build/linux'
Syncing with dir ../../media
Applying patches for kernel 6.5.3-1-vanilla
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/ccs.patch
patch -s -f -N -p1 -i ../backports/rc-cec.patch
Patched drivers/media/dvb-core/dvbdev.c
Patched drivers/media/v4l2-core/v4l2-dev.c
Patched drivers/media/rc/rc-main.c
Syncing with dir ../../media
make[2]: Leaving directory '/root/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/6.5.3-1-vanilla/build /lib/modules/6.5.3-1-vanilla/source 1
Preparing to compile for kernel version 6.5.3
WARNING: This is the V4L/DVB backport tree, with experimental drivers
         backported to run on legacy kernels from the development tree at:
                http://git.linuxtv.org/media-tree.git.
         It is generally safe to use it for testing a new driver or
         feature, but its usage on production environments is risky.
         Don't use it in production. You've been warned.
VIDEO_ROCKCHIP_ISP1: Requires at least kernel 9.255.255
VIDEO_XILINX: Requires at least kernel 9.255.255
VIDEO_OV772X: Requires at least kernel 9.255.255
VIDEO_OV9650: Requires at least kernel 9.255.255
VIDEO_TC358746: Requires at least kernel 9.255.255
INTEL_ATOMISP: Requires at least kernel 9.255.255
VIDEO_IPU3_IMGU: Requires at least kernel 9.255.255
Created default (all yes) .config file
./scripts/fix_kconfig.pl
make[1]: Leaving directory '/root/media_build/v4l'
make -C /root/media_build/v4l 
make[1]: Entering directory '/root/media_build/v4l'
./scripts/make_myconfig.pl
scripts/make_makefile.pl
[ ! -f "./config-mycompat.h" ] && echo "/* empty config-mycompat.h */" > "./config-mycompat.h" || true
make -C firmware prep
creating symbolic links...
perl scripts/make_config_compat.pl /lib/modules/6.5.3-1-vanilla/source ./.myconfig ./config-compat.h
make[2]: Entering directory '/root/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/root/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/root/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/root/media_build/v4l/firmware'
Kernel build directory is /lib/modules/6.5.3-1-vanilla/build
make -C ../linux apply_patches
make[2]: Entering directory '/root/media_build/linux'
Syncing with dir ../../media
Patches for 6.5.3-1-vanilla already applied.
make[2]: Leaving directory '/root/media_build/linux'
make -C /lib/modules/6.5.3-1-vanilla/build M=/root/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-6.5.3-1-obj/x86_64/vanilla'
  CC [M]  /root/media_build/v4l/max2175.o
  CC [M]  /root/media_build/v4l/ad5820.o
  CC [M]  /root/media_build/v4l/adp1653.o
  CC [M]  /root/media_build/v4l/adv7170.o
In file included from <command-line>:
/root/media_build/v4l/compat.h: In function 'get_user_pages_longterm':
In file included from <command-line>:
/root/media_build/v4l/compat.h: In function 'get_user_pages_longterm':
/root/media_build/v4l/compat.h:673:16: error: too many arguments to function 'get_user_pages'
  673 |         return get_user_pages(start, nr_pages, gup_flags, pages, vmas);
      |                ^~~~~~~~~~~~~~
In file included from /root/media_build/v4l/compat.h:668:
/usr/src/linux-6.5.3-1-vanilla/include/linux/mm.h:2430:6: note: declared here
 2430 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
/root/media_build/v4l/compat.h:673:16: error: too many arguments to function 'get_user_pages'
  673 |         return get_user_pages(start, nr_pages, gup_flags, pages, vmas);
      |                ^~~~~~~~~~~~~~
In file included from /root/media_build/v4l/compat.h:668:
/usr/src/linux-6.5.3-1-vanilla/include/linux/mm.h:2430:6: note: declared here
 2430 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
In file included from <command-line>:
/root/media_build/v4l/compat.h: In function 'get_user_pages_longterm':
/root/media_build/v4l/compat.h:673:16: error: too many arguments to function 'get_user_pages'
  673 |         return get_user_pages(start, nr_pages, gup_flags, pages, vmas);
      |                ^~~~~~~~~~~~~~
In file included from <command-line>:
/root/media_build/v4l/compat.h: In function 'get_user_pages_longterm':
In file included from /root/media_build/v4l/compat.h:668:
/usr/src/linux-6.5.3-1-vanilla/include/linux/mm.h:2430:6: note: declared here
 2430 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
/root/media_build/v4l/compat.h:673:16: error: too many arguments to function 'get_user_pages'
  673 |         return get_user_pages(start, nr_pages, gup_flags, pages, vmas);
      |                ^~~~~~~~~~~~~~
In file included from /root/media_build/v4l/compat.h:668:
/usr/src/linux-6.5.3-1-vanilla/include/linux/mm.h:2430:6: note: declared here
 2430 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-6.5.3-1-vanilla/scripts/Makefile.build:243: /root/media_build/v4l/adp1653.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [/usr/src/linux-6.5.3-1-vanilla/scripts/Makefile.build:243: /root/media_build/v4l/adv7170.o] Error 1
make[4]: *** [/usr/src/linux-6.5.3-1-vanilla/scripts/Makefile.build:243: /root/media_build/v4l/ad5820.o] Error 1
make[4]: *** [/usr/src/linux-6.5.3-1-vanilla/scripts/Makefile.build:243: /root/media_build/v4l/max2175.o] Error 1
make[3]: *** [/usr/src/linux-6.5.3-1-vanilla/Makefile:2034: /root/media_build/v4l] Error 2
make[2]: *** [../../../linux-6.5.3-1-vanilla/Makefile:234: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-6.5.3-1-obj/x86_64/vanilla'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/media_build/v4l'
make: *** [Makefile:26: all] Error 2

crazycat69, 2023-09-24:
https://github.com/tbsdtv/media_build/commit/d685613366ec38344d84b562e20171c2f77eb531

closed 2023-09-24 by TBScardOwner

TBScardOwner, 2023-09-24:
Thank you. Works fine.