tbsdtv/linux_media #325: TBS6590 device compiling error with kernel 6.5.4

alfrup, 2023-09-24:
I need to compile a driver for a TBS6590 device (kernel 6.5.4). When I execute make -j4 I get the following errors:

make -j4
make -C /root/media_build/v4l 
make[1]: Verzeichnis „/root/media_build/v4l“ wird betreten
No version yet, using 6.5.4-1-default
scripts/make_makefile.pl
Updating/Creating .config
make[2]: Entering directory '/root/media_build/linux'
make[2]: Entering directory '/root/media_build/linux'
Syncing with dir ../../media
Syncing with dir ../../media
Applying patches for kernel 6.5.4-1-default
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
make[3]: Entering directory '/root/media_build/linux'
Unapplying patches
patch -s -f -R -p1 -i ../backports/api_version.patch
make[3]: Leaving directory '/root/media_build/linux'
Applying patches for kernel 6.5.4-1-default
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/saa7146/saa7146_hlp.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/saa7146/saa7146_i2c.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/saa7146/saa7146_video.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/siano/smscoreapi.h.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/tveeprom.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/common/videobuf2/videobuf2-core.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-core/dmxdev.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-core/dvb_ca_en50221.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-core/dvb_demux.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-core/dvb_frontend.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-core/dvb_net.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-core/dvbdev.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/cxd2880/cxd2880_top.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib0070.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib0090.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib3000mb.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib3000mc.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib7000m.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib7000p.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib8000.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dib9000.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dibx000_common.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/drx39xyj/drxj.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/drxk_hard.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/dvb-pll.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/gp8psk-fe.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/lgdt3306a.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/nxt200x.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/nxt6000.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/dvb-frontends/or51211.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/mc/mc-devnode.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/btcx-risc.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-cards.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-driver.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-gpio.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-i2c.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-input.c.rej
patch -s -f -N -p1 -i ../backports/debug.patch
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-risc.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/bttv-vbi.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/dst.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/bt8xx/dvb-bt8xx.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/cx23885/altera-ci.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/cx23885/cx23885.h.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/cx25821/cx25821-alsa.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/cx25821/cx25821-cards.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/cx25821/cx25821-core.c.rej
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/pci/cx25821/cx25821-i2c.c.rej
[... 190 lines cut ...]
      |                ^~~~~~~~~~~~~~
In file included from /root/media_build/v4l/compat.h:668:
/usr/src/linux-6.5.4-1/include/linux/mm.h:2430:6: note: declared here
 2430 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
In file included from <command-line>:
/root/media_build/v4l/compat.h: In function 'get_user_pages_longterm':
/root/media_build/v4l/compat.h:673:16: error: too many arguments to function 'get_user_pages'
  673 |         return get_user_pages(start, nr_pages, gup_flags, pages, vmas);
      |                ^~~~~~~~~~~~~~
In file included from /root/media_build/v4l/compat.h:668:
/usr/src/linux-6.5.4-1/include/linux/mm.h:2430:6: note: declared here
 2430 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-6.5.4-1/scripts/Makefile.build:244: /root/media_build/v4l/ad5820.o] Fehler 1
make[4]: *** Es wird auf noch nicht beendete Prozesse gewartet …
make[4]: *** [/usr/src/linux-6.5.4-1/scripts/Makefile.build:244: /root/media_build/v4l/adp1653.o] Fehler 1
make[4]: *** [/usr/src/linux-6.5.4-1/scripts/Makefile.build:244: /root/media_build/v4l/adv7170.o] Fehler 1
make[4]: *** [/usr/src/linux-6.5.4-1/scripts/Makefile.build:244: /root/media_build/v4l/max2175.o] Fehler 1
make[3]: *** [/usr/src/linux-6.5.4-1/Makefile:2051: /root/media_build/v4l] Fehler 2
make[2]: *** [../../../linux-6.5.4-1/Makefile:234: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-6.5.4-1-obj/x86_64/default'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/root/media_build/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2

crazycat69, 2023-09-24:
Look like old media_build
Try latest [tbsdtv/media_build](https://github.com/tbsdtv/media_build) + [tbsdtv/linux_media](https://github.com/tbsdtv/linux_media)

miguel1963, 2023-09-24:
ok works in kernel 6.5.4 tumbleweed saa 7160 card

closed 2023-11-18 by crazycat69