tbsdtv/linux_media #326: Issue with mxl58x

bolzerrr, 2023-10-09:
[    4.876684] mc: loading out-of-tree module taints kernel.
[    4.876758] mc: module verification failed: signature and/or required key missing - tainting kernel
[    4.877706] mc: Linux media interface: v0.10
[    4.886127] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                bfcdb6fc52a62ee744969cc97017615aa8507c98 media: dvb: symbol fixup for dvb_attach()
[    4.891068] TBSECP3 driver 0000:01:00.0: TurboSight TBS 6909 DVB-S/S2 
[    4.914446] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    5.036285] TBSECP3 driver 0000:01:00.0: MAC address 00:22:ab:91:11:00
[    5.096314] i2c i2c-0: Hydra chip version 2
[    5.103342] i2c i2c-0: loading firmware, please wait...
[    5.131981] ------------[ cut here ]------------
[    5.131985] memcpy: detected field-spanning write (size 248) of single field "&buf[6]" at /home/loot/TBS/media_build/v4l/mxl58x.c:208 (size 242)
[    5.132016] WARNING: CPU: 3 PID: 2182 at /home/loot/TBS/media_build/v4l/mxl58x.c:208 write_fw_segment+0x171/0x252 [mxl58x]
[    5.132032] Modules linked in: mxl58x(OE) tbsecp3(OE+) gx1133(OE) tas2101(OE) dvb_core(OE) videobuf2_vmalloc(OE) videobuf2_memops(OE) videobuf2_common(OE) mc(OE) i2c_mux bochs drm_vram_helper drm_ttm_helper ttm agpgart intel_rapl_msr input_leds intel_rapl_common kvm_intel kvm irqbypass crct10dif_pclmul ghash_clmulni_intel sha512_ssse3 aesni_intel crypto_simd cryptd rapl qemu_fw_cfg evdev button hid_generic usbhid hid efivarfs virtio_scsi virtio_net net_failover failover virtio_blk virtio_console sr_mod cdrom crc32_pclmul xhci_pci xhci_pci_renesas xhci_hcd uhci_hcd ehci_pci ehci_hcd virtio_pci virtio_pci_legacy_dev virtio_pci_modern_dev virtio_ring virtio ahci libahci libata simpledrm drm_shmem_helper drm_kms_helper cfbfillrect syscopyarea cfbimgblt sysfillrect sysimgblt fb_sys_fops cfbcopyarea drm drm_panel_orientation_quirks loop ext4 crc32c_generic crc32c_intel crc16 mbcache jbd2 usb_storage usbcore usb_common sd_mod t10_pi crc64_rocksoft crc64 scsi_mod scsi_common
[    5.132119] CPU: 3 PID: 2182 Comm: modprobe Tainted: G           OE      6.1.55-1-lts #2-Alpine
[    5.132123] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 3.20230228-4 06/06/2023
[    5.132125] RIP: 0010:write_fw_segment+0x171/0x252 [mxl58x]
[    5.132134] Code: 3d 05 47 00 00 00 75 29 b9 f2 00 00 00 48 c7 c2 18 02 be c0 44 89 ee 48 c7 c7 58 02 be c0 c6 05 e6 46 00 00 01 e8 b1 2e 4c cb <0f> 0b 48 8b 44 24 18 48 89 df 48 89 c6 44 89 e9 ba 01 00 00 00 f3
[    5.132137] RSP: 0018:ffffa72a8048b778 EFLAGS: 00010246
[    5.132140] RAX: 0000000000000000 RBX: ffff922b06505476 RCX: 0000000000000000
[    5.132142] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[    5.132144] RBP: ffffa72a8050d018 R08: 0000000000000000 R09: 0000000000000000
[    5.132145] R10: 0000000000000000 R11: 0000000000000000 R12: ffff922b02bbf000
[    5.132147] R13: 00000000000000f8 R14: 0000000000008524 R15: 0000000000000000
[    5.132149] FS:  00007fb16fb08b08(0000) GS:ffff922e6fd80000(0000) knlGS:0000000000000000
[    5.132152] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    5.132154] CR2: 00007fdeab4e53f2 CR3: 0000000103128000 CR4: 0000000000750ee0
[    5.132160] PKRU: 55555554
[    5.132162] Call Trace:
[    5.132165]  <TASK>
[    5.132167]  ? write_fw_segment+0x171/0x252 [mxl58x]
[    5.132174]  ? __warn+0x79/0xc0
[    5.132181]  ? write_fw_segment+0x171/0x252 [mxl58x]
[    5.132188]  ? report_bug+0xf4/0x180
[    5.132194]  ? handle_bug+0x49/0xa0
[    5.132199]  ? exc_invalid_op+0x13/0x70
[    5.132203]  ? asm_exc_invalid_op+0x16/0x20
[    5.132209]  ? write_fw_segment+0x171/0x252 [mxl58x]
[    5.132216]  ? write_fw_segment+0x171/0x252 [mxl58x]
[    5.132223]  ? _raw_spin_unlock_irqrestore+0x20/0x50
[    5.132229]  mxl58x_attach.cold+0x314/0x1469 [mxl58x]
[    5.132238]  tbsecp3_frontend_attach.isra.0+0xa86/0x3780 [tbsecp3]
[    5.132254]  tbsecp3_dvb_init+0x17e/0x210 [tbsecp3]
[    5.132265]  tbsecp3_probe.cold+0x262/0x2e0 [tbsecp3]
[    5.132276]  local_pci_probe+0x3e/0x80
[    5.132282]  pci_device_probe+0xbd/0x250
[    5.132286]  ? sysfs_do_create_link_sd+0x6e/0xf0
[    5.132290]  really_probe+0xdb/0x390
[    5.132294]  ? pm_runtime_barrier+0x50/0x90
[    5.132301]  __driver_probe_device+0x78/0x130
[    5.132304]  driver_probe_device+0x1f/0xa0
[    5.132306]  __driver_attach+0xe6/0x1e0
[    5.132309]  ? __device_attach_driver+0x130/0x130
[    5.132311]  bus_for_each_dev+0x84/0xd0
[    5.132316]  bus_add_driver+0x1b9/0x210
[    5.132321]  driver_register+0x89/0xf0
[    5.132324]  ? 0xffffffffc0c67000
[    5.132327]  do_one_initcall+0x56/0x240
[    5.132332]  do_init_module+0x4a/0x200
[    5.132338]  __do_sys_finit_module+0xb8/0x140
[    5.132344]  do_syscall_64+0x59/0x90
[    5.132350]  entry_SYSCALL_64_after_hwframe+0x64/0xce
[    5.132354] RIP: 0033:0x7fb16faa007a
[    5.132358] Code: 4c 89 4c 24 48 4c 8b 4c 24 60 48 89 54 24 10 48 8b 7e 08 48 89 74 24 18 48 8b 51 18 c7 44 24 08 28 00 00 00 48 8b 76 10 0f 05 <48> 89 c7 e8 04 5f fe ff 48 83 c4 58 c3 41 57 41 56 49 89 d6 41 55
[    5.132360] RSP: 002b:00007ffc955b3c40 EFLAGS: 00000202 ORIG_RAX: 0000000000000139
[    5.132363] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007fb16faa007a
[    5.132365] RDX: 0000000000000000 RSI: 0000564fd91f3699 RDI: 000000000000000b
[    5.132366] RBP: 00007fb16fa66eb0 R08: 0000000000000000 R09: 0000000000000080
[    5.132368] R10: 0000000000000000 R11: 0000000000000202 R12: 0000000000060000
[    5.132370] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[    5.132373]  </TASK>
[    5.132374] ---[ end trace 0000000000000000 ]---

bolzerrr, 2023-10-23:
Workaround:

sed -i -r 's/(^CONFIG.*MXL.*=)./\1n/g' v4l/.config

crazycat69, 2023-10-23:
https://github.com/tbsdtv/linux_media/commit/acd9d20728471c705c49d25f52da44d62bb45670

bolzerrr, 2023-10-24:
Just made a fresh build, error is gone. 
Thank you!

closed 2023-10-24 by bolzerrr

ikosk, 2023-10-31:
After the last change, on the contrary, this error appears for me

$ sudo dmesg | grep frontend
acd9d20728471c705c49d25f52da44d62bb45670 media: dvb-frontends/mxl58x: Fix memcpy warning.
[   21.752945] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6903x (Dual DVB-S/S2/S2X))...
[   21.874346] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6903x (Dual DVB-S/S2/S2X))...

$ uname -a
Linux Desktop-PC 6.2.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Oct  6 10:23:26 UTC 2 x86_64 x86_64 x86_64 GNU/Linux