tbsdtv/linux_media #327: Build fails on Debian 12

ritzmann, 2023-10-22:
Latest media and media_build checkouts. Running `build` from media_build:
```
./build                                                                                                     
Checking if the needed tools for Debian GNU/Linux 12 (bookworm) are available
Needed package dependencies are met.

************************************************************
* This script will download the latest tarball and build it*
* Assuming that your kernel is compatible with the latest  *
* drivers. If not, you'll need to add some extra backports,*
* ./backports/<kernel> directory.                          *
* It will also update this tree to be sure that all compat *
* bits are there, to avoid compilation failures            *
************************************************************
************************************************************
* All drivers and build system are under GPLv2 License     *
* Firmware files are under the license terms found at:     *
* https://github.com/tbsdtv/media_build/releases/download/latest *
* Please abort in the next 5 secs if you don't agree with  *
* the license                                              *
************************************************************

Not aborted. It means that the licence was agreed. Proceeding...

****************************
Updating the building system
****************************
Von https://github.com/tbsdtv/media_build
 * branch            master     -> FETCH_HEAD
Bereits aktuell.
make: Verzeichnis „/home/user/src/tbsdtv/media_build/linux“ wird betreten
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
--2023-10-23 02:31:14--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5

...

******************
* Start building *
******************
make -C /home/user/src/tbsdtv/media_build/v4l allyesconfig
make[1]: Verzeichnis „/home/user/src/tbsdtv/media_build/v4l“ wird betreten
No version yet, using 6.1.0-13-amd64
make[2]: Entering directory '/home/user/src/tbsdtv/media_build/linux'
Applying patches for kernel 6.1.0-13-amd64
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/ccs.patch
patch -s -f -N -p1 -i ../backports/rc-cec.patch
patch -s -f -N -p1 -i ../backports/v6.4_pin_user_pages.patch
patch -s -f -N -p1 -i ../backports/v6.3_class_create.patch
patch -s -f -N -p1 -i ../backports/v6.3_check_endpoint.patch
patch -s -f -N -p1 -i ../backports/v6.2_class.patch
patch -s -f -N -p1 -i ../backports/v6.2_remove.patch
patch -s -f -N -p1 -i ../backports/v6.2_probe.patch
patch -s -f -N -p1 -i ../backports/v6.2_v4l2-core.patch
patch -s -f -N -p1 -i ../backports/v6.1_class.patch
patch -s -f -N -p1 -i ../backports/v6.1_i2c_probe.patch
patch -s -f -N -p1 -i ../backports/v6.1_videobuf2.patch
patch -s -f -N -p1 -i ../backports/v6.1_timer.patch
patch -s -f -N -p1 -i ../backports/v6.1_ipu3.patch
Patched drivers/media/dvb-core/dvbdev.c
Patched drivers/media/v4l2-core/v4l2-dev.c
Patched drivers/media/rc/rc-main.c
make[2]: Leaving directory '/home/user/src/tbsdtv/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/6.1.0-13-amd64/build /lib/modules/6.1.0-13-amd64/source 1
Preparing to compile for kernel version 6.1.0

***WARNING:*** You do not have the full kernel sources installed.
[... 148 lines cut ...]
make[3]: *** [/usr/src/linux-headers-6.1.0-13-common/scripts/Makefile.build:255: /home/user/src/tbsdtv/media_build/v4l/tbsmod.o] Fehler 1
make[3]: *** Es wird auf noch nicht beendete Prozesse gewartet....
/home/user/src/tbsdtv/media_build/v4l/tbsci.c: In function 'tbsci_ioctl':
/home/user/src/tbsdtv/media_build/v4l/tbsci.c:249:13: warning: unused variable 'clk_freq' [-Wunused-variable]
  249 |         u32 clk_freq;
      |             ^~~~~~~~
/home/user/src/tbsdtv/media_build/v4l/tbsci.c:255:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  255 |                 if (props.num == 1)
      |                    ^
/home/user/src/tbsdtv/media_build/v4l/tbsci.c:303:9: note: here
  303 |         case FE_ECP3FW_READ:
      |         ^~~~
/home/user/src/tbsdtv/media_build/v4l/tbsci.c:254:17: warning: ignoring return value of 'copy_from_user' declared with attribute 'warn_unused_result' [-Wunused-result]
  254 |                 copy_from_user(&props , (const char*)arg, sizeof(struct dtv_properties ));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/src/tbsdtv/media_build/v4l/tbsci.c:257:25: warning: ignoring return value of 'copy_from_user' declared with attribute 'warn_unused_result' [-Wunused-result]
  257 |                         copy_from_user(&prop , (const char*)props.props, sizeof(struct dtv_property ));
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-headers-6.1.0-13-common/Makefile:2039: /home/user/src/tbsdtv/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.1.0-13-amd64'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/home/user/src/tbsdtv/media_build/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2
build failed at ./build line 554
```

crazycat69, 2023-10-23:
use latest branch [tbsdtv/media_build](https://github.com/tbsdtv/media_build)

ritzmann, 2023-10-25:
Thanks, must have reset that accidentally when cleaning up the workspace. Failing even sooner now however:
```
$ git pull
Already up to date.
$ git status                                                                                                  
On branch latest
Your branch is up to date with 'origin/latest'.

nothing to commit, working tree clean
$ ./build
Checking if the needed tools for Debian GNU/Linux 12 (bookworm) are available
Needed package dependencies are met.

************************************************************
* This script will download the latest tarball and build it*
* Assuming that your kernel is compatible with the latest  *
* drivers. If not, you'll need to add some extra backports,*
* ./backports/<kernel> directory.                          *
* It will also update this tree to be sure that all compat *
* bits are there, to avoid compilation failures            *
************************************************************
************************************************************
* All drivers and build system are under GPLv2 License     *
* Firmware files are under the license terms found at:     *
* https://github.com/tbsdtv/media_build/releases/download/latest *
* Please abort in the next 5 secs if you don't agree with  *
* the license                                              *
************************************************************

Not aborted. It means that the licence was agreed. Proceeding...

****************************
Updating the building system
****************************
From https://github.com/tbsdtv/media_build
 * branch            master     -> FETCH_HEAD
dropping 556d2d2ac981257d3539ad9ec0ff2104e8d8266d Disable SND_BT87X for <5.11 -- patch contents already upstream
Auto-merging backports/backports.txt
CONFLICT (content): Merge conflict in backports/backports.txt
error: could not apply 4fd7679... Update backport patches for < 6.1
hint: Resolve all conflicts manually, mark them as resolved with
hint: "git add/rm <conflicted_files>", then run "git rebase --continue".
hint: You can instead skip this commit: run "git rebase --skip".
hint: To abort and get back to the state before "git rebase", run "git rebase --abort".
Could not apply 4fd7679... Update backport patches for < 6.1
make: Entering directory '/home/user/src/tbsdtv/media_build/linux'
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
--2023-10-25 20:12:58--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/89718596-4a50-4dbd-b984-f27e54da1ba8?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231025%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231025T171258Z&X-Amz-Expires=300&X-Amz-Signature=8e7f2d5640460117d85e6768fb3e5247038fc70b27de8375b834b5eb75555cca&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream [following]
--2023-10-25 20:12:58--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/89718596-4a50-4dbd-b984-f27e54da1ba8?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231025%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231025T171258Z&X-Amz-Expires=300&X-Amz-Signature=8e7f2d5640460117d85e6768fb3e5247038fc70b27de8375b834b5eb75555cca&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 91 [application/octet-stream]
Saving to: 'linux-media.tar.bz2.md5.tmp'

linux-media.tar.bz2.md5.tmp                100%[=======================================================================================>]      91  --.-KB/s    in 0s      

2023-10-25 20:12:58 (2.15 MB/s) - 'linux-media.tar.bz2.md5.tmp' saved [91/91]

cat: linux-media.tar.bz2.md5: No such file or directory
--2023-10-25 20:12:58--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/7414706d-935f-447e-b3d2-f6600546bdce?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231025%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231025T171259Z&X-Amz-Expires=300&X-Amz-Signature=dce5c79c4798a09076a4f3a0368a483530b3c90428bb77eaade1549307620140&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2&response-content-type=application%2Foctet-stream [following]
[... cut ...]
No version yet, using 6.1.0-13-amd64
make[2]: Entering directory '/home/user/src/tbsdtv/media_build/linux'
Can't process line 49
Version 6.1.0-13-amd64 not supported
/bin/sh: 7: exit: Illegal number: -1
make[2]: *** [Makefile:135: apply_patches] Error 2
make[2]: Leaving directory '/home/user/src/tbsdtv/media_build/linux'
make[1]: *** [Makefile:375: allyesconfig] Error 2
make[1]: Leaving directory '/home/user/src/tbsdtv/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2
can't select all drivers at ./build line 552
```

crazycat69, 2023-10-25:
try clear clone media_build

ritzmann, 2023-10-25:
Same failure, deleted and cloned both media_build and linux_media from scratch.

ritzmann, 2023-10-26:
```
$ dpkg --status linux-headers-amd64                                                                                                                   
Package: linux-headers-amd64
Status: install ok installed
```

I have been building this software on Debian successfully for a few years already. Mostly been using Debian 11 but when this started happening around two months ago, I upgraded to Debian 12, assuming that the latest stable version would work best. But that has been giving me similar compilation errors. I am surprised that this would suddenly stop right at the beginning of the process already. Just tried again but it fails consistently with `Version 6.1.0-13-amd64 not supported`.

crazycat69, 2023-10-26:
Just test Debian 12 with same kernel in VirtualBox. Build success

ritzmann, 2023-10-26:
I found the cause of the Version not supported error:
```
****************************
Updating the building system
****************************
From https://github.com/tbsdtv/media_build
 * branch            master     -> FETCH_HEAD
dropping 556d2d2ac981257d3539ad9ec0ff2104e8d8266d Disable SND_BT87X for <5.11 -- patch contents already upstream
Auto-merging backports/backports.txt
CONFLICT (content): Merge conflict in backports/backports.txt
error: could not apply 4fd7679... Update backport patches for < 6.1
hint: Resolve all conflicts manually, mark them as resolved with
hint: "git add/rm <conflicted_files>", then run "git rebase --continue".
hint: You can instead skip this commit: run "git rebase --skip".
hint: To abort and get back to the state before "git rebase", run "git rebase --abort".
Could not apply 4fd7679... Update backport patches for < 6.1
```

That leaves the workspace in a state with a rebase in progress and the backports/backports.txt file contains a merge conflict that causes the script linux/patches_for_kernel.pl to fail. Why you cannot reproduce that is beyond me.

crazycat69, 2023-10-26:
Just make fresh clone media_build and run build script:

```
git clone https://github.com/tbsdtv/media_build
cd media_build
./build

```

ritzmann, 2023-10-27:
```
$ rm -rf media_build/                                                                                                         
$ git clone https://github.com/tbsdtv/media_build                                                                                      
Klone nach 'media_build'...
remote: Enumerating objects: 7355, done.
remote: Counting objects: 100% (1399/1399), done.
remote: Compressing objects: 100% (474/474), done.
remote: Total 7355 (delta 965), reused 1346 (delta 922), pack-reused 5956
Empfange Objekte: 100% (7355/7355), 1.52 MiB | 1.24 MiB/s, fertig.
Löse Unterschiede auf: 100% (5294/5294), fertig.
$ cd media_build                                                                                                                       
$ ./build                                                                                                 
Checking if the needed tools for Debian GNU/Linux 12 (bookworm) are available
Needed package dependencies are met.

************************************************************
* This script will download the latest tarball and build it*
* Assuming that your kernel is compatible with the latest  *
* drivers. If not, you'll need to add some extra backports,*
* ./backports/<kernel> directory.                          *
* It will also update this tree to be sure that all compat *
* bits are there, to avoid compilation failures            *
************************************************************
************************************************************
* All drivers and build system are under GPLv2 License     *
* Firmware files are under the license terms found at:     *
* https://github.com/tbsdtv/media_build/releases/download/latest *
* Please abort in the next 5 secs if you don't agree with  *
* the license                                              *
************************************************************

Not aborted. It means that the licence was agreed. Proceeding...

****************************
Updating the building system
****************************
Von https://github.com/tbsdtv/media_build
 * branch            master     -> FETCH_HEAD
Weglassen von 556d2d2ac981257d3539ad9ec0ff2104e8d8266d Disable SND_BT87X for <5.11 -- Patch-Inhalte sind bereits im Upstream-Branch
automatischer Merge von backports/backports.txt
KONFLIKT (Inhalt): Merge-Konflikt in backports/backports.txt
Fehler: Konnte 4fd7679... (Update backport patches for < 6.1) nicht anwenden
Hinweis: Resolve all conflicts manually, mark them as resolved with
Hinweis: "git add/rm <conflicted_files>", then run "git rebase --continue".
Hinweis: You can instead skip this commit: run "git rebase --skip".
Hinweis: To abort and get back to the state before "git rebase", run "git rebase --abort".
Konnte 4fd7679... (Update backport patches for < 6.1) nicht anwenden
make: Verzeichnis „/home/user/src/tbsdtv/media_build/linux“ wird betreten
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
--2023-10-27 23:18:37--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5
Auflösen des Hostnamens github.com (github.com)… 140.82.121.3
Verbindungsaufbau zu github.com (github.com)|140.82.121.3|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/6b9adf2a-2557-452a-b759-ae4e8654d50c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231027%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231027T201838Z&X-Amz-Expires=300&X-Amz-Signature=eeef0d098d26e2ed58f001e705957eb4eeff2c981b6f7b8dd5ffcd1dd13748d2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream [folgend]
--2023-10-27 23:18:38--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/6b9adf2a-2557-452a-b759-ae4e8654d50c?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231027%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231027T201838Z&X-Amz-Expires=300&X-Amz-Signature=eeef0d098d26e2ed58f001e705957eb4eeff2c981b6f7b8dd5ffcd1dd13748d2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream
Auflösen des Hostnamens objects.githubusercontent.com (objects.githubusercontent.com)… 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Verbindungsaufbau zu objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 91 [application/octet-stream]
Wird in »linux-media.tar.bz2.md5.tmp« gespeichert.

linux-media.tar.bz2.md5.tmp                100%[=======================================================================================>]      91  --.-KB/s    in 0s      

2023-10-27 23:18:38 (3,01 MB/s) - »linux-media.tar.bz2.md5.tmp« gespeichert [91/91]

cat: linux-media.tar.bz2.md5: Datei oder Verzeichnis nicht gefunden
--2023-10-27 23:18:38--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2
Auflösen des Hostnamens github.com (github.com)… 140.82.121.3
Verbindungsaufbau zu github.com (github.com)|140.82.121.3|:443 … verbunden.
[... cut ...]
make[2]: Entering directory '/home/user/src/tbsdtv/media_build/linux'
Can't process line 49
Version 6.1.0-13-amd64 not supported
/bin/sh: 7: exit: Illegal number: -1
make[2]: *** [Makefile:135: apply_patches] Error 2
make[2]: Leaving directory '/home/user/src/tbsdtv/media_build/linux'
make[1]: *** [Makefile:375: allyesconfig] Fehler 2
make[1]: Verzeichnis „/home/user/src/tbsdtv/media_build/v4l“ wird verlassen
make: *** [Makefile:26: allyesconfig] Fehler 2
can't select all drivers at ./build line 552
```

ritzmann, 2023-10-27:
OK, I think I found the reason why it is not compiling:
https://github.com/tbsdtv/media_build/blob/latest/build#L526
When I remove the `master` parameter from there, the build script is getting beyond the above failure:
```
****************************
Updating the building system
****************************
From https://github.com/tbsdtv/media_build
 * branch            HEAD       -> FETCH_HEAD
Current branch latest is up to date.
make: Entering directory '/home/user/src/tbsdtv/media_build/linux'
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
...
```

Patches apply cleanly:
```
...
******************
* Start building *
******************
make -C /home/user/src/tbsdtv/media_build/v4l allyesconfig
make[1]: Entering directory '/home/user/src/tbsdtv/media_build/v4l'
No version yet, using 6.1.0-13-amd64
make[2]: Entering directory '/home/user/src/tbsdtv/media_build/linux'
Applying patches for kernel 6.1.0-13-amd64
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/ccs.patch
patch -s -f -N -p1 -i ../backports/rc-cec.patch
patch -s -f -N -p1 -i ../backports/v6.4_pin_user_pages.patch
patch -s -f -N -p1 -i ../backports/v6.3_class_create.patch
patch -s -f -N -p1 -i ../backports/v6.3_check_endpoint.patch
patch -s -f -N -p1 -i ../backports/v6.2_class.patch
patch -s -f -N -p1 -i ../backports/v6.2_remove.patch
patch -s -f -N -p1 -i ../backports/v6.2_probe.patch
patch -s -f -N -p1 -i ../backports/v6.2_v4l2-core.patch
patch -s -f -N -p1 -i ../backports/v6.1_class.patch
patch -s -f -N -p1 -i ../backports/v6.1_i2c_probe.patch
patch -s -f -N -p1 -i ../backports/v6.1_videobuf2.patch
patch -s -f -N -p1 -i ../backports/v6.1_timer.patch
patch -s -f -N -p1 -i ../backports/v6.1_ipu3.patch
Patched drivers/media/dvb-core/dvbdev.c
Patched drivers/media/v4l2-core/v4l2-dev.c
Patched drivers/media/rc/rc-main.c
make[2]: Leaving directory '/home/user/src/tbsdtv/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/6.1.0-13-amd64/build /lib/modules/6.1.0-13-amd64/source 1
Preparing to compile for kernel version 6.1.0
...
```

And compilation/build and install is finally running through! :relieved:

Of course that build script hasn't changed in years. Not sure why I started bumping into this only lately.

crazycat69, 2023-10-27:
https://github.com/tbsdtv/media_build/commit/cb372162778cddbd1bf1fafda4db8c8fd16d32b5

closed 2023-11-18 by crazycat69