tbsdtv/linux_media #328: modprobe problem ubuntu 20.04/22.04, and menuconfig error

zsirmo, 2023-10-23:
Hello,

I found two problems:
1. ubuntu 20.04 and 22.04 "make menuconfig" not working:, error messge:
`/usr/src/20231022/media_build# make menuconfig
make -C /usr/src/20231022/media_build/v4l menuconfig
make[1]: Entering directory '/usr/src/20231022/media_build/v4l'
/lib/modules/6.2.0-1015-lowlatency/build/scripts/kconfig/mconf ./Kconfig
Kconfig.kern:102: syntax error
Kconfig.kern:101: unknown statement "option"
make[1]: *** [Makefile:371: menuconfig] Error 1
make[1]: Leaving directory '/usr/src/20231022/media_build/v4l'
make: *** [Makefile:26: menuconfig] Error 2`

I checked Kconfig.kern, and found orphaned line:
`# Needed by MEDIA_ATTACH, always
config MODULES
        tristate
        default y

        option modules
# Needed by VIDEO_RENESAS_VSP1, VIDEO_RENESAS_FDP1
config ARM64
        tristate
        default n`

2. build&install is done, but modprobe failed:
`modprobe  dvb-usb-tbs5580
modprobe: ERROR: could not insert 'dvb_usb_tbs5580': Exec format error`

dmesg output:
`[ 2764.761706] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 0000000054650132, val ffffffffc0887960`

crazycat69, 2023-10-23:
https://github.com/tbsdtv/media_build/commit/5287ad1b58b858c5d8f55e8839f937a68661ccbd

zsirmo, 2023-10-23:
okay, i try , thank you

build problem , i try unsigned kernel  and WORKING, but install modules-extra package, i2c-mux depending!

closed 2023-11-18 by crazycat69