tbsdtv/linux_media #330: TBS6903X - problem after locking data transponders

ikor7, 2023-11-06:
After locking any data transponders for example Eutelsat 16A, freq. 12507 H., is it becomes impossible to tune DVB-S2 until reloading the tbsecp3 driver or restart the PC.

crazycat69, 2023-11-06:
https://github.com/tbsdtv/linux_media/commit/bf63058ea6207875dd5e856af91d11376334d052

ikor7, 2023-11-06:
TP lock, but it won't search for any stations... Only a driver restart will help...
`
----------------------------------> Using DVB-S2
>>> tune to: 12507:HM2O20S1:S16E:1590:
DVB-S IF freq is 1907000
status 1f | signal strength  44% | snr  33% | ber 0 | unc 0
WARNING: filter timeout pid 0x0011
WARNING: filter timeout pid 0x0000
WARNING: filter timeout pid 0x0010
----------------------------------> Using DVB-S2
>>> tune to: 12516:VC23M2O20S1:S16E:19365:
DVB-S IF freq is 1916000
status 1f | signal strength  53% | snr  32% | ber 0 | unc 0
WARNING: filter timeout pid 0x0011
WARNING: filter timeout pid 0x0000
WARNING: filter timeout pid 0x0010

`

ikor7, 2023-11-07:
> [bf63058](https://github.com/tbsdtv/linux_media/commit/bf63058ea6207875dd5e856af91d11376334d052)

Updating the driver didn't help.

crazycat69, 2023-11-08:
https://github.com/tbsdtv/linux_media/commit/a2daf45472b598d845a24b37e8295d56890954c6

ikor7, 2023-11-08:
Solved. Well thank you.

closed 2023-11-08 by ikor7