tbsdtv/linux_media #331: Compilation error with Linux 6.6 - TBS6904SE

jansohn, 2023-11-12:
Starts failing to build with linux kernel 6.6 on my system. 6.5 was working correctly.

```
make -C /lib/modules/6.6.1-arch1-1/build M=/var/lib/dkms/tbs_6904_se/6.0.11/build/media_build/v4l  modules
make[2]: Entering directory '/usr/lib/modules/6.6.1-arch1-1/build'
  CC [M]  /var/lib/dkms/tbs_6904_se/6.0.11/build/media_build/v4l/m88rs6000t.o
In file included from <command-line>:
/var/lib/dkms/tbs_6904_se/6.0.11/build/media_build/v4l/compat.h:1197:19: error: static declaration of 'dev_err_probe' follows non-static declaration
 1197 | static inline int dev_err_probe(const struct device *dev, int err, const char *fmt, ...)
      |                   ^~~~~~~~~~~~~
In file included from ./include/linux/device.h:15,
                 from ./include/linux/input.h:19,
                 from /var/lib/dkms/tbs_6904_se/6.0.11/build/media_build/v4l/compat.h:47:
./include/linux/dev_printk.h:277:20: note: previous declaration of 'dev_err_probe' with type 'int(const struct device *, int,  const char *, ...)'
  277 | __printf(3, 4) int dev_err_probe(const struct device *dev, int err, const char *fmt, ...);
      |                    ^~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:243: /var/lib/dkms/tbs_6904_se/6.0.11/build/media_build/v4l/m88rs6000t.o] Error 1
make[3]: *** [/usr/lib/modules/6.6.1-arch1-1/build/Makefile:1913: /var/lib/dkms/tbs_6904_se/6.0.11/build/media_build/v4l] Error 2
make[2]: *** [Makefile:234: __sub-make] Error 2
```

crazycat69, 2023-11-18:
[tbsdtv/media_build](https://github.com/tbsdtv/media_build) and [tbsdtv/linux_media](https://github.com/tbsdtv/linux_media) updated for 6.6 compatibility.
Please check

jansohn, 2023-11-18:
Just tested it, unfortunately it's not working yet.

```
make -C firmware prep
make[2]: Entering directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/firmware'
Kernel build directory is /lib/modules/6.6.1-arch1-1/build
make -C ../linux apply_patches
make[2]: Entering directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/linux'
Syncing with dir ../../media
Patches for 6.6.1-arch1-1 already applied.
make[2]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/linux'
make -C /lib/modules/6.6.1-arch1-1/build M=/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l  modules
make[2]: Entering directory '/usr/lib/modules/6.6.1-arch1-1/build'
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6000t.o
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tuner-simple.o
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tuner-types.o
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tas2101.o
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tas2101.c: In function 'tas2101_attach':
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tas2101.c:683:1: warning: label 'err1' defined but not used [-Wunused-label]
  683 | err1:
      | ^~~~
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/gx1133.o
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/gx1133.c: In function 'gx1133_i2c_select':
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/gx1133.c:747:12: warning: unused variable 'temp' [-Wunused-variable]
  747 |         u8 temp;
      |            ^~~~
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/gx1133.c: In function 'gx1133_set_frontend':
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/gx1133.c:1179:9: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
 1179 |         if (ret)
      |         ^~
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/gx1133.c:1182:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
 1182 |                 if (fe->ops.tuner_ops.set_params) {
      |                 ^~
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.o
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c: In function 'si5351_write':
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c:353:12: warning: unused variable 'val' [-Wunused-variable]
  353 |         u8 val;
      |            ^~~
In file included from ./include/linux/device.h:15,
                 from ./include/linux/input.h:19,
                 from /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/compat.h:47,
                 from <command-line>:
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c: In function 'si5351_write_bulk':
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c:387:25: warning: format '%x' expects argument of type 'unsigned int', but argument 5 has type 'u8 *' {aka 'unsigned char *'} [-Wformat=]
  387 |                         "si5351(ret=%i, reg=0x%02x, value=0x%02x)\n",
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/dev_printk.h:110:30: note: in definition of macro 'dev_printk_index_wrap'
  110 |                 _p_func(dev, fmt, ##__VA_ARGS__);                       \
      |                              ^~~
./include/linux/dev_printk.h:144:56: note: in expansion of macro 'dev_fmt'
  144 |         dev_printk_index_wrap(_dev_err, KERN_ERR, dev, dev_fmt(fmt), ##__VA_ARGS__)
      |                                                        ^~~~~~~
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c:386:17: note: in expansion of macro 'dev_err'
  386 |                 dev_err(&i2c->dev,
      |                 ^~~~~~~
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c:387:64: note: format string is defined here
  387 |                         "si5351(ret=%i, reg=0x%02x, value=0x%02x)\n",
      |                                                             ~~~^
      |                                                                |
      |                                                                unsigned int
      |                                                             %02hhn
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/m88rs6060.c:374:12: warning: unused variable 'val' [-Wunused-variable]
  374 |         u8 val;
      |            ^~~
[... 84 lines cut ...]
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tbsecp3-dvb.c:869:12: warning: 'max_send_burst' defined but not used [-Wunused-function]
  869 | static int max_send_burst(struct dvb_frontend *fe, enum fe_sec_mini_cmd burst)
      |            ^~~~~~~~~~~~~~
/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tbsecp3-dvb.c:864:12: warning: 'max_send_master_cmd' defined but not used [-Wunused-function]
  864 | static int max_send_master_cmd(struct dvb_frontend *fe, struct dvb_diseqc_master_cmd *cmd)
      |            ^~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tbsecp3-ca.o
  CC [M]  /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/tbsecp3-asi.o
[... cut ...]
make[4]: *** [scripts/Makefile.modpost:145: /var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l/Module.symvers] Error 1
make[3]: *** [/usr/lib/modules/6.6.1-arch1-1/build/Makefile:1865: modpost] Error 2
make[2]: *** [Makefile:234: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.6.1-arch1-1/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build/v4l'
make: *** [Makefile:26: all] Error 2
make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.6/build/media_build'
```

jansohn, 2023-11-22:
Thanks, that did it!

closed 2023-11-22 by jansohn

vmc-coding, 2023-11-22:
Works for me too, thx.