tbsdtv/linux_media #333: TBS 6285 DVB-T2 Not working

Dadeos, 2023-11-19:
I have a TBS 6285 card that I have been successfully using, for both DVB-T (i.e. UK Freeview SD) and DVB-T2 (i.e. UK Freeview HD), reception with the closed-source (i.e. [v170330](https://www.tbsiptv.com/download/common/tbs-linux-drivers_v170330.zip)) drivers for many years. I managed to tweak the closed-source drivers to work with Linux Kernels up to version 4.18 and have basically stuck with that Linux Kernel.

I now have cause, for various reasons, to upgrade Linux Kernel - I'm trying 6.1.62 - and therefore am attempting to transition to the, now recommended, open-source drivers. I have managed to build and install the latest (i.e. 5bc283d) version, and DVB-T functionality appears to be working fine. However, DVB-T2 reception does not appear to work. Attempts to tune to previously stored HD channels fail, and attempts to scan for new channels do not detect any HD transports.

Is this a known issue? Is there any additional configuration required to enable DVB-T2 reception with this card when using the open-source drivers?

```
01:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)
        Subsystem: Device 6285:0001
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 256 bytes
        Interrupt: pin A routed to IRQ 94
        Region 0: Memory at fb800000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
                Address: 00000000fee05000  Data: 0023
        Capabilities: [50] Express (v1) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset- SlotPowerLimit 10W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr+ FatalErr- UnsupReq+ AuxPwr- TransPend-
                LnkCap: Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 128 bytes, Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1
                        TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
        Capabilities: [74] Power Management version 2
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Vendor Specific Information: Len=50 <?>
        Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
        Kernel driver in use: SAA716x Budget
        Kernel modules: saa716x_tbs_dvb
```
```
mc: loading out-of-tree module taints kernel.
mc: Linux media interface: v0.10
WARNING: You are using an experimental version of the media stack.
        As the driver is backported to an older kernel, it doesn't offer
        enough quality for its usage in production.
        Use it with care.
Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):

dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
i2c i2c-9: Added multiplexed i2c bus 10
si2168 9-0064: Silicon Labs Si2168-A20 successfully identified
si2168 9-0064: firmware version: A 2.0.3
si2157 10-0062: Silicon Labs Si2157 successfully attached
SAA716x Budget 0000:01:00.0: TurboSight TBS 6285  MAC[0]=ff:ff:ff:ff:ff:ff
SAA716x Budget 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
i2c i2c-9: Added multiplexed i2c bus 11
si2168 9-0066: Silicon Labs Si2168-A20 successfully identified
si2168 9-0066: firmware version: A 2.0.3
si2157 11-0060: Silicon Labs Si2157 successfully attached
SAA716x Budget 0000:01:00.0: TurboSight TBS 6285  MAC[1]=ff:ff:ff:ff:ff:ff
SAA716x Budget 0000:01:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
i2c i2c-8: Added multiplexed i2c bus 12
si2168 8-0064: Silicon Labs Si2168-A20 successfully identified
si2168 8-0064: firmware version: A 2.0.3
si2157 12-0062: Silicon Labs Si2157 successfully attached
SAA716x Budget 0000:01:00.0: TurboSight TBS 6285  MAC[2]=ff:ff:ff:ff:ff:ff
SAA716x Budget 0000:01:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
i2c i2c-8: Added multiplexed i2c bus 13
si2168 8-0066: Silicon Labs Si2168-A20 successfully identified
si2168 8-0066: firmware version: A 2.0.3
si2157 13-0060: Silicon Labs Si2157 successfully attached
SAA716x Budget 0000:01:00.0: TurboSight TBS 6285  MAC[3]=ff:ff:ff:ff:ff:ff
SAA716x Budget 0000:01:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
si2168 9-0064: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
si2168 9-0064: firmware version: \x81 2.0.35
si2157 10-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
si2168 9-0066: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
si2168 9-0066: firmware version: \x81 2.0.35
si2157 11-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
si2168 8-0064: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
si2168 8-0064: firmware version: \x81 2.0.35
si2157 12-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
si2168 8-0066: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
si2168 8-0066: firmware version: \x81 2.0.35
si2157 13-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
```

crazycat69, 2023-11-19:
what app used for reception ? look like linux-tbs-drivers support standard autodetect, but open-source si2168 frontend driver require proper delivery system setting

Dadeos, 2023-11-19:
I'm using MythTV, which has worked with the closed-source drivers. Everything about the system is identical, simply rebooting using a different Linux Kernel and TBS driver configuration.

crazycat69, 2023-11-19:
MythTV allow select standard (delivery system) for muxes ?

Dadeos, 2023-11-19:
I'm sorry, I have no idea. I'm assuming, perhaps incorrectly, that if the system works with the closed-source driver then it would/should also work with the open-source driver.

Can you suggest some other diagnostic approach that I could try to verify whether the issue is indeed with the driver configuration, or perhaps identify that it's actually being caused by MythTV?

Dadeos, 2023-11-19:
Ah, OK; your hint regarding "delivery system" caused me to discover the [dvb-fe-tool](https://www.linuxtv.org/wiki/index.php/Dvb-fe-tool) - I was previously unaware of this tool.

I can now see that the frontends are configured for "DVBT" delivery, and whilst I can configured them to "DVBT2" using the tool, running MythTV appears to configure them back to "DVBT". I suspect that this is where I need to explore next.

I presume that the closed-source drivers do not support this interface, and as such do not suffer this inconvenience.

Do you happen to know whether it's necessary to reconfigure the frontends when switching between "DVBT" and "DVBT2" channels, or is the "DVBT2" delivery system a superset of "DVBT"? (i.e. if I can persuade my system to stop reconfiguring the frontends to "DVBT" and get them to remain configured for "DVBT2", may I expect all SD/HD channel to become available, or will I only be able to view the HD channels using such a configuration?)

Thanks for the pointer!

crazycat69, 2023-11-19:
What MythTV version ?

Dadeos, 2023-11-19:
I'm using a really old version of MythTV, likely prior to the introduction of support for "delivery systems".

As I stated in my original post; I've been stuck using the closed-source driver on Linux Kernel 4.18 - which dates my last efforts in this regard around 2018. I can't really remember that long ago, but I suspect that I probably encountered this issue (i.e. the open-source drivers not supporting DVB-T2 with the version of MythTV I'm using) and I just kept with what worked. I suspect that I need to explore either upgrading MythTV or hacking the version I'm currently using to request the "DVBT2" delivery system, which the [link](https://www.mythtv.org/wiki/Channel_Scanning#Delivery_system) you provided seems to suggest is the desired configuration if one wishes to receive both SD and HD channels.

Thank you for pointing me in the right direction; you've doubtless save me hours of digging otherwise required to work out where I was going wrong.

closed 2023-11-19 by Dadeos