tbsdtv/linux_media #335: Build fails with latest Debian 12 kernel 6.1.0-14-amd64

ritzmann, 2023-12-09:
```
$ git clean -dfx
$ git pull
Already up to date.
$ ./build
Checking if the needed tools for Debian GNU/Linux 12 (bookworm) are available
Needed package dependencies are met.

************************************************************
* This script will download the latest tarball and build it*
* Assuming that your kernel is compatible with the latest  *
* drivers. If not, you'll need to add some extra backports,*
* ./backports/<kernel> directory.                          *
* It will also update this tree to be sure that all compat *
* bits are there, to avoid compilation failures            *
************************************************************
************************************************************
* All drivers and build system are under GPLv2 License     *
* Firmware files are under the license terms found at:     *
* https://github.com/tbsdtv/media_build/releases/download/latest *
* Please abort in the next 5 secs if you don't agree with  *
* the license                                              *
************************************************************

Not aborted. It means that the licence was agreed. Proceeding...

****************************
Updating the building system
****************************
From https://github.com/tbsdtv/media_build
 * tag               latest     -> FETCH_HEAD
Auto-merging v4l/versions.txt
CONFLICT (content): Merge conflict in v4l/versions.txt
error: could not apply e683266... Enable two drivers with backports
hint: Resolve all conflicts manually, mark them as resolved with
hint: "git add/rm <conflicted_files>", then run "git rebase --continue".
hint: You can instead skip this commit: run "git rebase --skip".
hint: To abort and get back to the state before "git rebase", run "git rebase --abort".
Could not apply e683266... Enable two drivers with backports
make: Entering directory '/home/user/src/tbsdtv/media_build/linux'
wget https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5 -O linux-media.tar.bz2.md5.tmp
--2023-12-09 19:40:40--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2.md5
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/a0884b3b-e946-4614-8657-18bb9812b5fc?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231209T174040Z&X-Amz-Expires=300&X-Amz-Signature=277a2cacb46d19c7e91a21df2a0195eaa9eb8724bc9b1aeaae5c4189096e126f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream [following]
--2023-12-09 19:40:40--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/a0884b3b-e946-4614-8657-18bb9812b5fc?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231209T174040Z&X-Amz-Expires=300&X-Amz-Signature=277a2cacb46d19c7e91a21df2a0195eaa9eb8724bc9b1aeaae5c4189096e126f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2.md5&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 87 [application/octet-stream]
Saving to: ‘linux-media.tar.bz2.md5.tmp’

linux-media.tar.bz2.md5.tmp                            100%[=========================================================================================================================>]      87  --.-KB/s    in 0s      

2023-12-09 19:40:41 (6.30 MB/s) - ‘linux-media.tar.bz2.md5.tmp’ saved [87/87]

cat: linux-media.tar.bz2.md5: No such file or directory
--2023-12-09 19:40:41--  https://github.com/tbsdtv/media_build/releases/download/latest/linux-media.tar.bz2
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/ec1190fb-3d56-4220-af4f-080eb22053e5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231209T174041Z&X-Amz-Expires=300&X-Amz-Signature=194af4280c26a4d35270b432d15826d224c3faae9061adebfe4da35b19fc8126&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2&response-content-type=application%2Foctet-stream [following]
--2023-12-09 19:40:41--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/61525972/ec1190fb-3d56-4220-af4f-080eb22053e5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231209%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231209T174041Z&X-Amz-Expires=300&X-Amz-Signature=194af4280c26a4d35270b432d15826d224c3faae9061adebfe4da35b19fc8126&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=61525972&response-content-disposition=attachment%3B%20filename%3Dlinux-media.tar.bz2&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.108.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8262762 (7.9M) [application/octet-stream]
Saving to: ‘linux-media.tar.bz2’

[... 148 lines cut ...]
The text leading up to this was:
--------------------------
|diff --git a/drivers/media/cec/cec-api.c b/drivers/media/cec/cec-api.c
|index 10b67fc..64b0250 100644
|--- a/drivers/media/cec/cec-api.c
|+++ b/drivers/media/cec/cec-api.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
The text leading up to this was:
--------------------------
|diff --git a/drivers/media/media-device.c b/drivers/media/media-device.c
|index 35e81f7..9c3361c 100644
|--- a/drivers/media/media-device.c
|+++ b/drivers/media/media-device.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/user/src/tbsdtv/media_build/linux'
make[1]: *** [Makefile:386: allyesconfig] Error 2
make[1]: Leaving directory '/home/user/src/tbsdtv/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2
can't select all drivers at ./build line 552
```

crazycat69, 2023-12-09:
try fresh clone [media_build](https://github.com/tbsdtv/media_build)

ritzmann, 2023-12-10:
Deleted and cloned media_build: Exactly same error.

crazycat69, 2023-12-10:
https://github.com/tbsdtv/media_build/commit/88764363a3e3d36b3c59a0a2bf2244e262035d47

ritzmann, 2023-12-10:
That seems to have done the trick, thanks! I did not do an entirely clean update because Debian pulled the 6.1.0-14 kernel yesterday and released 6.1.0-15 today with a critical bug fix but I doubt that made a difference.

closed 2023-12-10 by ritzmann

crazycat69, 2023-12-11:
look like this related to your git settings related to merge strategy.