tbsdtv/linux_media #337: TBS 6909X v2  - i2c errors

byserkhan, 2024-01-12:
Hi to all.

system configuration like this;

OS: Ubuntu Server 20.04
kernel version: 5.4.0
TBS card: 1x 6909, 1x 6909x v2

There is no issue on 6909
There are too many issue on 6909x v2 and i am still facing with that problem. Sometimes getting "restarting adapter" error on cesbo.
Also after a while server hang couse of that i2c error.

Here is the dmesg errors
[image]

Also cesbo reporting that issue. After reporting adapter restart automaticly.
[image]

crazycat69, 2024-01-12:
look like 6909x HW problem

byserkhan, 2024-01-12:
> look like 6909x HW problem

HW meaning tbs cards somehow broken or what else ?

crazycat69, 2024-01-13:
yes