tbsdtv/linux_media #338: Raspberry pi 5 - Debian GNU/Linux 12 (bookworm) -driver TBS 5520se fails

Jaro6904, 2024-01-30:
Hello,

I followed all installation steps :
cd media_build
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j4
sudo make install

All installation steps went without any problem. Only some warning in make -j4 step.

Tvheadend installation was successfull too. Tvheadend does not show TBS drivers.

I dont know where the problem is but command **dmesg | grep frontend**
 shows : 
                81ba127ee153c1f1565fbb33a65ed36ad13c52bd media: dvb-frontends/gx1133: PLS support.
                81ba127ee153c1f1565fbb33a65ed36ad13c52bd media: dvb-frontends/gx1133: PLS support.
[   16.361299] dvb-usb: **no frontend was attached** by 'TurboSight TBS 5520SE'
[   17.980380] usb 1-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   17.980476] usb 1-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   31.942405] usb 1-1: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[   33.545322] dvb-usb: no frontend was attached by 'TurboSight TBS 5520SE'
[   35.077568] usb 1-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   35.077649] usb 1-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   49.378101] usb 1-1: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[   50.909512] dvb-usb: no frontend was attached by 'TurboSight TBS 5520SE'
[   52.446080] usb 1-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   52.446164] usb 1-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   53.229412] usb 1-1: dvb_frontend_stop: warning: thread 000000001d60019f won't exit
[   54.294785] dvb-usb: no frontend was attached by 'TurboSight TBS 5520SE'
[   55.851114] usb 1-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   55.851200] usb 1-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...

Can you please help me with ?

Thank you in advance.

Jaromir

mareksaw-cg, 2024-02-07:
Are you sure the tuner is OK? Could you check it in Windows?

Jaro6904, 2024-02-07:
I am sure, the tuner is 100% OK. I have second raspberry pi 4 with Libreelec installation with Crazycat drivers and tuner works fine with it

Jaro6904, 2024-02-07:
Do you run it on rpi4 or rpi5 similar to me ? I think that your version of linux is not compatible with rpi5.

I have opened it in https://www.tbsdtv.com/forum/viewtopic.php?f=87&t=25916&sid=235527af11fb554cce38ae1b48dd50b0 . But without response yet :-(

mareksaw-cg, 2024-02-07:
Yes, it is a rpi5 with 5520SE attached. This was just an output of uname -a from a working machine.

Jaro6904, 2024-02-07:
Can you please share download link with your version of Raspbian with me ? Raspberri Pi imager offer me only last version of Raspbian.

Thank you in advance.

mareksaw-cg, 2024-02-07:
Unfortunately I have no such a link. I created my system on 28.12.2023 and I installed the current OS version which was available at that time.
I am not sure if it is at all possible to install an older version.

Jaro6904, 2024-02-07:
Great. I am going to reinstall my version with the older one. I share output after I'll test instalation of TBS drivers.
Thank you for your help

mareksaw-cg, 2024-02-07:
By the way... Did you upgraded the kernel after installing the current version (sudo apt-get upgrade)?

Jaro6904, 2024-02-07:
Yes, I did it. It is usually one of my first command after I installed a new system. 
sudo apt-get update and then sudo apt-get upgrade

mareksaw-cg, 2024-02-07:
Just do update after installing, no upgrade.
It seems that the 11.12.2023 is the latest release. And with this version I have the 6.1.63 kernel, while your uname -a says you have 6.1.73. It may be so that something from the TBS drivers is not compatible with newest kernel? So check TBS without upgrading first.

crazycat69, 2024-02-07:
> Tvheadend installation was successfull too. Tvheadend does not show TBS drivers.
> I dont know where the problem is but command dmesg | grep frontend

Please provide full dmesg log

Jaro6904, 2024-02-08:
Hello,
I reinstall version of Raspberry to Linux raspberrypi 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

All installation steps were withou any error . But I can not see TBS in tvheadend.

The list of all commands that I run during installation :
 1  sudo apt-get update
    2  uname -a
    3  mkdir tbs
    4  cd tbs
    5  sudo apt-get install gcc-8
    6  apt-get install g++-8
    7  apt-get install gcc
    8  apt-get install linux-headers-generic
    9  apt-get install patchutils
   10  apt-get install libproc-processtable-perl
   11  git clone https://github.com/tbsdtv/media_build.git
   12  apt-get install git
   13  git clone https://github.com/tbsdtv/media_build.git
   14  git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
   15  cd media_build
   16  make dir DIR=../media
   17  make allyesconfig
   18  sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
   19  sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
   20  make -j4
   21  make install
   22  reboot
   23  cd tbs
   24  wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
   25  tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/
   26  lsusb
   27  reboot
   28  apt-get install tvheadend
   29  reboot
   30  lsusb
   31  dmesg | grep frontend
   32  history

Jaro6904, 2024-02-08:
**dmesg**
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.0-rpi7-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001a000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s177448 r8192 d43736 u229376
[    0.000000] pcpu-alloc: s177448 r8192 d43736 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 260608
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:ED:1D:71 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=c7827dd3-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 8, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 7, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 3796512K/4186112K available (12352K kernel code, 2186K rwdata, 4112K rodata, 4480K init, 1216K bss, 61920K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40061 entries in 40 pages
[    0.000000] ftrace: allocated 40 pages with 2 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000154] Console: colour dummy device 80x25
[    0.000299] printk: console [tty1] enabled
[    0.000317] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000327] pid_max: default: 32768 minimum: 301
[    0.000354] LSM: Security Framework initializing
[    0.000412] Mount-cache hash table entries: 8192 (order: 2, 65536 bytes, linear)
[    0.000429] Mountpoint-cache hash table entries: 8192 (order: 2, 65536 bytes, linear)
[    0.000843] cgroup: Disabling memory control group subsystem
[    0.001318] cblist_init_generic: Setting adjustable number of callback queues.
[    0.001329] cblist_init_generic: Setting shift to 2 and lim to 1.
[... 929 lines cut ...]
[   40.084044] usb 3-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   40.094742] usb 3-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   40.105027] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[   40.118655] usb usb3-port2: over-current change #5
[   40.259635] usb 3-1: USB disconnect, device number 13
[... cut ...]
[   41.774287] si2183 13-0067: Silicon Labs Si2183 successfully attached
[   41.784322] si2157 14-0061: Silicon Labs Si2157 successfully attached
[   41.791749] i2c i2c-14: av201x: Airoha Technology AV201x successfully attached
[   41.860078] usb 3-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   41.870785] usb 3-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   41.880943] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.

Jaro6904, 2024-02-08:
I found in Tvheadend log :
2024-02-08 12:30:31.769 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:32.146 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:32.499 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:35.141 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:35.520 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:35.875 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:38.553 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:38.932 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:39.291 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:39.431 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
2024-02-08 12:30:41.933 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:42.313 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:42.667 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:45.309 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:45.686 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:46.039 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:48.685 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:49.064 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:49.420 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:49.437 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
2024-02-08 12:30:52.061 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:52.440 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:52.795 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:55.441 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:55.821 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:56.175 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:58.817 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:30:59.197 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:30:59.552 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:02.197 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:02.577 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:02.931 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:05.573 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:05.952 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:06.308 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:06.450 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
2024-02-08 12:31:08.949 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:09.326 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:09.684 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:12.329 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:12.709 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:13.063 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:15.705 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:16.085 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:16.439 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:16.465 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
2024-02-08 12:31:19.081 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:19.461 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:19.815 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:22.461 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:22.841 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:23.195 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:25.837 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:26.216 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:26.572 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:26.771 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
2024-02-08 12:31:29.253 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:29.632 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:29.987 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:32.637 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:33.014 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:33.367 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:33.780 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
2024-02-08 12:31:36.009 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:36.389 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:36.743 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:39.386 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:31:39.766 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:31:40.119 linuxdvb: adapter removed /dev/dvb/adapter0
[... 101 lines cut ...]
2024-02-08 12:33:24.503 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:33:24.855 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:27.501 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:27.882 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:33:28.236 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:30.878 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:31.258 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:33:31.611 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:31.782 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400
[... cut ...]
2024-02-08 12:33:41.751 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:44.393 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:44.770 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:33:45.123 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:47.765 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:48.145 linuxdvb: adapter added /dev/dvb/adapter0
2024-02-08 12:33:48.499 linuxdvb: adapter removed /dev/dvb/adapter0
2024-02-08 12:33:48.620 http: 192.168.252.4: HTTP/1.1 POST (3) /api/hardware/tree -- 400

mareksaw-cg, 2024-02-08:
But as I can see tuner is now properly recognized by OS.
I had a similar issue with TBS-5530. DVB-S tuner was not recognized by tvheadend. The log from tvheadend was similar to yours.

alexiedb, 2025-03-23:
> > But as I can see tuner is now properly recognized by OS. I had a similar issue with TBS-5530. DVB-S tuner was not recognized by tvheadend. The log from tvheadend was similar to yours.
> 
> frontend 1 is use for dvbs mode, so you need add the link for frontend (demux,dvr and net). #cd /dev/dvb/adapter0 #ln -s dvr0 dvr1 #ln -s demux0 demux1 #ln -s net0 net1 then restart tvh.

For me this does not work, I still get this error:
"
Mar 23 17:13:53 alexie-VirtualBox tvheadend[3032]: linuxdvb: adapter added /dev/dvb/adapter0
Mar 23 17:14:35 alexie-VirtualBox tvheadend[3032]: scanfile: DVB-S - loaded 1 regions with 116 networks
Mar 23 17:14:35 alexie-VirtualBox tvheadend[3032]: scanfile: DVB-T - loaded 46 regions with 1134 networks
Mar 23 17:14:35 alexie-VirtualBox tvheadend[3032]: scanfile: DVB-C - loaded 20 regions with 82 networks
Mar 23 17:14:35 alexie-VirtualBox tvheadend[3032]: scanfile: ATSC-T - loaded 2 regions with 13 networks
Mar 23 17:14:35 alexie-VirtualBox tvheadend[3032]: scanfile: ATSC-C - loaded 1 regions with 5 networks
Mar 23 17:14:35 alexie-VirtualBox tvheadend[3032]: scanfile: ISDB-T - loaded 2 regions with 1297 networks
Mar 23 17:15:39 alexie-VirtualBox tvheadend[3032]: **linuxdvb: unable to open /dev/dvb/adapter0/frontend1**
Mar 23 17:15:39 alexie-VirtualBox tvheadend[3032]: dvr: Creating new configuration ''
Mar 23 17:15:39 alexie-VirtualBox tvheadend[3032]: access: No access entries loaded
Mar 23 17:15:39 alexie-VirtualBox tvheadend[3032]: epggrab: module uk_freesat created

"
and DVB-S tuner doesn't show up in tvheadend.

Is there any way to make it working with tvheadend?