tbsdtv/linux_media #340: TBS 6981 with Debian 12, 6.1.0-17-amd64, error running FW, failed to set voltage

PlutoHDDev, 2024-02-07:
- Debian 12
- Kernel: 6.1.0-17-amd64
- TBS 6981
- Installed driver as described here: https://www.linuxtv.org/wiki/index.php/TBS_driver_installation#Buidling_TBS'_forked_driver

`dmesg` (and shown after boot):
```
i2c i2c-1: cx24117_load_firmware() Error running FW.
i2c i2c-1: cx24117_firmware_ondemand: Writing firmware failed
```

TVHeadend when running a scan:
```
mpegts: 11494H in Astra 19.2 - tuning on TurboSight TBS 6981 DVB-S/S2 #1 : DVB-S #0
diseqc: failed to set voltage (e=I/O error)
```
[image]

What may cause this issue? Installing the driver seems to work just fine, I can set up the tuners in TVHeadend as you can see in the screenshot, but something still does not work.
Driver/Firmware issue? BIOS setting? Anything else? I have tried/researched many things but nothing helped unfortunately.

PlutoHDDev, 2024-02-07:
This is the `lsmod` response: https://pastebin.com/K1k9rAd8

crazycat69, 2024-02-07:
full dmesg log please

PlutoHDDev, 2024-02-07:
<details><summary>dmesg log</summary>
<p>

```
[    0.000000] Linux version 6.1.0-17-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30)
[    0.000000] Command line: BOOT_IMAGE=/@/boot/vmlinuz-6.1.0-17-amd64 root=UUID=37baa075-3644-4071-83b0-00648435a514 ro rootflags=subvol=@ quiet
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009c3efff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009c3f000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20b000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b020000-0x0000000039ff8fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000039ff9000-0x000000003c9cafff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003c9cb000-0x000000003ca0cfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000003ca0d000-0x000000003d0bcfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000003d0bd000-0x000000003df79fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003df7a000-0x000000003dffefff] type 20
[    0.000000] BIOS-e820: [mem 0x000000003dfff000-0x000000003effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003f000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd100000-0x00000000fd1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000fea0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec30000-0x00000000fec30fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc2000-0x00000000fedcffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd4000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000083f33ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000083f340000-0x000000083fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI=0x3d0a6000 ACPI 2.0=0x3d0a6014 TPMFinalLog=0x3d070000 SMBIOS=0x3de2b000 SMBIOS 3.0=0x3de2a000 MEMATTR=0x363d8018 ESRT=0x382a0e18 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. B550I AORUS PRO AX/B550I AORUS PRO AX, BIOS F14 10/29/2021
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3493.475 MHz processor
[    0.000638] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000641] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000653] last_pfn = 0x83f340 max_arch_pfn = 0x400000000
[    0.000869] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001057] e820: update [mem 0xc0000000-0xffffffff] usable ==> reserved
[    0.001078] last_pfn = 0x3f000 max_arch_pfn = 0x400000000
[    0.005879] esrt: Reserving ESRT space from 0x00000000382a0e18 to 0x00000000382a0e50.
[    0.005893] e820: update [mem 0x382a0000-0x382a0fff] usable ==> reserved
[    0.005955] Using GB pages for direct mapping
[    0.006487] RAMDISK: [mem 0x2e9a6000-0x3093afff]
[    0.006493] ACPI: Early table checksum verification disabled
[    0.006496] ACPI: RSDP 0x000000003D0A6014 000024 (v02 ALASKA)
[    0.006500] ACPI: XSDT 0x000000003D0A5728 0000D4 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.006506] ACPI: FACP 0x000000003C9F6000 000114 (v06 ALASKA A M I    01072009 AMI  00010013)
[    0.006511] ACPI: DSDT 0x000000003C9EF000 006097 (v02 ALASKA A M I    01072009 INTL 20190509)
[    0.006514] ACPI: FACS 0x000000003D0A0000 000040
[    0.006517] ACPI: SSDT 0x000000003CA02000 00A4F8 (v02 GBT    GSWApp   00000001 INTL 20190509)
[    0.006520] ACPI: SSDT 0x000000003C9FC000 005419 (v02 AMD    Artic    00000002 MSFT 04000000)
[    0.006523] ACPI: SSDT 0x000000003C9F8000 003D06 (v01 AMD    AMD AOD  00000001 INTL 20190509)
[    0.006526] ACPI: SSDT 0x000000003C9F7000 0000C5 (v02 ALASKA CPUSSDT  01072009 AMI  01072009)
[    0.006529] ACPI: FIDT 0x000000003C9EE000 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.006532] ACPI: MCFG 0x000000003C9ED000 00003C (v01 ALASKA A M I    01072009 MSFT 00010013)
[    0.006535] ACPI: HPET 0x000000003C9EC000 000038 (v01 ALASKA A M I    01072009 AMI  00000005)
[    0.006538] ACPI: BGRT 0x000000003C9EB000 000038 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.006540] ACPI: TPM2 0x000000003C9EA000 00004C (v04 ALASKA A M I    00000001 AMI  00000000)
[... 1568 lines cut ...]
[   31.951635] br-13b35e68f1a4: port 8(veth3acbce2) entered blocking state
[... cut ...]
[   44.484480] cx23885 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
[   45.664461] i2c i2c-1: cx24117_cmd_execute_nolock() Firmware not responding
[   47.580488] i2c i2c-1: cx24117_cmd_execute_nolock() Firmware not responding
[  316.448533] EXT4-fs (md0): error count since last fsck: 2
[  316.448543] EXT4-fs (md0): initial error at time 1700252531: htree_dirblock_to_tree:1082: inode 246680215
[  316.448557] EXT4-fs (md0): last error at time 1700413493: ext4_validate_block_bitmap:420
```

</p>
</details>

crazycat69, 2024-02-07:
Not need install TBS media_build for 6980/6981,
Support for this cards available in kernel with RC support
Only need install firmwares from [dvb-firmwares.tar.bz2](https://github.com/tbsdtv/media_build/releases/download/latest/dvb-firmwares.tar.bz2). Need at least **v4l-cx23885-avcore-01.fw**, **dvb-fe-cx24117.fw** on /lib/firmware

PlutoHDDev, 2024-02-07:
Okay, thanks! Is there an easy way to uninstall media_build?

closed 2024-02-08 by PlutoHDDev