tbsdtv/linux_media #343: Compile for another platform

Tnix76, 2024-02-29:
Hi,

I'm on Debian x64 system and I can compile succesfully TBS drivers for my installation. But how can I compile for another Linux machine or distro? Should I download the target kernel headers and configure anywhere in the source tree? How does it do?

Thanks.

crazycat69, 2024-02-29:
yes, you need target kernel headers

Tnix76, 2024-02-29:
Ok, and how can I instruct the compiler or make process to find the target kernel headers instead my local machine ones? Maybe any env variable in a Makefile? I'm lost and I don't know it ...

Thanks.

crazycat69, 2024-03-01:
build script support version option:

> root@computer:/usr/local/src/media_build$ ./build --help
> Usage:
>     build [--help] [--man] [--verbose] [--check-only] [<--git> [URL]
>     [BRANCH]] [--main-git] [--depth [DEPTH]] [--version [KERNELVERSION]]
> 
> Options:
>     --help  Print a brief help message and exits.
> 
>     --man   Prints the manual page and exits.
> 
>     --verbose
>             Be more verbose.
> 
>     --check-only
>             Don't do anything, except for checking if the needed
>             dependencies are there.
> 
>     --git [URL] [BRANCH]
>             Allows specifying a URL and a git branch, instead of the default
>             ones. Currently, only linuxtv.org git URL's are supported, as
>             the build needs to warrant an unique namespace for git remotes.
> 
>     --main-git
>             Use the main development git tree, as found at
>             <http://git.linuxtv.org/media_tree.git>.
> 
>     --depth [DEPTH]
>             When doing a git clone (e. g. with --git or --main-git), pass
>             the depth parameter, in order to get a smaller tree.
> 
>             That helps to reduce disk storage and download time.
> 
>     --version [KERNELVERSION]
>             Try to build for a particular kernel version. The format of the
>             version string is the same as the output from 'uname -r'.

closed 2024-08-02 by crazycat69