tbsdtv/linux_media #344: build failing on kernel 6.8 (Ubuntu)

ashleyw-gh, 2024-03-08:
hi, build seems to be failing on Ubuntu on latest 6.8 kernel with the following message;
```
#  uname -a
Linux xxxxxx 6.8.0-11-generic #11-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 14 00:29:05 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

/root/tbsdriver/media_build/v4l/tbs_pcie.c:112:17: error: implicit declaration of function 'strlcpy'; did you mean 'strscpy'? [-Werror=implicit-function-declaration]
  112 |                 strlcpy(f->description, "YUV 4:2:2", sizeof(f->description));
      |                 ^~~~~~~
      |                 strscpy
```

build script i've been using for a long time is the following;
```
cd /root
rm -r -f tbsdriver
mkdir tbsdriver
cd tbsdriver
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make distclean
make -j4
make install
rm -f dvb-firmwares.tar.bz2
wget https://github.com/tbsdtv/media_build/releases/download/latest/dvb-firmwares.tar.bz2
tar jxvf dvb-firmwares.tar.bz2 -C /lib/firmware/
chown root:root /lib/firmware/* -R
reboot
```
any ideas?

ashleyw-gh, 2024-03-09:
right i think ive been able to get to the bottom of things;
the strlcpy API has been removed form kernel 6.8;

https://lwn.net/Articles/958178/
The task of [removing the strlcpy() API](https://lwn.net/Articles/905777/) from the kernel is now complete.

I tried to work around the issue by doing an;
apt-get install libbsd-dev

but I don't know what i'd need to do to be able to compile the drivers.
Please can the TBS driver code be modified to remove the strlcpy dependencies to make it easier to use?

ashleyw-gh, 2024-03-09:
in the meantime, I've reverted my system back to Ubuntu 23.10 which is running kernel 6.5.0-25 and the TBS drivers compile as expected. It would be great if @crazycat69 could work his magic on the TBS drivers to modernise the code to prepare it for 6.8 kernel versions. thanks.

crazycat69, 2024-03-10:
f5c03ca466888bf6a9d5322c71493c173fb0fc35
82aaaf6cb36e8050bccf84d3f116c5cb316f238e
b3d6de4067b98e76868ba41b569e3b1d1ab0f2e2
282b6872f825a15a5a16e1abb6eb618af2840778
48a32d973a1bac68bb7d9b7a9cea9a7963b2bf79
804a1febcc3dbf61e9140584a004233abc8c25f6
508dae75e4d1c2b0af0fb9ba4aea702e20f84af3

mochoandre, 2024-03-11:
yesterday I compiled in Ubuntu 22.04 with kernel 6.5. Now It's not compiling with 

```
make[2]: Entering directory '/usr/src/linux-headers-6.5.0-25-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
make[2]: Leaving directory '/usr/src/linux-headers-6.5.0-25-generic'
./scripts/rmmod.pl check
found 582 modules
make[1]: Leaving directory '/root/media_build/v4l'
root@Ubuntu2204TVH:~/media_build# uname -a
Linux Ubuntu2204TVH 6.5.0-25-generic #25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Feb 20 16:09:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

This changes?

crazycat69, 2024-03-11:
Maybe compiled before and no changes :)

ashleyw-gh, 2024-03-22:
I've finally got around to attempting an upgrade to Ubuntu 24.04 again (to get kernel 6.8) and I've attempted a compile and I'm getting the following error;
``
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-p2m.o
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-v4l2.o
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-tw28.o
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-gpio.o
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-disp.o
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-enc.o
  CC [M]  /root/tbsdriver/media_build/v4l/solo6x10-v4l2-enc.o
gcc-13: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [scripts/Makefile.build:243: /root/tbsdriver/media_build/v4l/solo6x10-enc.o] Error 1
make[4]: *** Waiting for unfinished jobs....
gcc-13: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [scripts/Makefile.build:243: /root/tbsdriver/media_build/v4l/solo6x10-core.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.8.0-11-generic/Makefile:1926: /root/tbsdriver/media_build/v4l                                                       ] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-11-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/tbsdriver/media_build/v4l'
make: *** [Makefile:26: all] Error 2
``
When I switch back to Ubuntu 23.10 the same compile scripts work fine using kernel 6.5.0-26.
Any ideas @crazycat69 ?
love your work by the way.

ashleyw-gh, 2024-04-07:
just to close the loop on this, I've just attempted another upgrade to Ubuntu 24.04 and the build is now working, so i think the previous changes that @crazycat69 made are working well  and it appears that the version dependencies on the Ubuntu side with gcc are now resolving themselves as 24.04 launch gets closer. thanks again!

```
root@????:~# lsb_release -d
No LSB modules are available.
Description:    Ubuntu Noble Numbat (development branch)
root@????:~# uname -a
Linux ???? 6.8.0-11-generic #11-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 14 00:29:05 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
root@????:~# dmesg|grep 6205
[    0.306205] pci 0000:00:18.4: bridge window [io  0x1000-0x0fff] to [bus 1f] add_size 1000
[    4.273507] TBSECP3 driver 0000:03:00.0: TurboSight TBS 6205 DVB-T/T2/C
[    4.404464] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[    4.534453] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[    4.667977] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[    4.801250] TBSECP3 driver 0000:03:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[    4.801298] TBSECP3 driver 0000:03:00.0: TurboSight TBS 6205 DVB-T/T2/C : PCI 0000:03:00.0, IRQ 66, MMIO 0xfea00000
```

ashleyw-gh, 2024-06-07:
in case this helps anyone else, the driver compilation was also failing on Ubuntu x64 again, (for different reasons mostly threading on unrelated sub components). I stumbled across this; https://forums.gentoo.org/viewtopic-p-8363634.html
I suspect like many I have a card (like the TBS6205) and I simply want to compile the drivers for that card etc to get it working under Linux.
So in my case I modified my build scrips to specifically build the required components so I can use the card under tvheadend.
It would be really useful if a simple menu system could ask what type of card you were using eg. 6205/6209 etc and then create a simple v4l ".config" file that can be used to minimally compile the required drivers. The problem most people have is that the entire v4l tree is compiled and a single error on an unrelated component can cause the whole compile to crash out.
Most users of TBS cards are likely to not have worked with "make config" etc nor have any idea as to which components are required to get a successful compilation and are only going through this process as the common TBS drivers themselves are not part of the kernel.
And yeah a minimal DKMS script that the average user could use would be even more useful especially if it allowed for specific drivers to be chosen.
Thanks for all your on-going hard work helping to keep the TBS hardware running as well as it did when it was released!

I've pasted in the repeatable code below, I use to compile the driver for the 625 card specially - it runs far faster than compiling the complete driver tree obviously. 

```
cd /root
rm -r -f tbsdriver
mkdir tbsdriver
cd tbsdriver
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make distclean

# from https://forums.gentoo.org/viewtopic-p-8363634.html
cat << 'EOF' >v4l/.config
CONFIG_RATIONAL=y
CONFIG_NET=y
CONFIG_FB_CFB_COPYAREA=m
CONFIG_PNP=y
CONFIG_FIREWIRE=m
CONFIG_MMU=y
CONFIG_TTY=y
CONFIG_GENERIC_ALLOCATOR=y
CONFIG_DMA_SHARED_BUFFER=y
CONFIG_HAS_DMA=y
CONFIG_CRC32=y
CONFIG_SYSFS=y
CONFIG_I2C_MUX=m
CONFIG_HAVE_CLK=y
CONFIG_SND_PCM=m
CONFIG_USB_ARCH_HAS_HCD=y
CONFIG_REGMAP_I2C=y
CONFIG_I2C=y
CONFIG_PCI_MSI=y
CONFIG_ACPI=y
CONFIG_NEW_LEDS=y
CONFIG_FONT_8x16=y
CONFIG_MODULES=y
CONFIG_FB_CFB_FILLRECT=m
CONFIG_ISA_DMA_API=y
CONFIG_FW_LOADER=y
CONFIG_LEDS_CLASS=y
CONFIG_PCI=y
CONFIG_VIRT_TO_BUS=y
CONFIG_I2C_ALGOBIT=m
CONFIG_INPUT_EVDEV=y
CONFIG_COMMON_CLK=y
CONFIG_FONT_SUPPORT=m
CONFIG_USB=y
CONFIG_FB=m
CONFIG_FB_CFB_IMAGEBLIT=m
CONFIG_HDMI=y
CONFIG_SND=y
CONFIG_HAS_IOMEM=y
CONFIG_INET=y
CONFIG_BITREVERSE=y
CONFIG_X86=y
CONFIG_SERIO=y
CONFIG_INPUT=y
CONFIG_REGMAP=y
CONFIG_USB_ACM=m
CONFIG_DEBUG_FS=y
CONFIG_MEDIA_SUPPORT=m
CONFIG_MEDIA_DIGITAL_TV_SUPPORT=y
CONFIG_CEC_CORE=m
CONFIG_MEDIA_CEC_SUPPORT=y
CONFIG_USB_PULSE8_CEC=m
CONFIG_DVB_CORE=m
CONFIG_DVB_NET=y
CONFIG_DVB_MAX_ADAPTERS=48
CONFIG_DVB_DYNAMIC_MINORS=y
CONFIG_MEDIA_PCI_SUPPORT=y
CONFIG_DVB_TBSECP3=m
CONFIG_DVB_PLATFORM_DRIVERS=y
CONFIG_MEDIA_SUBDRV_AUTOSELECT=y
CONFIG_MEDIA_ATTACH=y
CONFIG_MEDIA_TUNER=m
CONFIG_MEDIA_TUNER_SIMPLE=m
CONFIG_MEDIA_TUNER_SI2157=m
CONFIG_DVB_SI2168=m
CONFIG_DVB_TAS2101=m
CONFIG_MISC_DEVICES=y
CONFIG_ALTERA_STAPL=m
EOF

make -j4
make install
cd /root
rm -f tbs-tuner-firmwares_v1.0.tar.bz2
wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/
reboot

```

closed 2024-07-01 by crazycat69