tbsdtv/linux_media #345: TBS driver causing issues with graphics acceleration

Liam-Whiteside, 2024-03-11:
Every time I update the kernel (Ubuntu 22.04), I can use hardware acceleration and apps like intel_gpu_top work as expected. modprobe i915 works.

Once I've installed the TBS drivers using the script below, these stop working eg
$ sudo intel_gpu_top
No device filter specified and no discrete/integrated i915 devices found
$ sudo modprobe i915
modprobe: ERROR: could not insert 'i915': Unknown symbol in module, or unknown parameter (see dmesg)

Is there a way to have the drivers installed but avoid these problems?
```
#!/bin/bash

if [ "$EUID" -ne 0 ]
  then echo "Please run as root!"
  exit
fi

echo 'Prepare system'

if [ -n "`which apt-get`" ]; then

V=$(uname -r | awk -F. '{print $1*100+$2}')
if [ $V -lt 41 ]; then
    echo "Please, install kernel ...."
    exit 1
fi
    apt-get update
    apt-get -y upgrade
    apt-get dist-upgrade
    apt-get -y install \
        build-essential \
        patchutils \
        libproc-processtable-perl \
        linux-headers-$(uname -r) \
        git
systemctl disable apt-daily.service
systemctl disable apt-daily.timer
rm -rf /usr/src/media_build /usr/src/media /usr/src/dvb-firmwares.tar.bz2
cd /usr/src
git clone --depth=1 https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media

elif [ -n "`which yum`" ]; then
        yum -y update
        yum -y upgrade
        yum -y group install "Development Tools"
        yum -y install epel-release
        yum -y install \
        perl-core \
        perl-Proc-ProcessTable \
        perl-Digest-SHA \
        kernel-headers \
        kernel-devel \
        elfutils-libelf-devel
rm -rf /usr/src/media_build /usr/src/media /usr/src/dvb-firmwares.tar.bz2
cd /usr/src
git clone --depth=1 https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
sed -i '/vm_fault_t;/d' /usr/src/media_build/v4l/compat.h
sed -i '/add v4.20_access_ok.patch/d' /usr/src/media_build/backports/backports.txt

fi

rm -rf /lib/modules/$(uname -r)/extra
rm -rf /lib/modules/$(uname -r)/kernel/drivers/media
rm -rf /lib/modules/$(uname -r)/kernel/drivers/staging/media

curl -L https://github.com/tbsdtv/media_build/releases/download/latest/dvb-firmwares.tar.bz2 | \
    tar -jxf - -C /lib/firmware/

cd /usr/src/media_build
make dir DIR=../media && \
    make allyesconfig && \
    make -j4 && \
    make install && \

echo 'Done! Please reboot the server'
```

crazycat69, 2024-03-11:
- not need remove kernel media drivers
- disable IR/RC support in media_build

See [media_build/install.sh](https://github.com/tbsdtv/media_build/blob/latest/install.sh) script for details

Liam-Whiteside, 2024-03-11:
Thanks, not sure why those rm -rf lines were in the script, or where that script originally came from, but it's been working after every OS update for a few years. Removing them appears to be avoiding the problem.

I'll leave disabling IR/RC support as a further step if required.

Liam-Whiteside, 2024-03-11:
Closing as solution provided worked.

closed 2024-03-11 by Liam-Whiteside