tbsdtv/linux_media #346: TBS6280 + opensuse 5.14.21-150500.55.52 + Unable to find symbol cxd2820r_attach()

Mr-Edgard, 2024-04-03:
Hello,

Should be something wrong on my side, but need help to find what...
After my system update, I tried to compile and install driver for my DVB card and I've this issue when booting:
_[    3.132609] DVB: Unable to find symbol cxd2820r_attach()
[    3.132657] SAA716x Budget 0000:04:00.0: TurboSight TBS 6280  frontend 0 attach failed
[    3.132876] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    3.141431] DVB: Unable to find symbol cxd2820r_attach()
[    3.141479] SAA716x Budget 0000:04:00.0: TurboSight TBS 6280  frontend 1 attach failed_

Of course, this issue make DVB card unavailable to mythbackend:
_Error: Failed to open device /dev/dvb/adapter0/frontend0_
I only get _demux0_, _dvr0_ and _net0_.

The method I use to compile is:
_rm -rf media_build media
VERS=latest
git clone https://github.com/tbsdtv/media_build.git -b $VERS
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b $VERS ./media
cd media_build
make dir DIR=../media
make allyesconfig
sed -i -r 's/(^CONFIG.*_RC.*=)./\1n/g' v4l/.config
sed -i -r 's/(^CONFIG.*_IR.*=)./\1n/g' v4l/.config
make -j4
cd media_build && make install
reboot_
I also suppress _v5.14_bus_void_return.patch_ and _v5.17_iosys.patch_ to be compatible with the system.
No error during compilation except _Skipping BTF generation_ but I don't think it's the issue.
I tried some tricks (ie: add _cxd2820r_ in _/etc/modules-load.d/dvb.conf_) but didn't found any way to fix this :-(

Hope somebody can helps,
MrEd

crazycat69, 2024-04-03:
`modprobe cxd2820r`

Mr-Edgard, 2024-04-04:
Hi,
Thank you for your answer!
Unfortunately, _modprobe_ command doesn't solve the issue.

It seems _dvbdev_ starts before I can load _cxd2820r_ then stay in error state. Loading _cxd2820r_ after (I think it's what modules-load.d do) doesn't fix it and I cannot restart _dvbdev_ init (I'm not sure this is a crystal clear explanation :-()...
Some extracts from _dmesg_:
```
[    2.465871] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                c3c0c45405d1da857fd1fb3071f24ae3249d0a60 media: dvb-usb/tbs5230: Minor init changes.
[    2.467721] SAA716x Budget 0000:04:00.0: enabling device (0000 -> 0002)
[    2.475727] libata version 3.00 loaded.
[    2.484093] sata_via 0000:02:00.0: version 2.6
[    2.484200] sata_via 0000:02:00.0: enabling device (0000 -> 0001)
[    2.484269] sata_via 0000:02:00.0: routed to hard irq line 11
[    2.484595] scsi host0: sata_via
[    2.486005] ahci 0000:00:17.0: version 3.0
[    2.492957] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
...
[    3.132609] DVB: Unable to find symbol cxd2820r_attach()
[    3.132657] SAA716x Budget 0000:04:00.0: TurboSight TBS 6280  frontend 0 attach failed
[    3.132876] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    3.141431] DVB: Unable to find symbol cxd2820r_attach()
[    3.141479] SAA716x Budget 0000:04:00.0: TurboSight TBS 6280  frontend 1 attach failed
```

And the result of _lsmod | egrep 'dvb|cxd'_:
```
cxd2820r               49152  0
regmap_i2c             16384  1 cxd2820r
saa716x_tbs_dvb        73728  0
tas2101                28672  1 saa716x_tbs_dvb
saa716x_core           86016  1 saa716x_tbs_dvb
dvb_core              176128  3 cxd2820r,saa716x_tbs_dvb,saa716x_core
cx24117                40960  1 saa716x_tbs_dvb
videobuf2_vmalloc      20480  2 uvcvideo,dvb_core
videobuf2_common       73728  5 videobuf2_vmalloc,videobuf2_v4l2,uvcvideo,dvb_core,videobuf2_memops
mc                     73728  5 videodev,videobuf2_v4l2,uvcvideo,dvb_core,videobuf2_common
i2c_algo_bit           16384  2 saa716x_tbs_dvb,i915
```

I don't know a lot about, but it looks like _tbs5230_ is loaded where _cxd2820r_ should...
Thanks again for your help!

Mr-Edgard, 2024-04-04:
Hi,
After browsing a little bit the code, I saw missing _cxd2820r_ happened in _saa716x_tbs_dvb_ module call...
So my work around is:
```
rmmod saa716x_tbs_dvd
modprobe  cxd2820r
modprobe saa716x_tbs_dvb
systemctl restart mythbackend
``` 
(well, except _restart_ which was for testing purpose, I put all the stuff in a script called from _ExecStartPre_ of _mytbackend systemd_)
After that, I've  _/dev/dvb/adapter0/frontend0_ (and _adapter1_ ;-))
In my _dmesg_, I now have:
```
[131530.133256] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[131530.750191] cxd2820r 1-006c: Sony CXD2820R successfully identified
[131530.763864] tda18212 1-0060: NXP TDA18212HN/M successfully identified
[131530.764884] SAA716x Budget 0000:04:00.0: TurboSight TBS 6280  MAC[0]=ff:ff:ff:ff:ff:ff
[131530.764896] SAA716x Budget 0000:04:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6280 DVB-T/T2/C)...
```
(and same for 2nd tuner).
So it works again but do you have any idea about the reason it doesn't start automatically like before?

Thanx and BR

crazycat69, 2024-04-04:
this some suse/rhel specific issue with module loading

Mr-Edgard, 2024-04-04:
OK... Don´t know if I'm able to close the ticket... This workaround satisfy me ;-).

closed 2024-04-04 by Mr-Edgard