tbsdtv/linux_media #348: snd_usb_audio error

sq7r0, 2024-04-21:
Hello,

I have an issue running TBS 5927 USB card on Linux box. Im using two usb sound cards and after latest updates (to be honest dont know which one) i have lost possibility to use both - TBS DVB card and soundcard in the same time. When i delete dvb drivers from modules im able to use my soundcards. When i install the driver (built from sources using https://github.com/tbsdtv/linux_media/wiki) TBS card is working fine, but the usb card driver cannot be loaded and i can see errors like below :

[    9.295408] snd_usb_audio: disagrees about version of symbol media_devnode_remove
[    9.295412] snd_usb_audio: Unknown symbol media_devnode_remove (err -22)
[    9.295566] snd_usb_audio: disagrees about version of symbol __media_device_register
[    9.295568] snd_usb_audio: Unknown symbol __media_device_register (err -22)
[    9.295614] snd_usb_audio: disagrees about version of symbol media_create_intf_link
[    9.295615] snd_usb_audio: Unknown symbol media_create_intf_link (err -22)
[    9.295621] snd_usb_audio: disagrees about version of symbol media_create_pad_link
[    9.295623] snd_usb_audio: Unknown symbol media_create_pad_link (err -22)
[    9.295694] snd_usb_audio: disagrees about version of symbol media_device_register_entity
[    9.295695] snd_usb_audio: Unknown symbol media_device_register_entity (err -22)
[    9.295814] snd_usb_audio: disagrees about version of symbol media_entity_pads_init
[    9.295815] snd_usb_audio: Unknown symbol media_entity_pads_init (err -22)
[    9.295846] snd_usb_audio: disagrees about version of symbol media_remove_intf_link
[    9.295847] snd_usb_audio: Unknown symbol media_remove_intf_link (err -22)
[    9.295952] snd_usb_audio: disagrees about version of symbol media_device_delete
[    9.295953] snd_usb_audio: Unknown symbol media_device_delete (err -22)
[    9.296002] snd_usb_audio: disagrees about version of symbol media_devnode_create
[    9.296003] snd_usb_audio: Unknown symbol media_devnode_create (err -22)
[    9.296014] snd_usb_audio: disagrees about version of symbol media_device_unregister_entity
[    9.296015] snd_usb_audio: Unknown symbol media_device_unregister_entity (err -22)
[    9.296049] snd_usb_audio: disagrees about version of symbol media_device_usb_allocate
[    9.296050] snd_usb_audio: Unknown symbol media_device_usb_allocate (err -22)

I guess its something with Linux media interface: v0.10 Currently im using kernel 6.8.7 but also tested on 6.6 longterm with the same effect.

thecosmicslug, 2024-06-19:
I am getting this error also. Now my USB speaker is no longer working! Does anyone have a work around?

kaswa42, 2025-01-04:
+1
my USB audio devices don't work after compiling the drivers

nick-reachcomp, 2025-03-12:
I'm also affected by this issue.
For me the command `sudo modprobe -f snd_usb_audio`
gives the error modprobe: `ERROR: could not insert 'snd_usb_audio': Exec format error`
that means I'm trying to use a module compiled for a different kernel version.
I'm on Xubuntu 24.04.2 LTS, kernel 6.8.0-55-generic https://github.com/tbsdtv/linux_media/pull/57 x86_64 updated with drivers for TBS 6280 on 2025-03-08.