tbsdtv/linux_media #349: Build fails on Fedora 40

jal-frezie, 2024-04-25:
Newly upgraded. 
$ uname -a
Linux hotwheels 6.8.7-300.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Apr 17 19:21:08 UTC 2024 x86_64 GNU/Linux
./install.sh 
make -C /home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l stagingconfig
make[1]: Entering directory '/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l'
No version yet, using 6.8.7-300.fc40.x86_64
make[2]: Entering directory '/home/jaspert/Build/TBS6280/tbsdriver/media_build/linux'
Syncing with dir ../../media
Patches for 6.8.7-300.fc40.x86_64 already applied.
make[2]: Leaving directory '/home/jaspert/Build/TBS6280/tbsdriver/media_build/linux'
./scripts/make_kconfig.pl /lib/modules/6.8.7-300.fc40.x86_64/build /lib/modules/6.8.7-300.fc40.x86_64/source 1 1
Preparing to compile for kernel version 6.8.7

***WARNING:*** You do not have the full kernel sources installed.
.
.
.
  CC [M]  /home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.o
.
.
/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.c: In function 'mxl603_ctrl_programRegisters':
/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.c:58:79: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   58 |                         ret = reg_read(dev->client,ctrlRegInfoPtr[i].regAddr, (int)&tmp);
      |                                                                               ^
/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.c:58:79: error: passing argument 3 of 'reg_read' makes pointer from integer without a cast [-Wint-conversion]
   58 |                         ret = reg_read(dev->client,ctrlRegInfoPtr[i].regAddr, (int)&tmp);
      |                                                                               ^~~~~~~~~
      |                                                                               |
      |                                                                               int
/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.c:5:63: note: expected 'u8 *' {aka 'unsigned char *'} but argument is of type 'int'
    5 | static int reg_read(struct i2c_client *client,u8 reg_addr,u8 *val)
      |                                                           ~~~~^~~
/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.c: In function 'mxl603_init':
/home/jaspert/Build/TBS6280/tbsdriver/media_build/v4l/mxl603.c:83:12: warning: unused variable 'dfeRegData' [-Wunused-variable]
   83 |         u8 dfeRegData;
      |            ^~~~~~~~~~

Any help welcome

crazycat69, 2024-04-25:
https://github.com/tbsdtv/linux_media/commit/eb50749344d4df1b15667308c6fc8b179a77d820

jal-frezie, 2024-04-26:
Thanks CC

closed 2024-08-02 by crazycat69