tbsdtv/linux_media #351: Compilation error with Linux 6.9 - TBS6904SE

jansohn, 2024-05-23:
Starts failing to build with linux kernel 6.9 on my system. 6.8 was working correctly.

    DKMS make.log for tbs_6904_se-6.9 for kernel 6.9.1-arch1-2 (x86_64)
    Thu May 23 11:36:52 PM CEST 2024
    make: Entering directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build'
    make -C linux/ dir DIR="../../media"
    make[1]: Entering directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build/linux'
    rm -rf drivers include sound mm .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
    Searching in ../../media/Makefile for kernel version.
    ./use_dir.pl ../../media
    sync file: include/linux/mmc/sdio_ids.h
    sync file: include/uapi/linux/v4l2-mediabus.h
    sync file: include/trace/events/v4l2.h
    sync file: include/misc/altera.h
    sync file: include/linux/dma/xilinx_dma.h
    sync file: include/uapi/linux/ivtvfb.h
    sync file: include/uapi/linux/v4l2-dv-timings.h
    sync file: sound/pci/bt87x.c
    sync file: include/uapi/linux/usb/video.h
    sync file: include/uapi/linux/cec-funcs.h
    sync file: include/uapi/linux/xilinx-v4l2-controls.h
    sync file: include/linux/atmel-isc-media.h
    sync file: include/trace/events/pwc.h
    sync file: include/uapi/linux/media-bus-format.h
    sync file: include/trace/events/vb2.h
    sync file: include/linux/usb/uvc.h
    sync file: include/linux/overflow.h
    sync file: include/linux/fixp-arith.h
    sync file: include/uapi/linux/smiapp.h
    sync file: include/linux/of_graph.h
    sync file: include/uapi/linux/videodev2.h
    sync file: include/linux/kconfig.h
    sync file: include/uapi/linux/media.h
    sync file: include/linux/ti_wilink_st.h
    sync file: include/linux/pci_ids.h
    sync file: include/uapi/linux/max2175.h
    sync file: include/uapi/linux/ccs.h
    sync file: include/uapi/linux/v4l2-subdev.h
    sync file: include/uapi/linux/uvcvideo.h
    sync file: include/uapi/linux/v4l2-common.h
    sync file: include/uapi/linux/lirc.h
    sync file: include/linux/via-core.h
    sync file: include/uapi/linux/cec.h
    sync file: include/uapi/linux/v4l2-controls.h
    sync file: include/linux/hdmi.h
    sync file: include/uapi/linux/ivtv.h
    sync file: include/sound/aci.h
    sync dir: drivers/media/
    sync dir: include/dt-bindings/media/
    sync dir: include/linux/platform_data/media/
    sync dir: drivers/misc/altera-stapl/
    sync dir: include/uapi/linux/dvb/
    sync dir: drivers/staging/media/
    sync dir: include/media/
    make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build/linux'
    make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build'
    make: Entering directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build'
    make -C /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l
    make[1]: Entering directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l'
    No version yet, using 6.7.0-arch3-1
    Updating/Creating .config
    make[2]: Entering directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build/linux'
    Syncing with dir ../../media
    Applying patches for kernel 6.9.1-arch1-2
    patch -s -f -N -p1 -i ../backports/api_version.patch
    patch -s -f -N -p1 -i ../backports/pr_fmt.patch
    patch -s -f -N -p1 -i ../backports/debug.patch
    patch -s -f -N -p1 -i ../backports/drx39xxj.patch
    patch -s -f -N -p1 -i ../backports/ccs.patch
    patch -s -f -N -p1 -i ../backports/rc-cec.patch
[... 221 lines cut ...]
                     from <command-line>:
    /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l/v4l2-spi.c: At top level:
    /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l/v4l2-spi.c:78:19: error: conflicting types for 'v4l2_spi_new_subdev'; have 'struct v4l2_subdev *(struct v4l2_device *, struct spi_master *, struct spi_board_info *)'
       78 | EXPORT_SYMBOL_GPL(v4l2_spi_new_subdev);
          |                   ^~~~~~~~~~~~~~~~~~~
    ./include/linux/export.h:56:28: note: in definition of macro '__EXPORT_SYMBOL'
       56 |         extern typeof(sym) sym;                                 \
          |                            ^~~
    ./include/linux/export.h:69:41: note: in expansion of macro '_EXPORT_SYMBOL'
       69 | #define EXPORT_SYMBOL_GPL(sym)          _EXPORT_SYMBOL(sym, "GPL")
          |                                         ^~~~~~~~~~~~~~
    /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l/v4l2-spi.c:78:1: note: in expansion of macro 'EXPORT_SYMBOL_GPL'
       78 | EXPORT_SYMBOL_GPL(v4l2_spi_new_subdev);
          | ^~~~~~~~~~~~~~~~~
    /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l/../linux/include/media/v4l2-common.h:286:21: note: previous declaration of 'v4l2_spi_new_subdev' with type 'struct v4l2_subdev *(struct v4l2_device *, struct spi_master *, struct spi_board_info *)'
      286 | struct v4l2_subdev *v4l2_spi_new_subdev(struct v4l2_device *v4l2_dev,
          |                     ^~~~~~~~~~~~~~~~~~~
    make[4]: *** [scripts/Makefile.build:244: /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l/v4l2-spi.o] Error 1
    make[3]: *** [/usr/lib/modules/6.9.1-arch1-2/build/Makefile:1919: /var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l] Error 2
    make[2]: *** [Makefile:240: __sub-make] Error 2
    make[2]: Leaving directory '/usr/lib/modules/6.9.1-arch1-2/build'
    make[1]: *** [Makefile:53: default] Error 2
    make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build/v4l'
    make: *** [Makefile:26: all] Error 2
    make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.9/build/media_build'

TBScardOwner, 2024-05-24:
I can confirm error on OpenSuSE Tumblewed with Kernel 6.9.1

alfrup, 2024-05-26:
I can confirm error on OpenSuSE Tumblewed with Kernel 6.9.1  too.

miguel1963, 2024-05-26:
Also same bug in Tumbleweed 6.9.1

jal-frezie, 2024-06-30:
To fix this you have to replace every occurrence of 'spi_master' in the code with 'spi_controller', and every occurrence of '[un]register_master' with '[un]register_controller'.

crazycat69, 2024-07-01:
[tbsdtv/media_build](https://github.com/tbsdtv/media_build) and [tbsdtv/linux_media](https://github.com/tbsdtv/linux_media) updated for 6.9 compatibility.
Please check

jansohn, 2024-07-10:
Working for me now, thank you very much!

closed 2024-07-10 by jansohn