tbsdtv/linux_media #353: Building on Pi 4 CM4 GCC13

SonnyWalkman, 2024-06-02:
Hello,
I've tried many attempts to build the TBS driver v4l tree on the Pi 4 running 24.04LTS Ubuntu server headless.

What could be my issue?   solo6x10 seems the issue when compiling.

Is there dpkg available?

Error I get below. 

> ........ compilation ..........

> CC [M]  /home/dvb/media_build/v4l/solo6x10-enc.o
> CC [M]  /home/dvb/media_build/v4l/solo6x10-v4l2-enc.o
> gcc-13: fatal error: Killed signal terminated program cc1
> compilation terminated.
> make[4]: *** [scripts/Makefile.build:243: /home/dvb/media_build/v4l/solo6x10-core.o] Error 1
> make[4]: *** Waiting for unfinished jobs....
> gcc-13: fatal error: Killed signal terminated program cc1
> compilation terminated.
> make[4]: *** [scripts/Makefile.build:243: /home/dvb/media_build/v4l/solo6x10-p2m.o] Error 1
> gcc-13: fatal error: Killed signal terminated program cc1
> compilation terminated.
> make[4]: *** [scripts/Makefile.build:243: /home/dvb/media_build/v4l/solo6x10-enc.o] Error 1
> make[3]: *** [/usr/src/linux-headers-6.8.0-1004-raspi/Makefile:1926: /home/dvb/media_build/v4l] Error 2
> make[2]: *** [Makefile:240: __sub-make] Error 2
> make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-1004-raspi'
> make[1]: *** [Makefile:53: default] Error 2
> make[1]: Leaving directory '/home/dvb/media_build/v4l'
> make: *** [Makefile:26: all] Error 2

SonnyWalkman, 2024-06-02:
Is there a method to just compile the TBS drivers (TBS-6522H) without a full line-up of cards?
Much easier and don't have to spend hours on the build process.

If so, how do you set the parameters?

crazycat69, 2024-06-02:
**make menuconfig**

SonnyWalkman, 2024-06-03:
> **make menuconfig**

Wish it was that simple.Unfortunately make menuconfig doesn't work

SonnyWalkman, 2024-06-03:
I've contacted Jack at TBS. He's advised building for arm64 doesn't work and offered a working arm32 image to test. Unfortunately, I'm using the newer TBS-6522H multisystem quad card and drivers aren't ready.

He's offered to build drivers for my card so hopefully I'll get it all working.

Having drivers working on arm64 should be the goal from here on?

If it not able to compile on arm64 why did the build script let me build 75% before crashing??

crazycat69, 2024-06-03:
arm64 supported, but you have some compiler issue (maybe too many CPU cores used for make ?)

SonnyWalkman, 2024-06-04:
Hi @crazycat69 

Tried using make with one processor.

Slow, slow and ended in tears with the same error. Obviously something not right with the code.

crazycat69, 2024-06-04:
Some internal compiler error, present in ubuntu 24.04 beta for x86_64 (resolved in release build)
https://github.com/tbsdtv/linux_media/issues/344

try disable **CONFIG_VIDEO_SOLO6X10** in **v4l/.config**:
> CONFIG_VIDEO_SOLO6X10 n

SonnyWalkman, 2024-06-05:
Hi again @crazycat69 

Thanks for the tip. I've commented out and are now recompiling.

I'm just wanting to compile the TBS card drivers. What other dependencies do I need to do this?  Could I simply comment most of the drivers except tbs in v4l/.config? I'm assuming there'll need some base drives which I don't know which one's I require for a working system?

crazycat69, 2024-06-05:
better use **make menuconfig** (require ncurces lib) or **make config** (interactive console config)
Disable RC and CEC support, Cameras and video grabbers, Analog TV, AM/FM receivers, SDR, Paltform specific drivers and unused USB, PCI media devices.

SonnyWalkman, 2024-06-05:
Thanks @crazycat69 

I needed to add ```dtoverlay=pcie-32bit-dma

```to the config.txt file.

I'll install ncurses and try make menu config to select my single tbs
Below is a view of the lsmod command

![image](https://github.com/tbsdtv/linux_media/assets/54062411/7038aaac-a723-4db5-a609-81109e8ceb2f)

sunsettrack4, 2024-06-27:
> Hi @crazycat69
> 
> Tried using make with one processor.
> 
> Slow, slow and ended in tears with the same error. Obviously something not right with the code.
> 
> Tried using gcc 12 and gcc 13. Both pulled the same error.
> 
> Not sure where to go from here?

Alternative solution: Just use a larger swap file (4096 MB): https://pimylifeup.com/raspberry-pi-swap-file/ and one processor

closed 2024-08-02 by crazycat69