tbsdtv/linux_media #354: TBS 6209 on ubuntu 24.04

xlazom00, 2024-06-02:
TBS 6209 on ubuntu 24.04

[    5.125881] TBSECP3 driver 0000:02:00.0: i2c nack (7)
[    5.125951] TBSECP3 driver 0000:02:00.0: i2c nack (7)
[    5.125960] si2157: probe of 19-0063 failed with error -5
[    5.125974] TBSECP3 driver 0000:02:00.0: TBSECP3 frontend 4 attach failed
[    5.125979] TBSECP3 driver 0000:02:00.0: frontend attach failed
[    5.126253] TBSECP3 driver 0000:02:00.0: adapter4 attach failed
[    5.126260] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    5.253282] TBSECP3 driver 0000:02:00.0: MAC address 00:22:ab:b1:4d:85
[    5.255503] si2183 4-0067: Silicon Labs Si2183 successfully attached
[    5.257657] TBSECP3 driver 0000:02:00.0: i2c nack (7)
[    5.257718] TBSECP3 driver 0000:02:00.0: i2c nack (7)
[    5.257724] si2157: probe of 19-0060 failed with error -5
[    5.257733] TBSECP3 driver 0000:02:00.0: TBSECP3 frontend 5 attach failed
[    5.257738] TBSECP3 driver 0000:02:00.0: frontend attach failed
[    5.257995] TBSECP3 driver 0000:02:00.0: adapter5 attach failed

2 devices are missing
with latest drivers build

crazycat69, 2024-06-02:
look like hw issue, one si21832 chip failed (two logical demods)

closed 2024-08-02 by crazycat69

jdiaz080592, 2024-09-05:
> TBS 6209 on ubuntu 24.04
>
> [ 5.125881] TBSECP3 driver 0000:02:00.0: i2c nack (7) [ 5.125951] TBSECP3 driver 0000:02:00.0: i2c nack (7) [ 5.125960] si2157: probe of 19-0063 failed with error -5 [ [...]

Did u Fix this error? I got the same problem with my TBS6909x V2

xlazom00, 2024-09-05:
> Did u Fix this error? I got the same problem with my TBS6909x V2

All fixed on my side

jdiaz080592, 2024-09-05:
> All fixed on my side

Can u tell me how did u fix it?

xlazom00, 2024-09-05:
> Can u tell me how did u fix it?

I did pull latest code from git
And all works as expected