tbsdtv/linux_media #356: DVB-GSE pseudo TS seems to have changed (TBS6903X)

mrwish7, 2024-06-13:
It used to be possible to decode the DVB-S2 GSE transmission on 23.5°E 12641 V 1500 with my TBS6903X card. Since recently though, I'm no longer able to decode the encapsulated data (EDI with DAB multiplexes) using the pts2bbf ( https://github.com/newspaperman/bbframe-tools/ ) and eti-tools ( https://github.com/piratfm/eti-tools ).

The card will still successfully lock the transmission with the above parameters, but the pseudo-TS PID seems to now be 280 (0x0118) instead of 270 (0x010e). Even modifying the pts2bbf tool to take into account this PID change, the GSE data is no longer successfully decoded.

I've had confirmation from @andimik that the transmission works in the same way as before with his setup (without having updated any drivers in several months). The transmission should be GSE with two MIS streams (ISI 1 and ISI 2), each containing EDI UDP streams.

Confusingly, I have not been able to fix this by downgrading the drivers, so I'm unsure how this still works in some cases but not for me! I may not have successfully downgraded though. I can confirm that everything worked fine as of the latest version of the drivers in March this year.

I've attached a sample pseudo TS file recorded this morning, as you can see the encapsulated PID is 280 for the data. Just to confirm, this is definitely from Linux rather than Windows! There doesn't seem to be any problem with the stream itself, there's no missing packets and the signal level is good (13dB C/N).

[12641V_GSE.zip](https://github.com/user-attachments/files/15817005/12641V_GSE.zip)

crazycat69, 2024-06-13:
**options stid135 bbframe=1**

crazycat69, 2024-06-13:
And you can use [gse branch](https://github.com/tbsdtv/linux_media/tree/gse) with GSE decapsulation support in dvbnet subsystem. Work only with stid135 in bbframe=0 mode.

P.S, dvbnet tools with GSE support
https://github.com/tbsdtv/dvb-apps

mrwish7, 2024-06-14:
Thanks, I've set the bbframe option in the stid135 mod config and it now works perfectly as before.

```
tune-s2 12641 V 1500 -lnb UNIVERSAL -system DVB-S2
```
```
dvbstream -o 8192 | pts2bbf | bbfedi2eti -mis 1 | ni2out --list
dvbstream v0.7.01 - (C) Dave Chapman 2001-2004
Released under the GPL.
Latest version available from http://dvbtools.sourceforge.net/
dvbstream will stop after -1 seconds (71582788 minutes)
option misOutput to stdout
Streaming 1 stream
[06/14/24 10:08:25] Initialise next pseq to 24824
[06/14/24 10:08:25] EDI-AF: Packet sequence error
DR Deutschland   (0x10bc)
 0 : KLASSIK RADIO    (0xd75b) Pri subch= 6 start=192 CUs= 54 PL=eep-3a bitrate=72 DAB+
 1 : SUNSHINE LIVE    (0x15dc) Pri subch=21 start=330 CUs= 54 PL=eep-3a bitrate=72 DAB+
 2 : radio horeb      (0xd01c) Pri subch= 5 start=156 CUs= 36 PL=eep-3a bitrate=48 DAB+
 3 : ERF Plus         (0x1a64) Pri subch= 2 start=  0 CUs= 48 PL=eep-3a bitrate=64 DAB+
 4 : Dlf              (0xd210) Pri subch=10 start=480 CUs=104 PL=eep-2a bitrate=104 DAB+
 5 : Dlf Kultur       (0xd220) Pri subch=11 start=584 CUs=112 PL=eep-2a bitrate=112 DAB+
 6 : Dlf Nova         (0xd230) Pri subch=12 start=696 CUs=104 PL=eep-2a bitrate=104 DAB+
 7 : DRadio DokDeb    (0xd240) Pri subch=13 start=800 CUs= 36 PL=eep-3a bitrate=48 DAB+
 8 : Schwarzwaldradio (0x100d) Pri subch= 7 start=246 CUs= 42 PL=eep-2b bitrate=64 DAB+
 9 : ENERGY           (0x1a45) Pri subch= 4 start=102 CUs= 54 PL=eep-3a bitrate=72 DAB+
10 : SCHLAGERPARADIES (0x10c3) Pri subch=20 start=288 CUs= 42 PL=eep-2b bitrate=64 DAB+
11 : Absolut relax    (0x17fa) Pri subch= 3 start= 48 CUs= 54 PL=eep-3a bitrate=72 DAB+
12 : RADIO BOB!       (0x15dd) Pri subch=22 start=384 CUs= 54 PL=eep-3a bitrate=72 DAB+
```

closed 2024-06-14 by mrwish7