tbsdtv/linux_media #357: Compilation error with Linux 6.10 - TBS6904SE

jansohn, 2024-07-26:
Here we go again... Starts failing to build with linux kernel 6.10 on my system. 6.9 was working correctly.

    DKMS make.log for tbs_6904_se-6.10 for kernel 6.10.1-arch1-1 (x86_64)
    Fri Jul 26 10:34:35 PM CEST 2024
    make: Entering directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build'
    make -C linux/ dir DIR="../../media"
    make[1]: Entering directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build/linux'
    rm -rf drivers include sound mm .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
    Searching in ../../media/Makefile for kernel version.
    ./use_dir.pl ../../media
    sync file: include/trace/events/vb2.h
    sync file: include/misc/altera.h
    sync file: include/uapi/linux/smiapp.h
    sync file: include/linux/via-core.h
    sync file: include/uapi/linux/media.h
    sync file: include/uapi/linux/v4l2-mediabus.h
    sync file: include/uapi/linux/ivtv.h
    sync file: include/sound/aci.h
    sync file: include/uapi/linux/v4l2-subdev.h
    sync file: include/linux/mmc/sdio_ids.h
    sync file: include/uapi/linux/media-bus-format.h
    sync file: include/linux/ti_wilink_st.h
    sync file: include/uapi/linux/usb/video.h
    sync file: include/linux/dma/xilinx_dma.h
    sync file: include/uapi/linux/cec-funcs.h
    sync file: include/linux/pci_ids.h
    sync file: include/trace/events/pwc.h
    sync file: include/uapi/linux/v4l2-dv-timings.h
    sync file: include/uapi/linux/uvcvideo.h
    sync file: include/linux/atmel-isc-media.h
    sync file: include/uapi/linux/v4l2-common.h
    sync file: include/uapi/linux/v4l2-controls.h
    sync file: include/linux/fixp-arith.h
    sync file: include/trace/events/v4l2.h
    sync file: include/uapi/linux/videodev2.h
    sync file: include/linux/overflow.h
    sync file: sound/pci/bt87x.c
    sync file: include/uapi/linux/xilinx-v4l2-controls.h
    sync file: include/uapi/linux/cec.h
    sync file: include/linux/usb/uvc.h
    sync file: include/uapi/linux/ccs.h
    sync file: include/linux/of_graph.h
    sync file: include/linux/hdmi.h
    sync file: include/uapi/linux/lirc.h
    sync file: include/uapi/linux/max2175.h
    sync file: include/linux/kconfig.h
    sync file: include/uapi/linux/ivtvfb.h
    sync dir: drivers/media/
    sync dir: include/media/
    sync dir: drivers/misc/altera-stapl/
    sync dir: include/dt-bindings/media/
    sync dir: drivers/staging/media/
    sync dir: include/linux/platform_data/media/
    sync dir: include/uapi/linux/dvb/
    make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build/linux'
    make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build'
    make: Entering directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build'
    make -C /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l
    make[1]: Entering directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l'
    No version yet, using 6.9.7-arch1-1
    Updating/Creating .config
    make[2]: Entering directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build/linux'
    Syncing with dir ../../media
    Applying patches for kernel 6.10.1-arch1-1
    patch -s -f -N -p1 -i ../backports/api_version.patch
    patch -s -f -N -p1 -i ../backports/pr_fmt.patch
    patch -s -f -N -p1 -i ../backports/debug.patch
    patch -s -f -N -p1 -i ../backports/drx39xxj.patch
    patch -s -f -N -p1 -i ../backports/ccs.patch
    patch -s -f -N -p1 -i ../backports/rc-cec.patch
[... 72 lines cut ...]
    In file included from /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l/gx1133.c:26:
    ./include/linux/i2c-mux.h:58:5: note: declared here
       58 | int i2c_mux_add_adapter(struct i2c_mux_core *muxc,
          |     ^~~~~~~~~~~~~~~~~~~
    /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l/gx1133.c:816:15: error: too many arguments to function 'i2c_mux_add_adapter'
      816 |         ret = i2c_mux_add_adapter(priv->muxc, 0, 1, 0);
          |               ^~~~~~~~~~~~~~~~~~~
    ./include/linux/i2c-mux.h:58:5: note: declared here
       58 | int i2c_mux_add_adapter(struct i2c_mux_core *muxc,
          |     ^~~~~~~~~~~~~~~~~~~
    /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l/gx1133.c: In function 'gx1133_set_frontend':
    /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l/gx1133.c:1537:9: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
     1537 |         if (ret)
          |         ^~
    /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l/gx1133.c:1540:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
     1540 |                 if (fe->ops.tuner_ops.set_params) {
          |                 ^~
    make[4]: *** [scripts/Makefile.build:244: /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l/gx1133.o] Error 1
    make[3]: *** [/usr/lib/modules/6.10.1-arch1-1/build/Makefile:1934: /var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l] Error 2
    make[2]: *** [Makefile:240: __sub-make] Error 2
    make[2]: Leaving directory '/usr/lib/modules/6.10.1-arch1-1/build'
    make[1]: *** [Makefile:53: default] Error 2
    make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build/v4l'
    make: *** [Makefile:26: all] Error 2
    make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.10/build/media_build'

TBScardOwner, 2024-08-01:
make -C /lib/modules/6.10.2-1-default/build M=/root/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-6.10.2-1-obj/x86_64/default'
  LD [M]  /root/media_build/v4l/ccs.o
  LD [M]  /root/media_build/v4l/cx25840.o
  LD [M]  /root/media_build/v4l/et8ek8.o
  LD [M]  /root/media_build/v4l/msp3400.o
  LD [M]  /root/media_build/v4l/s5c73m3.o
  LD [M]  /root/media_build/v4l/tda18271.o
  CC [M]  /root/media_build/v4l/af9013.o
  CC [M]  /root/media_build/v4l/atbm8830.o
  CC [M]  /root/media_build/v4l/au8522_common.o
  CC [M]  /root/media_build/v4l/au8522_dig.o
/root/media_build/v4l/af9013.c: In function 'af9013_probe':
/root/media_build/v4l/af9013.c:1483:15: error: too many arguments to function 'i2c_mux_add_adapter'
 1483 |         ret = i2c_mux_add_adapter(state->muxc, 0, 0, 0);
      |               ^~~~~~~~~~~~~~~~~~~
In file included from /root/media_build/v4l/af9013_priv.h:18,
                 from /root/media_build/v4l/af9013.c:11:
/usr/src/linux-6.10.2-1/include/linux/i2c-mux.h:58:5: note: declared here
   58 | int i2c_mux_add_adapter(struct i2c_mux_core *muxc,
      |     ^~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-6.10.2-1/scripts/Makefile.build:245: /root/media_build/v4l/af9013.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-6.10.2-1/Makefile:1951: /root/media_build/v4l] Error 2
make[2]: *** [../../../linux-6.10.2-1/Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-6.10.2-1-obj/x86_64/default'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/media_build/v4l'
make: *** [Makefile:26: all] Error 2

crazycat69, 2024-08-02:
[tbsdtv/media_build](https://github.com/tbsdtv/media_build) and [tbsdtv/linux_media](https://github.com/tbsdtv/linux_media) updated for 6.10 compatibility.
Please check

TBScardOwner, 2024-08-02:
Thank you, works well for me.

jansohn, 2024-08-02:
Works for me, too. Thanks!

closed 2024-08-02 by jansohn