tbsdtv/linux_media #359: Failed compilation on kernel 6.11.0-1

alfrup, 2024-09-30:
I need to compile a driver for a TBS6590 device. When I execute make -j4 I get the following errors
/usr/src/linux-6.11.0-1/include/linux/minmax.h:167:9: note: this is the location of the previous definition
  167 | #define min3(x, y, z) \
      |         ^~~~
/usr/local/sbin/media_build/v4l/compat.h:239:9: warning: "max3" redefined
  239 | #define max3(x, y, z) max((typeof(x))max(x, y), z)
      |         ^~~~
/usr/src/linux-6.11.0-1/include/linux/minmax.h:176:9: note: this is the location of the previous definition
  176 | #define max3(x, y, z) \
      |         ^~~~
make[3]: *** [/usr/src/linux-6.11.0-1/Makefile:1949: /usr/local/sbin/media_build/v4l] Fehler 2
make[2]: *** [../../../linux-6.11.0-1/Makefile:224: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-6.11.0-1-obj/x86_64/default'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/usr/local/sbin/media_build/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2
[TBS6590.txt](https://github.com/user-attachments/files/17184713/TBS6590.txt)

crazycat69, 2024-10-03:
[tbsdtv/media_build](https://github.com/tbsdtv/media_build) and [tbsdtv/linux_media](https://github.com/tbsdtv/linux_media) updated for 6.11 compatibility.
Please check

alfrup, 2024-10-04:
> Antwort an: tbsdtv/linux_media
> (Issue #359)
>
> tbsdtv/media_build [1] and tbsdtv/linux_media [2] updated for 6.11

OK,
Thank you

[4].

[1] tbsdtv/media_build https://github.com/tbsdtv/media_build
[2] tbsdtv/linux_media https://github.com/tbsdtv/linux_media
[3] view it on GitHub
    https://github.com/tbsdtv/linux_media/issues/359#issuecomment-2392283548
[4] unsubscribe