tbsdtv/linux_media #360: 6903x Under Linux: Very Slow Zapping/Tuning With TVHeadend

ilikenwf, 2024-10-01:
I got headend setup and have attempted this with both @deeptho and your stock kernel modules under Archlinux, kernel 6.10.10. 

For whatever reason my 6903x is very slow to zap - 1-3 minutes before the channels appear. Right now I'm testing on 99.2W, which has several PBS channels on Ku band, relatively low symbol rates, but I have a lock and my set top receiver has no trouble locking and switching between them rapidly.

I have not yet tried any other software for tuning with this card; and it does look like someone put the capacitor mod on it before I purchased it second hand.

For all I know I'm doing something wrong with the headend config but I don't think I am; and I'm currently at a loss.

TVHeadend Output:
```
Oct 01 05:55:10 settop tvheadend[779]: htsp: 127.0.0.1 [ Kodi Media Center ]: Identified as user 'kodi'
Oct 01 05:55:10 settop tvheadend[779]: htsp: 127.0.0.1 [ kodi | Kodi Media Center ]: Privileges updated
Oct 01 05:55:28 settop tvheadend[779]: mpegts: 12095H in 99.2W - tuning on TurboSight TBS 6903x  #0 : DVB-S #0
Oct 01 05:55:30 settop tvheadend[779]: subscription: 0001: "127.0.0.1 [ kodi | Kodi Media Center ]" subscribing on channel "99.2W/12095H/{PMT:1000}", weight: 150, adapter: "TurboSi>
Oct 01 06:09:34 settop tvheadend[779]: http: x.x.x.x: using ticket BB019EA5B982E733A6D3F9C1A0F05C86C3A133C1 for /stream/service/2c90edc9dd79b02d3c92a31fdfe44379
Oct 01 06:09:34 settop tvheadend[779]: subscription: 0002: "HTTP" subscribing to service "99.2W/12095H/{PMT:1000}", weight: 100, adapter: "TurboSight TBS 6903x  #0 : DVB-S #0", net>
Oct 01 06:09:46 settop tvheadend[779]: subscription: 0002: "HTTP" unsubscribing, hostname="x.x.x.x", client="VLC/3.0.21 LibVLC/3.0.21"
Oct 01 06:09:49 settop tvheadend[779]: subscription: 0001: "127.0.0.1 [ kodi | Kodi Media Center ]" unsubscribing from "99.2W/12095H/{PMT:1000}", hostname="127.0.0.1", username="ma>
Oct 01 06:10:09 settop tvheadend[779]: http: x.x.x.x: using ticket 076680E580FBF428678CC6D2CE4D1B0FC1EAED54 for /stream/service/4deace5972b7c5c941cb5982661261be
Oct 01 06:10:09 settop tvheadend[779]: mpegts: 12140V in 99.2W - tuning on TurboSight TBS 6903x  #0 : DVB-S #0
Oct 01 06:10:10 settop tvheadend[779]: subscription: 0003: "HTTP" subscribing to service "99.2W/12140V/HD03", weight: 100, adapter: "TurboSight TBS 6903x  #0 : DVB-S #0", network: >
Oct 01 06:10:29 settop tvheadend[779]: webui: Stop streaming /stream/service/4deace5972b7c5c941cb5982661261be?ticket=076680E580FBF428678CC6D2CE4D1B0FC1EAED54, timeout waiting for p>
Oct 01 06:10:29 settop tvheadend[779]: subscription: 0003: "HTTP" unsubscribing, hostname="x.x.x.x", client="VLC/3.0.21 LibVLC/3.0.21"
Oct 01 06:10:29 settop tvheadend[779]: http: x.x.x.x: using ticket 076680E580FBF428678CC6D2CE4D1B0FC1EAED54 for /stream/service/4deace5972b7c5c941cb5982661261be
Oct 01 06:10:29 settop tvheadend[779]: mpegts: 12140V in 99.2W - tuning on TurboSight TBS 6903x  #0 : DVB-S #0
Oct 01 06:10:32 settop tvheadend[779]: subscription: 0004: "HTTP" subscribing to service "99.2W/12140V/HD03", weight: 100, adapter: "TurboSight TBS 6903x  #0 : DVB-S #0", network: >
Oct 01 06:10:50 settop tvheadend[779]: webui: Stop streaming /stream/service/4deace5972b7c5c941cb5982661261be?ticket=076680E580FBF428678CC6D2CE4D1B0FC1EAED54, timeout waiting for p>
Oct 01 06:10:50 settop tvheadend[779]: subscription: 0004: "HTTP" unsubscribing, hostname="x.x.x.x", client="VLC/3.0.21 LibVLC/3.0.21"
Oct 01 06:12:43 settop tvheadend[779]: http: x.x.x.x: using ticket 076680E580FBF428678CC6D2CE4D1B0FC1EAED54 for /stream/service/4deace5972b7c5c941cb5982661261be
Oct 01 06:12:43 settop tvheadend[779]: mpegts: 12140V in 99.2W - tuning on TurboSight TBS 6903x  #0 : DVB-S #0
Oct 01 06:12:46 settop tvheadend[779]: subscription: 0005: "HTTP" subscribing to service "99.2W/12140V/HD03", weight: 100, adapter: "TurboSight TBS 6903x  #0 : DVB-S #0", network: >
Oct 01 06:12:49 settop tvheadend[779]: subscription: 0005: "HTTP" unsubscribing, hostname="x.x.x.x", client="VLC/3.0.21 LibVLC/3.0.21"
Oct 01 06:12:53 settop tvheadend[779]: mpegts: 12095H in 99.2W - tuning on TurboSight TBS 6903x  #0 : DVB-S #0
Oct 01 06:12:55 settop tvheadend[779]: subscription: 0006: "127.0.0.1 [ kodi | Kodi Media Center ]" subscribing on channel "99.2W/12095H/{PMT:1000}", weight: 150, adapter: "TurboSi>

```

Getting lots of dmesg spam:

```
[ 1152.412181] FE_STiD135_GetDemodLock:2608 demod=0: fec=0 dmd=0 signal=0 carr=0 vit=0 sync=0 timedout=1 lock=0
[ 1152.412190] fe_stid135_search:3147 demod=0: error=0
[ 1152.414873] stid135_set_parameters:1418 set state->signal_info pls_mode=0x0 pls_code=0x1 stream_id=0ffffffff
[ 1152.415174] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1152.415183] fe_stid135_get_signal_info:3838 demod=0: Called fe_stid135_get_lock_status: carrier=0 sync=0
[ 1152.502760] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1152.504386] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1152.504401] dvb_frontend: dvb_frontend_thread:854: Adapter=0 Adding event val=0x21 old=0x0
[ 1152.504467] stid135_send_long_master_cmd:2009 demod=0.0 diseqc
[ 1152.513145] fe_stid135_diseqc_init:6346 DISEQC rf_in=0 error=0 mode=2 mode_fld=2 envelope_fld=0
[ 1152.577935] stid135_send_long_master_cmd:2071 demod=0.0 diseqc sent: rf_in=0; tuner[0].use_count=1 rf_in[0].use_count=1
[ 1152.577951] dvb_frontend: dvb_frontend_handle_ioctl:3377: sending master_cmd done err=0
[ 1153.557139] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1153.557157] stid135_tune_:1716 demod=0 re_tune=0
[ 1153.557536] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1153.559094] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1153.559104] dvb_frontend: dvb_frontend_thread:854: Adapter=0 Adding event val=0x21 old=0x0
[ 1154.581138] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1154.581157] stid135_tune_:1716 demod=0 re_tune=0
[ 1154.581534] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1154.583101] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1155.605106] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1155.605124] stid135_tune_:1716 demod=0 re_tune=0
[ 1155.605503] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1155.607044] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1156.629147] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1156.629165] stid135_tune_:1716 demod=0 re_tune=0
[ 1156.629574] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1156.631099] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1157.653137] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[... 41 lines cut ...]
[ 1167.893452] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1167.894942] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1168.917144] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1168.917168] stid135_tune_:1716 demod=0 re_tune=0
[ 1168.917668] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1168.919714] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1169.941123] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1169.941142] stid135_tune_:1716 demod=0 re_tune=0
[ 1169.941586] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1169.943280] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1170.965123] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1170.965147] stid135_tune_:1716 demod=0 re_tune=0
[ 1170.965671] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1170.967417] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1171.989110] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1171.989136] stid135_tune_:1716 demod=0 re_tune=0
[ 1171.989524] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1171.991296] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1173.013125] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1173.013150] stid135_tune_:1716 demod=0 re_tune=0
[ 1173.013622] fe_stid135_get_lock_status:1781 demod=0 demodState=0/0
[ 1173.015269] stid135_tune_:1758 demod=0 LOCK TIME 2024ms locked=0
[ 1174.037124] stid135_set_sec_ready_:1687 demod=0.0 Marking rf_in as configured active_rf_in=000000002f6d2793 old=1
[ 1174.037149] stid135_tune_:1716 demod=0 re_tune=0
```

crazycat69, 2024-10-01:
What modulation and SR ?

ilikenwf, 2024-10-01:
It applies to any of the FTA channels on 99.2W - for 12095H it's:

DVB-S2 QPSK 3676 4/5

https://www.lyngsat.com/Galaxy-16-and-T11-T14.html

ilikenwf, 2024-10-02:
@crazycat69 I should also mention that I know I have a good signal and everything - I also have an enigma2 box that flips through all the channels very quickly by comparison on this sat.

ilikenwf, 2024-10-02:
Looking at the headend logs, it seems to me more that this may be some driver issue going by the persistend dmesg output? I can try the other coax port, or try the build against a different kernel, currently running the 6.10.10-zen kernel.

ilikenwf, 2024-10-02:
> Your logs only contain part of the story. It shows the services don't lock (at first). Most likely cause is an incorrect configuration of tvheadend (e.g., incorrect adapter numbers). If multiple services are attached to the same channel, tvheadend tries and retries different configurations, which could explain why they eventually lock.

I can try to get a more complete log collection if that would help.

For adapters and my Diseqc setup everything functions properly including positioning, C and Ku bands at least, and htsp is setup in the default manner.

I'm happy to share any config files or screenshots as well if that would be helpful. My end goal is to have both Neumo and headend behaving properly.

While I have my e2 box for casual watching it would be preferable to get parity with the TBS card there for a number of reasons.

Oh and I don't post on the forums but I think we both frequent Rick's satellite? He isnjudt a few hours away from me and some of my equipment came from him.

Thank you for the input, please advise me on how I can further share to debug.

deeptho, 2024-10-02:
tvheadend is complex and in my experience it is very easy to get things wrong.
So if you can start with neumo, it will be much easier to debug.

I use tbs6909x myself with tvheadend. tbs6903x is only slightly different and less well tested, 
but since you have problems with both tbs and blindscan drivers, it is not likely to be a driver problem.

ilikenwf, 2024-10-04:
@deeptho @crazycat69 just for reference, on the low signal rate Ku TP I'm currently locked on, I'm getting a great signal:

9.2 dB SNR
-53.6 dBm signal strength

ilikenwf, 2024-10-08:
So I'm still fighting to get scanning/tuning to work in neumodvb - right now it crashes any time I try to do so. Here's my tuner settings in Headend...everything scans in ok, and I can usually, eventually get tuned in, but it spins for minutes sometimes.

[image]

ilikenwf, 2024-10-08:
Ok, so I think I figured it out. The rotor init time was set to 2 minutes for some reason under advanced, which obviously is far too long for diseqc...

closed 2024-10-08 by ilikenwf