tbsdtv/linux_media #363: Build fails on Ubuntu with kernel 6.8.0

mareklab, 2024-11-18:
Just to let you know, the build fails on Ubuntu 24.04 TLS with kernel 6.8.0, I've tried both the latest gcc-13 and older gcc-11
```
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-i2c.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-p2m.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-v4l2.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-tw28.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-gpio.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-disp.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-enc.o
  CC [M]  /root/tbs-driver/media_build/v4l/solo6x10-v4l2-enc.o
gcc-13: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [scripts/Makefile.build:243: /root/tbs-driver/media_build/v4l/solo6x10-p2m.o] Error 1
make[4]: *** Waiting for unfinished jobs....
gcc-13: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [scripts/Makefile.build:243: /root/tbs-driver/media_build/v4l/solo6x10-core.o] Error 1
gcc-13: fatal error: Killed signal terminated program cc1
compilation terminated.
make[4]: *** [scripts/Makefile.build:243: /root/tbs-driver/media_build/v4l/solo6x10-enc.o] Error 1
make[3]: *** [/usr/src/linux-headers-6.8.0-48-generic/Makefile:1925: /root/tbs-driver/media_build/v4l] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-48-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/root/tbs-driver/media_build/v4l'
```

Then it works fine on the same Ubuntu 24.04 TLS with my previous kernel 5.15, built with gcc-13

mareklab, 2024-11-18:
RAM 4 GB, CPU i3

crazycat69, 2024-11-19:
swap enabled ?
https://github.com/tbsdtv/linux_media/issues/358

mareklab, 2024-11-19:
Sure. Let me test the swap utilization when it fails... (update coming soon)

mareklab, 2024-11-19:
Okay, you were right. The problem is `make -j4` - since it runs 4 processes at once, it requires a lot of RAM to finish all build task. 

There  are 2 solutions: either just run `make`, or increase swap size (5GB worked for me, it failed with my initial 1GB)

Thanks for the hint!

closed 2024-11-19 by mareklab