tbsdtv/linux_media #367: Compilation error with Linux 6.12 - TBS6904SE

jansohn, 2024-12-02:
Seems like we need an update for 6.12 compatibility.

```
DKMS make.log for tbs_6904_se-6.11 for kernel 6.12.1-arch1-1 (x86_64)
Mon Dec  2 09:06:28 PM CET 2024
make: Entering directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build'
make -C linux/ dir DIR="../../media"
make[1]: Entering directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build/linux'
rm -rf drivers include sound mm .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
Searching in ../../media/Makefile for kernel version.
./use_dir.pl ../../media
sync file: include/trace/events/pwc.h
sync file: include/uapi/linux/media-bus-format.h
sync file: include/uapi/linux/uvcvideo.h
sync file: include/linux/kconfig.h
sync file: include/linux/of_graph.h
sync file: include/linux/usb/uvc.h
sync file: include/linux/atmel-isc-media.h
sync file: include/linux/overflow.h
sync file: include/linux/mmc/sdio_ids.h
sync file: include/uapi/linux/xilinx-v4l2-controls.h
sync file: include/uapi/linux/ivtv.h
sync file: include/misc/altera.h
sync file: include/sound/aci.h
sync file: include/uapi/linux/usb/video.h
sync file: include/trace/events/vb2.h
sync file: include/uapi/linux/v4l2-dv-timings.h
sync file: include/uapi/linux/lirc.h
sync file: include/uapi/linux/media.h
sync file: include/uapi/linux/v4l2-subdev.h
sync file: include/linux/ti_wilink_st.h
sync file: sound/pci/bt87x.c
sync file: include/trace/events/v4l2.h
sync file: include/uapi/linux/cec-funcs.h
sync file: include/uapi/linux/smiapp.h
sync file: include/uapi/linux/videodev2.h
sync file: include/linux/dma/xilinx_dma.h
sync file: include/uapi/linux/cec.h
sync file: include/linux/fixp-arith.h
sync file: include/linux/hdmi.h
sync file: include/linux/pci_ids.h
sync file: include/uapi/linux/v4l2-mediabus.h
sync file: include/uapi/linux/v4l2-controls.h
sync file: include/linux/via-core.h
sync file: include/uapi/linux/ccs.h
sync file: include/uapi/linux/max2175.h
sync file: include/uapi/linux/v4l2-common.h
sync file: include/uapi/linux/ivtvfb.h
sync dir: include/uapi/linux/dvb/
sync dir: drivers/misc/altera-stapl/
sync dir: include/dt-bindings/media/
sync dir: drivers/media/
sync dir: drivers/staging/media/
sync dir: include/media/
sync dir: include/linux/platform_data/media/
make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build/linux'
make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build'
make: Entering directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build'
make -C /var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l
make[1]: Entering directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l'
No version yet, using 6.11.4-arch1-1
Updating/Creating .config
make[2]: Entering directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build/linux'
Syncing with dir ../../media
Applying patches for kernel 6.12.1-arch1-1
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/ccs.patch
[... 117 lines cut ...]
 2215 |         struct i2c_adapter *i2c = dev->base->i2c;
      |                             ^~~
/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/m88rs6060.c: In function 'm88rs6060_get_frontend':
/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/m88rs6060.c:3186:29: warning: unused variable 'i2c' [-Wunused-variable]
 3186 |         struct i2c_adapter *i2c = dev->base->i2c;
      |                             ^~~
/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/m88rs6060.c: At top level:
/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/m88rs6060.c:645:13: warning: 'si5351_drive_strength' defined but not used [-Wunused-function]
  645 | static void si5351_drive_strength(struct si5351_priv *priv, enum si5351_clock clk, enum si5351_drive drive)
      |             ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/v4l2-dv-timings.o
  CC [M]  /var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/v4l2-dev.o
/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/v4l2-dev.c:487:19: error: 'no_llseek' undeclared here (not in a function); did you mean 'noop_llseek'?
  487 |         .llseek = no_llseek,
      |                   ^~~~~~~~~
      |                   noop_llseek
make[4]: *** [scripts/Makefile.build:229: /var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l/v4l2-dev.o] Error 1
make[3]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l] Error 2
make[2]: *** [Makefile:224: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build/v4l'
make: *** [Makefile:26: all] Error 2
make: Leaving directory '/var/lib/dkms/tbs_6904_se/6.11/build/media_build'
```

crazycat69, 2024-12-11:
[tbsdtv/media_build](https://github.com/tbsdtv/media_build) and [tbsdtv/linux_media](https://github.com/tbsdtv/linux_media) updated for 6.12 compatibility.
Please check

jansohn, 2024-12-13:
working, thanks for the update!

closed 2024-12-13 by jansohn