tbsdtv/linux_media #370: Compilation error: /media_build/v4l/ccs-core.c:668:21: error: too many arguments to function 'pm_runtime_get_if_active'

expupil, 2024-12-12:
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
make -j4

make -C /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l 
make[1]: Entering directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l'
scripts/make_makefile.pl
./scripts/make_myconfig.pl
[ ! -f "./config-mycompat.h" ] && echo "/* empty config-mycompat.h */" > "./config-mycompat.h" || true
make -C firmware prep
creating symbolic links...
perl scripts/make_config_compat.pl /lib/modules/6.8.0-50-generic/build ./.myconfig ./config-compat.h
make[2]: Entering directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/firmware'
Kernel build directory is /lib/modules/6.8.0-50-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/linux'
Syncing with dir ../../media
Patches for 6.8.0-50-generic already applied.
make[2]: Leaving directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/linux'
make -C /lib/modules/6.8.0-50-generic/build M=/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-6.8.0-50-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/max2175.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ad5820.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adp1653.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7170.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7175.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7180.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7183.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7343.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7393.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7511-v4l2.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7604.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/adv7842.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ak7375.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ak881x.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/aptina-pll.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ar0521.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/bt819.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/bt856.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/bt866.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-core.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-reg-access.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-quirk.o
  CC [M]  /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-limits.o
/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-core.c: In function 'ccs_set_ctrl':
/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-core.c:668:21: error: too many arguments to function 'pm_runtime_get_if_active'
  668 |         pm_status = pm_runtime_get_if_active(&client->dev, true);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-core.c:22:
./include/linux/pm_runtime.h:75:12: note: declared here
   75 | extern int pm_runtime_get_if_active(struct device *dev);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [scripts/Makefile.build:243: /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l/ccs-core.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.8.0-50-generic/Makefile:1925: /home/chris/Downloads/tbs-open-linux-driver/media_build/v4l] Error 2
make[2]: *** [Makefile:240: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.8.0-50-generic'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/chris/Downloads/tbs-open-linux-driver/media_build/v4l'
make: *** [Makefile:26: all] Error 2

Westfale63, 2024-12-12:
Exactly same problem for me.

Bazoogle, 2024-12-12:
Same thing over here as well

crazycat69, 2024-12-12:
Look like forced backport from newer kernels > 6.8
Disable **v6.8-ccs.patch** in [media_build/backports/backports.txt](https://github.com/tbsdtv/media_build/blob/latest/backports/backports.txt#L44)

Westfale63, 2024-12-12:
> Look like forced backport from newer kernels > 6.8 Disable **v6.8-ccs.patch** in [media_build/backports/backports.txt](https://github.com/tbsdtv/media_build/blob/latest/backports/backports.txt#L44)

No, sadly this doesn't work for me. Errors still appearing

crazycat69, 2024-12-12:
cd media_build
**make make cleanall**
make dir DIR=../media
./install.sh

Westfale63, 2024-12-12:
Sorry I made a mistake, the hint from crazycat69 one hour ago, works!
Thank you very much for your tip 👍

Bazoogle, 2024-12-12:
> Look like forced backport from newer kernels > 6.8 Disable **v6.8-ccs.patch** in [media_build/backports/backports.txt](https://github.com/tbsdtv/media_build/blob/latest/backports/backports.txt#L44)

I just deleted the media and media_build folders, redownloaded, commented out the line "add v6.8-ccs.patch" like you suggested, and retried it like normal and it got past the point it was getting stuck on with "ccs-core.c". Thank you for such quick responses and all your work on this project!

expupil, 2024-12-13:
Thanks for that advice, when using Kernel 6.8.0-50-generic, the code now compiles without errors. Unfortunately no drivers are loaded after the reboot.
I have reverted to booting to 6.8.0-49-generic for the moment as the driver is working here.

crazycat69, 2024-12-13:
need disable RC/IR support before compilation.
use [media-build/install.sh](https://github.com/tbsdtv/media_build/blob/latest/install.sh) script for config/compile/install

Madhorse1, 2025-03-04:
> need disable RC/IR support before compilation. use [media-build/install.sh](https://github.com/tbsdtv/media_build/blob/latest/install.sh) script for config/compile/install

Hi, can you please specify how to disable  v6.8-ccs.patch? I commented out the line 47 in the backports.txt file, but it did not solve the issue.
Thnx

Madhorse1, 2025-03-04:
Correction. It works if done from the beginning.

seapoint, 2025-03-11:
> Look like forced backport from newer kernels > 6.8 Disable **v6.8-ccs.patch** in [media_build/backports/backports.txt](https://github.com/tbsdtv/media_build/blob/latest/backports/backports.txt#L44)

Thanks a million Crazycat69 - this sorted my problem for me. I wonder will this be resolved in a future build?

burny0202, 2025-03-29:
Yep - Just had this issue. Commented out v6.8-ccs.patch  media_build/backports/backports.txt and seems to work

Delitants, 2025-06-04:
Why not to apply this fix to master branch? It's annoying.

crazycat69, 2025-06-04:
This not fix, ths workaround for custom case with forced backports on ubuntu 6.8 kernels..
LTS kernels like 6.1, 5.15, 5.4, 4.19 require this patch for compilation.

Bazoogle, 2025-06-04:
> This not fix, this workaround for custom case with forced backports on ubuntu 6.8 kernels..

Would the fix be to check Kernel version and exclude the patch for certain ones? Or any other check that could be done so it still works on the newer version without manually commenting it out?

crazycat69, 2025-06-04:
Ubuntu kernels before 6.8.0-50 require this patch, but later forced backport patches require disable it. Kernel version still same. 

P.S. Redhat users (CentsOS, SuSe and etc) accept it and don't ask any questions :) Use true LTS distribs with LTS kernels :)

Delitants, 2025-06-05:
> Ubuntu kernels before 6.8.0-50 require this patch, but later forced backport patches require disable it. Kernel version still same.
> 
> P.S. Redhat users (CentsOS, SuSe and etc) accept it and don't ask any questions :) Use true LTS distribs with LTS kernels :)

I use Ubuntu 24.04.2 LTS with default kernel it comes with it, -55