tbsdtv/linux_media #371: tbs_pcie.c:1519:10: error: 'struct snd_pcm_ops' has no member named 'copy_user' in RHEL9.5

pszemus, 2024-12-13:
Hello
I am having problems building the `linux_media` project for RedHat Enterprise Linux 9 (RHEL 9).
I've successfully built `linux_media` for RHEL < 9.5 (with some backport patches handling), but now, in RHEL 9.5, kernel `5.14.0-503.15.1.el9_5` has backported changes from kernel 6.6 and the `snd_pcm_opts` struct has the field `copy_user` renamed to `copy`.

This line tries to handle that: https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/pci/tbscapture/tbs_pcie.c#L1518 but fails for RHEL's kernel because it's version is 5.14.0 while at the same time the `copy` field should be used.

```
v4l/tbs_pcie.c:1519:10: error: 'struct snd_pcm_ops' has no member named 'copy_user'
 1519 |         .copy_user =    tbs_pcie_audio_copy_user,
```

```
# grep -F *copy /usr/src/kernels/5.14.0-503.15.1.el9_5.x86_64/include/sound/pcm.h 
	int (*copy)(struct snd_pcm_substream *substream, int channel,
```

How to handle that case? Should we remove the `#if LINUX_VERSION_CODE` completely and relay only on backport patches in  [media_build](https://github.com/tbsdtv/media_build) project?

crazycat69, 2024-12-13:
change #if LINUX_VERSION_CODE < KERNEL_VERSION(6, 6, 0) to #if LINUX_VERSION_CODE < KERNEL_VERSION(5, 14, 0)
https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/pci/tbscapture/tbs_pcie.c#L1483
https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/pci/tbscapture/tbs_pcie.c#L1518

pszemus, 2024-12-13:
@crazycat69 Are you sure? The general approach seems right as `snd_pcm_ops->copy` was introduced in kernel 6.6.0. It is RedHat that backported this change to RHEL 9.5 (5.14.0-503.15.1.el9_5)

crazycat69, 2024-12-13:
i not use distribs with agressively patched old kernels, but in your case need change tbs_pcie.c and disable v6.5_pcm_copy.patch

tuxjz, 2025-04-06:
I know this is RHEL related issue, but I think Rocky Linux 9 (Rocky 9.5 - `5.14.0-503.35.1.el9_5.x86_64`) has the same issue...

I have tried to change the `tbs_pcie.c` and disabled pcm_copy patch... and it ended up even worse I think:

```
v4l/solo6x10-g723.c:266:10: error: 'const struct snd_pcm_ops' has no member named 'copy_user'
  266 |         .copy_user = snd_solo_pcm_copy_user,
      |          ^~~~~~~~~
v4l/solo6x10-g723.c:266:22: error: initialization of 'int (*)(struct snd_pcm_substream *, struct timespec64 *, struct timespec64 *, struct snd_pcm_audio_tstamp_config *, struct snd_pcm_audio_tstamp_report *)' from incompatible pointer type 'int (*)(struct snd_pcm_substream *, int,  long unsigned int,  void *, long unsigned int)' [-Werror=incompatible-pointer-types]
  266 |         .copy_user = snd_solo_pcm_copy_user,
      |                      ^~~~~~~~~~~~~~~~~~~~~~
v4l/solo6x10-g723.c:266:22: note: (near initialization for 'snd_solo_pcm_ops.get_time_info')
v4l/solo6x10-g723.c:267:10: error: 'const struct snd_pcm_ops' has no member named 'copy_kernel'
  267 |         .copy_kernel = snd_solo_pcm_copy_kernel,
      |          ^~~~~~~~~~~
v4l/solo6x10-g723.c:267:24: error: initialization of 'int (*)(struct snd_pcm_substream *, int,  long unsigned int,  long unsigned int)' from incompatible pointer type 'int (*)(struct snd_pcm_substream *, int,  long unsigned int,  void *, long unsigned int)' [-Werror=incompatible-pointer-types]
  267 |         .copy_kernel = snd_solo_pcm_copy_kernel,
```

My next attempt instead of linux_media wiki docs (as that already failed as well) was using the steps in this script (respectively the related forum post): https://github.com/nt74/rocky9-installers/blob/main/install-tbsdtv-drivers.sh but that haven't helped either.

So I will just let it there for others and follow this issue for updates... thanks for any help :)

closed 2026-02-06 by Davin622